zephyrproject-rtos/zephyr #63564: Spurious `warning: division by zero is undefined` on `K_THREAD_DEFINE` when building with clang

fabiobaltieri, 2023-10-05:
**Describe the bug**
Hi, bumped into this compiler warning on `K_THREAD_DEFINE` since https://github.com/zephyrproject-rtos/zephyr/pull/62900 got merged. One of our project that is building for an stm32 with LLVM started failing with a compiler warning:

```
/mnt/host/source/src/platform/ec/zephyr/shim/src/tasks.c:51:1251: error: division by zero is undefined [-Werror,-Wdivision-by-zero]
 struct z_thread_stack_element __attribute__((section("." "noinit" "." "\"EC_BASE/ec/zephyr/shim/src/tasks.c\"" "." "21"))) __attribute__((__aligned__(((((0x40) > (0)) ? (0x40) : (0)))))) _k_thread_stack_RWSIG[(((((unsigned long)(1536) + ((unsigned long)(8) - 1)) / (unsigned long)(8)) * (unsigned long)(8)) + ((size_t)0x40))]; struct k_thread _k_thread_obj_RWSIG; __attribute__((__aligned__(__alignof(struct _static_thread_data)))) struct _static_thread_data _k_thread_data_RWSIG __attribute__((section("." "__static_thread_data" "." "static" "." "_k_thread_data_RWSIG_"))) __attribute__((__used__)) = { .init_thread = (&_k_thread_obj_RWSIG), .init_stack = (_k_thread_stack_RWSIG), .init_stack_size = (1536), .init_entry = (k_thread_entry_t)rwsig_task, .init_p1 = (void *)0, .init_p2 = (void *)((void*)0), .init_p3 = (void *)((void*)0), .init_prio = ((EC_TASK_PRIO_COUNT - EC_TASK_RWSIG_PRIO - 1)), .init_options = (0), .init_delay = ((k_timeout_t) { .ticks = (((-1)) == (-1) ? ((k_ticks_t) -1) : ((k_ticks_t)((1) ? ( ((10000) == (1000)) ? (uint64_t) (((((-1)) > (0)) ? ((-1)) : (0))) : ((1000) > (10000) && (1000) % (10000) == 0U) ? ((uint64_t) (((((-1)) > (0)) ? ((-1)) : (0))) + ((0) ? ((1000) / (10000)) / 2 : (1) ? ((1000) / (10000)) - 1 : 0)) / ((1000) / (10000)) : ((10000) > (1000) && (10000) % (1000) == 0U) ? (uint64_t) (((((-1)) > (0)) ? ((-1)) : (0)))*((10000) / (1000)) : ((((((365 * 24ULL * 3600ULL * 1000) + (4294967295U) - 1) / (4294967295U)) * 10000) <= 4294967295U) ? (((uint64_t) (((((-1)) > (0)) ? ((-1)) : (0)))*(10000) + ((0) ? (1000) / 2 : (1) ? (1000) - 1 : 0)) / (1000)) : (((uint64_t) (((((-1)) > (0)) ? ((-1)) : (0))) / (1000))*(10000) + (((uint64_t) (((((-1)) > (0)) ? ((-1)) : (0))) % (1000))*(10000) + ((0) ? (1000) / 2 : (1) ? (1000) - 1 : 0)) / (1000))) ) : (((uint64_t) (((((-1)) > (0)) ? ((-1)) : (0))) / (1000))*(10000) + (((uint64_t) (((((-1)) > (0)) ? ((-1)) : (0))) % (1000))*(10000) + ((0) ? (1000) / 2 : (1) ? (1000) - 1 : 0)) / (1000)) ))) }), .init_name = "RWSIG", }; const k_tid_t RWSIG = (k_tid_t)&_k_thread_obj_RWSIG;
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                  ^ ~~~~~~~~~~~~~~~~~~
```

The cause seems to be the introduction of `SYS_TIMEOUT_MS` in `Z_THREAD_INITIALIZER`, the macro eventually expands into some call of `z_tmcvt_64` which includes a `/ (Z_HZ_ms / Z_HZ_cyc)`, that becomes a `/ 0` in most tickless setups since `Z_HZ_ms = 1000` and `Z_HZ_cyc = CONFIG_SYS_CLOCK_TICKS_PER_SEC = 10000`. The ternary pick the valid value and GCC is happy with that, but clang seems to be firing a warning.

I managed to reproduce this with this example:

```c
#include <stdio.h>

int ticks = 0 ? 1 / 0 : 0 ;

int main(int argc, char **argv)
{
        printf("%d\n", ticks);
}
```

```
$ gcc -Wall -o test test.c
$ clang -Wall -o test test.c
test.c:3:19: warning: division by zero is undefined [-Wdivision-by-zero]
int ticks = 0 ? 1 / 0 : 0 ;
                  ^ ~
1 warning generated.
```

Note that for some reason the warning does not fire if the condition is moved inside the function.

The CI did not catch it because the clang build that we run is all for `native_posix` and runs with `SYS_CLOCK_TICKS_PER_SEC = 100`, but this can be reproduced upstream fairly easily with:

```
$ export ZEPHYR_TOOLCHAIN_VARIANT=llvm
$ west build -p -b native_posix_64 samples/basic/threads -DCONFIG_SYS_CLOCK_TICKS_PER_SEC=10000
...
/usr/local/google/home/fabiobaltieri/zephyrproject/zephyr/samples/basic/threads/src/main.c:111:1: warning: division by zero is undefined [-Wdivision-by-zero]
K_THREAD_DEFINE(blink0_id, STACKSIZE, blink0, NULL, NULL, NULL,
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
...
```

To get around this in our application we ended up selectively disabling the warning (https://chromium-review.googlesource.com/4907933), but that's obviously not a scalable workaround. Unfortunately the way our CI is setup does not give us much visibility into it, but there's probably many users compling with clang for tickless systems and I think this should be looked into before the release.

I tried to look into the https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/sys/time_units.h definitions but can't quite see a decent workaround, looking for more ideas around this.

**To Reproduce**
```
$ export ZEPHYR_TOOLCHAIN_VARIANT=llvm
$ west build -p -b native_posix_64 samples/basic/threads -DCONFIG_SYS_CLOCK_TICKS_PER_SEC=10000
```

**Expected behavior**
No warning

**Impact**
Build fail if warning as treated as errors.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: clang 14.0.6 and 17.0.0

fabiobaltieri, 2023-10-05:
cc @keith-packard @andyross @nashif @ceolin @dcpleung @peter-mitsis @cfriedt

cfriedt, 2023-10-05:
I would suggest reverting for now. Can respond after the release

fabiobaltieri, 2023-10-05:
> I would suggest reverting for now. Can respond after the release

It's certainly an option but let's hear what the author and reviewers think about it, if can come up with a workaround before RC2 I think we should give it a go.

fabiobaltieri, 2023-10-05:
Maybe we can just cap that fraction result to a minimum of 1?

```diff
--- a/include/zephyr/sys/time_units.h
+++ b/include/zephyr/sys/time_units.h
@@ -175,7 +175,7 @@ static TIME_CONSTEXPR inline int sys_clock_hw_cycles_per_sec(void)
 #define z_tmcvt_int_div_64(__t, __from_hz, __to_hz, __round_up, __round_off) \
        ((uint64_t) (__t) + z_tmcvt_off_div(__from_hz, __to_hz,         \
                                            __round_up, __round_off)) / \
-       ((__from_hz) / (__to_hz))
+       (MAX((__from_hz) / (__to_hz), 1))
 
 /* Integer multiplcation 64-bit conversion */
 #define z_tmcvt_int_mul_64(__t, __from_hz, __to_hz)    \
```

I'll open a PR see if that would pass CI.

fabiobaltieri, 2023-10-05:
> The logic _should_ be using a multiply instead of divide for this case -- `* (Z_HZ_cyc / Z_HZ_ms)` instead of `/ (Z_HZ_ms / Z_HZ_cyc)`. Sounds like it should also be fairly easy to fix once I figure out why it did the wrong thing :-)

Ok it looks like there's more into it -- I closed https://github.com/zephyrproject-rtos/zephyr/pull/63580, will let you come up with the right fix before I mess things up even more. :-)

fgrandel, 2023-10-05:
> once I figure out why it did the wrong thing :-)

@keith-packard I think it *does* the right thing, the problem seems to be a compiler artifact as the nice example @fabiobaltieri constructed shows:

> int ticks = 0 ? 1 / 0 : 0 ;

Found https://stackoverflow.com/questions/26923182/how-to-eliminate-divide-by-0-error-in-template-code - maybe related?

I agree this should be easy to fix this. Already started experimenting...

keith-packard, 2023-10-05:
As far as I can determine, the compiler is complaining about a divide-by-zero in a branch *which will not be taken*.

Take a look at this code:
```
int i = (10 > 100 ? (20 / (10 / 100)) : (20 * (100 / 10)));
```

```
$ clang -c -O2 -Wall -Wextra foo.c
foo.c:1:25: warning: division by zero is undefined [-Wdivision-by-zero]
int i = (10 > 100 ? (20 / (10 / 100)) : (20 * (100 / 10)));
                        ^ ~~~~~~~~~~
1 warning generated.
```
I've submitted a PR which works around the warning by using 1 when the value would be zero. It cannot affect the result
because this code is in a conditional branch where from_hz (1000) is strictly greater than to_hz (10000).

fabiobaltieri, 2023-10-05:
> As far as I can determine, the compiler is complaining about a divide-by-zero in a branch _which will not be taken_.

Yes that is the case in my tests (the build fails with a different error if that branch is taken), and from my understanding the condition it cannot be taken in any case.

fgrandel, 2023-10-05:
> As far as I can determine, the compiler is complaining about a divide-by-zero in a branch which will not be taken.

Exactly!

To me this looks like a bug in CLang: See https://stackoverflow.com/questions/18385020/can-code-that-will-never-be-executed-invoke-undefined-behavior

keith-packard, 2023-10-05:
> To me this looks like a bug in CLang: See https://stackoverflow.com/questions/18385020/can-code-that-will-never-be-executed-invoke-undefined-behavior

Yeah, thinking about this again, the C 'short circuit' rules apply to the conditional operator which means that the undefined operation (divide by zero) will not happen, and hence the program is not invoking undefined behavior.

It's no different from using an if statement to protect the division operation.

fgrandel, 2023-10-05:
> It's no different from using an if statement to protect the division operation.

Haha, my example in https://github.com/llvm/llvm-project/issues/68339

keith-packard, 2023-10-05:
> > It's no different from using an if statement to protect the division operation.
> 
> Haha, my example in [llvm/llvm-project#68339](https://github.com/llvm/llvm-project/issues/68339) - first time ever I open a bug against a compiler, probably quite ingenuous on my part. Let's see whether they laugh at me...

I'd bet they'll come back with chapter-and-verse from the standard explaining why the compiler should work this way. That's the usual result of filing compiler issues.

fgrandel, 2023-10-05:
> I'd bet they'll come back with chapter-and-verse from the standard

Nope - we just earned a "confirmed" label - see https://github.com/llvm/llvm-project/issues/68339

Nice.

closed 2023-10-06 by jhedberg

marc-hb, 2023-11-30:
The warning is back. Bigger hammer submitted:
- https://github.com/zephyrproject-rtos/zephyr/pull/65953

PR #63589 (2023-10-06): sys/time_units.h: Work around clang div-by-zero warning
clang emits a warning for this code:

  foo.c:1:25: warning: division by zero is undefined [-Wdivision-by-zero]
  int i = (10 > 100 ? (20 / (10 / 100)) : (20 * (100 / 10)));
	                        ^ ~~~~~~~~~~

The warning is generated for code whose value does not affect the expression result, but technically it is still 'undefined behavior'.

Work around this warning by checking for a zero divisor and substituting one to make the compiler happy.

Closes: #63564