zephyrproject-rtos/zephyr #63586: device: DEVICE_DECLARE doesn't work in C++

jamie-bes, 2023-10-05:
**Describe the bug**

I'm working on a Zephyr module that uses C++20 and involves several custom drivers. Some incompatibility issues with C++ and Zephyr's driver macros were fixed in #56333, but I ran into a new issue yesterday regarding the `DEVICE_DECLARE` macro. Here's a minimal reproducible example in which the exact code that builds correctly in C results a compile error when C++ is enabled.

**To Reproduce**
1. `west init` in new directory, then `west update`.
2. `cd zephyr/samples/hello_world`
3. Change the contents of `main.c` to the following:

```
#include <stdio.h>
#include <zephyr/device.h>

DEVICE_DECLARE(my_driver);

void test_device_get(void) {
	const struct device* my_driver_lookup = DEVICE_GET(my_driver);
}

int my_driver_init(const struct device *dev) {
	test_device_get();

    return 0;
}

DEVICE_DEFINE(
    my_driver,
    "My Driver",
    my_driver_init,
    NULL,
    NULL,
    NULL,
    POST_KERNEL,
    CONFIG_KERNEL_INIT_PRIORITY_DEFAULT,
    NULL
);

int main(void) {
	printf("Hello World! %s\n", CONFIG_BOARD);
	return 0;
}
```

4. Build with `west build -b nucleo_f207zg`. This should build successfully.
5. Now rename `main.c` to `main.cpp`, change `target_sources(app PRIVATE src/main.c)` in `CMakeLists.txt` to `target_sources(app PRIVATE src/main.cpp)`, and change the contents of `prj.conf` to the following:

```
CONFIG_CPP=y
CONFIG_STD_CPP20=y
CONFIG_REQUIRES_FULL_LIBCPP=y
```

6. Build with `west build -b nucleo_f207zg --pristine` and it will fail.

The issue appears to be with `DEVICE_DECLARE` and not with `DEVICE_DEFINE` -- if I delete `DEVICE_DECLARE` and reorder things in the file such that the "forward declaration" of `my_driver` isn't needed, then it'll build correctly in both C and C++.

**Logs and console output**

```
west build -b nucleo_f207zg --pristine
-- west build: making build dir /home/jamie/test_device_declare/zephyr/samples/hello_world/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/jamie/test_device_declare/zephyr/samples/hello_world
-- CMake version: 3.22.4
-- Found Python3: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/jamie/.cache/zephyr
-- Zephyr version: 3.5.0-rc1 (/home/jamie/test_device_declare/zephyr)
-- Found west (found suitable version "1.1.0", minimum required is "0.14.0")
-- Board: nucleo_f207zg
[... 90 lines cut ...]
   85 | #define DEVICE_NAME_GET(dev_id) _CONCAT(__device_, dev_id)
      |                                         ^~~~~~~~~
/home/jamie/test_device_declare/zephyr/include/zephyr/toolchain/common.h:137:26: note: in definition of macro '_DO_CONCAT'
  137 | #define _DO_CONCAT(x, y) x ## y
      |                          ^
/home/jamie/test_device_declare/zephyr/include/zephyr/device.h:85:33: note: in expansion of macro '_CONCAT'
   85 | #define DEVICE_NAME_GET(dev_id) _CONCAT(__device_, dev_id)
      |                                 ^~~~~~~
/home/jamie/test_device_declare/zephyr/include/zephyr/device.h:325:36: note: in expansion of macro 'DEVICE_NAME_GET'
  325 |         static const struct device DEVICE_NAME_GET(dev_id)
      |                                    ^~~~~~~~~~~~~~~
/home/jamie/test_device_declare/zephyr/samples/hello_world/src/main.cpp:5:1: note: in expansion of macro 'DEVICE_DECLARE'
    5 | DEVICE_DECLARE(my_driver);
      | ^~~~~~~~~~~~~~
[36/135] Building C object zephyr/CMakeFiles/zephyr.dir/soc/arm/st_stm32/common/soc_config.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /home/jamie/.local/bin/cmake --build /home/jamie/test_device_declare/zephyr/samples/hello_world/build
```

**Environment:**
 - OS: Ubuntu 20.04.6 LTS
 - Toolchain:
   - Zephyr SDK: 0.16.3
   - West: v1.1.0
 - Commit SHA or Version used: `236318332b`

gmarull, 2023-10-06:
C++ doesn't seem to allow having a tentative definition of a variable (what DEVICE_DECLARE does). So I guess the only thing you can do is to define the device early.

Note that writing drivers in C++ is not officially supported in Zephyr. Also, using DEVICE_DEFINE is discouraged nowadays, better look into DT-based devices (where this problem will actually disappear because DT devices have external visibility).

jamie-bes, 2023-10-06:
Ah, I wasn't familiar with C's rules for tentative definitions and had assumed `DEVICE_DECLARE` was only doing a declaration rather than a definition. But I understand it better now that I'm looking at how the macro is written.

From `device.h`:

```
#define DEVICE_DECLARE(dev_id)                                                 \
	static const struct device DEVICE_NAME_GET(dev_id)
```

However, I tried changing it to a declaration with `extern` and that actually compiles in both C and C++.

```
#define DEVICE_DECLARE(dev_id)                                                 \
	extern const struct device DEVICE_NAME_GET(dev_id)
```

I'm not sure if making that change would end up breaking something else. But this is the behavior that I'd expect based on the name `DEVICE_DECLARE`, and it also eliminates the inconsistency between C and C++ here.

I use `DEVICE_DT_INST_DEFINE` for most normal drivers, but there are some special cases where having the user explicitly create an instance of a device through the device tree isn't suitable, and that's where `DEVICE_DEFINE` comes in. (My particular use case is effectively dependency injection of one driver into another, with the dependency created by a module and selected by the user with a string property in a device tree overlay. I've got it working without `DEVICE_DECLARE`, but my current workaround is to do a runtime lookup of the dependency with `device_get_binding()` rather than a static lookup with `DEVICE_GET()`.)

cfriedt, 2023-10-06:
I don't think there is an easy solution for `DEVICE_DECLARE()` unfortunately as what you're describing seems to be a [language limitation](https://stackoverflow.com/a/936589).

It might make sense to trigger a build assert if `DEVICE_DECLARE()` is used from C++.

A kind of weird alternative would be to do this:

main.cpp
```cpp
extern "C" {
extern const struct device *my_driver_ptr;
};
```
devices.c
```cpp
DEVICE_DEFINE(my_driver, "My Driver", my_driver_init, NULL, NULL, NULL, POST_KERNEL,
	      CONFIG_KERNEL_INIT_PRIORITY_DEFAULT, NULL);
const struct device *my_driver_ptr = DEVICE_GET(my_driver);
```

cfriedt, 2023-10-06:
@jamie-bes - if you like, you could submit a PR for the `extern` workaround. That might be the simplest, but it would also need to work the same with all toolchains / architectures.

gmarull, 2023-10-06:
Note that non-DT devices do not have external visibility (they are declared as static). This is mainly because device name uniqueness is hard to guarantee, unlike DT where ordinal is used. DEVICE_DECLARE is meant to be used in the same C file, so using extern feels weird. I’ll take a closer look next week and see if there’s any solution…

closed 2023-12-21 by github-actions[bot]