zephyrproject-rtos/zephyr #63651: Add support for external sensor clocks

yperess, 2023-10-06:
## Introduction

Many sensors support having an external clock. This will likely be a simple device tree binding in the driver, but there's no way to see this information externally. This can become more important as sensors start reporting time in ticks of their own clocks which will increase accuracy.

### Problem description

My current first pass at the async sensor API uses nanosecond timestamps. While those work fine, they do have accumulated error over time. Using the internal clock of the SoC or the sensor's external clock can allow us to get much more accurate timestamps. This would mean converting the decoded timestamps to ticks, but also including the associate'd clock's `const struct device *`

### Proposed change

Add the clock's device to the sensor_info struct and the default binding. Then change the decoder API to use ticks and include the associated clock device.

yperess, 2023-10-06:
@teburd @MaureenHelm @tristan-google for visibility

teburd, 2023-10-12:
In thinking about this I'm not sure an external clock needs to do much besides provide metadata to the sensor driver about its clock rate and its presence for a sensor driver.

Ticks being used as the measurement makes me think we are talking about kernel ticks, most other APIs in zephyr when talking about cycle counts use the term cycles.

If we wish to use cycle accurate timestamps (makes sense) then I'd recommend that's the terminology we use rather than ticks.

yperess, 2023-10-17:
@teburd updated the proposal based on our discussion, I think that the Kconfig isn't as useful anymore since some sensors might want to use the system clock and others might want external ones.

KeHIntel, 2023-11-13:
@yperess @teburd I have some comments after reading this proposal:

1. CPU cycle may not be suitable for sensor timestamping, because CPU cycle accumulation is not linear to time, there are a lot of factors that could impact CPU cycle counts, like dynamic frequency, CPU idle, CPU power gate, etc.

2. Timer should be a common design for Zephyr RTOS, it shouldn't be a specific design for sensors. Zephyr should support different accuracy level timers based on HW capability (like in x86 case, we have RTC timer and HPET timer), as well as conversion and correction methods, for sensor and other subsystems to use.

teburd, 2023-11-13:
> @yperess @teburd I have some comments after reading this proposal:
>
> 1. CPU cycle may not be suitable for sensor timestamping, because CPU cycle accumulation is not linear to time, there are a lot of [...]

I don’t disagree, at the same time kernel ticks have very low precision. So I do think we need to have a better solution here of some sort. There are also devices which provide their own time stamp from their internal sample clock and it would be nice to reuse that!

yperess, 2025-01-15:
> [@teburd](https://github.com/teburd) updated the proposal based on our discussion, I think that the Kconfig isn't as useful anymore since some sensors might want to use the system clock and others might want external ones.

I take this back. I did some work exploring this and I feel that it's a nightmare to provide different clocks to different sensors. It makes sensor fusion nearly impossible. I'm happy with the implementation we landed on in https://github.com/zephyrproject-rtos/zephyr/pull/80189 as it provides a compile time wrapping for multiple clock APIs.

closed 2025-01-15 by yperess