zephyrproject-rtos/zephyr #63660: MAX17048 driver does not retrieve properties if the battery is fully charged

martincalsyn, 2023-10-07:
**Describe the bug**
The drivers/fuel_gauge/max17048 driver code in max17048.c skips retrieval of fuel_gauge properties if the battery
is fully charged but still attached to an external power source.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
    All - this is not platform-dependent
 - What have you tried to diagnose or workaround this issue?
    Step through with debugger. Retrieve values through full charge/discharge cycles.
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
   No. This appears to have been in the code since it's contribution.

**To Reproduce**
Steps to reproduce the behavior:
1. Using either a real max17048, battery, and charging subsystem, charge the battery and leave the charger attached
while looping to retrieve all fuel_gauge properties.
3. Note that when the battery is fully charged, the charge rate goes to 0 and other properties (like voltage) are no longer returned.
4. Also note that the conditional at line 237 uses the charging rate to skip over the block of code that retrieves other properties. A fully charged battery that is still attached to the charger will have a charge rate of 0 (neither charging nor discharging) but should still report values for voltage, etc.

**Expected behavior**
That a fully charged battery still return valid values for voltage even if the charge/discharge rate is zero.

**Impact**
When a battery is fully charged, all returned properties are zero, which misrepresents the actual state of the battery.

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows) : All
 - Toolchain (e.g Zephyr SDK, ...) : Zephyr SDK 0.16.3
 - Commit SHA or Version used : Zephyr 3.4.0

**Additional context**
I have a pull request I can submit that accommodates the situation where the charge rate is 0 due to a fully-charged battery that is still receiving charging current.

aaronemassey, 2023-10-07:
@martincalsyn 

I think I actually just fixed this last week. I noticed it [while doing a refactor and moved the property fetching out of the branch](https://github.com/zephyrproject-rtos/zephyr/commit/12cbfcf3976dce5220f44a514f7e9b704d7e6a4e). Could you try again on tip of tree or at least with that PR merged?

martincalsyn, 2023-10-07:
Oh! Right you are. Thanks! I'll close this out.

closed 2023-10-07 by martincalsyn