zephyrproject-rtos/zephyr #63727: Blinky and button samples not working for Nucleo F411RE

gabohc, 2023-10-10:
Blinky and button samples not working for Nucleo F411RE

Successfully built the sample, without issues or errors, with:

```
west build -p always -b nucleo_f411re samples/basic/blinky
```
Programming the board directly with the binary file (.hex, .bin) via STM-32 LINK UTILITY does not work.
 (`west flash` does not either, but that's for another issue)

However, using the blinky sample from Mbed IOS, compiling via Keil Studio Cloud and programming with the same method DOES work (ruling out faulty board, checking connection, etc)

**Expected behavior**
Samples working out of the box, given that it is a supported board

**Environment (listing the two that I used):**
 - WSL Ubuntu 22.04    -- Ubuntu 20.04
 - Zephyr SDK 0.16.3     -- 0.13.0
 - Zephyr 3.5.0rc2          -- 2.6.0

**Additional context**
I have successfully used Zephyr previously with a custom board, in another computer. Also tried building the binary with that environment (the second one listed, v2.6.0), with the same result

I also tried modifying the used gpios in the sample, to no avail.

FRASTM, 2023-10-10:
This seems to be a pin conflict on the GPIO PA5

erwango, 2023-10-10:
On my side, this sample works out of the box using default `west flash`.
I'm using Ubuntu 20.04, `v3.5.0-rc2-56`.

Please note that `Programming the board directly with the binary file (.hex, .bin) via STM-32 LINK UTILITY does not work.` is not the usual flashing process, even if I don't see why it wouldn't work.

FRASTM, 2023-10-10:
@gabohc  can you please check your hardware board : SB12 (NRST) is not open (on the bottom side of the MB1136 nucleo_f411re board)

[image]

gabohc, 2023-10-10:
@FRASTM I can confirm, it is **not** open. 

@erwango Yes, I know it's not the standard flashing, but I assumed it would be no problem since in the past I was able to manually flash the other board with the bin file via JTAG with a Raspberry Pi

erwango, 2023-10-11:
@gabohc What's your board revision ?

closed 2024-01-18 by github-actions[bot]

gabohc, 2024-05-12:
Hi apologies for the late reply

I tested this once again, both by repeating the process in Windows and Linux, but same result