zephyrproject-rtos/zephyr #63730: i2s_nrfx_write() race against data_handler() can lead to out-of-order i2s data writes

tmon-nordic, 2023-10-10:
**Describe the bug**
Data written with i2s_nrfx_write() can appear out-of-order (i.e. data passed in later call to i2s_nrfx_write() can be sent before earlier call to i2s_nrfx_write()) on the i2s bus.

**To Reproduce**
Create application that uses I2S in TX only mode with 3 i2s slab buffers. Keep feeding the buffer at varying time. The issue was observed in USB->I2S application where the USB clock was running slightly faster than ACLK which allowed to hit the issue once every 160 seconds (160000 frames).

**Expected behavior**
Data on i2s bus appear in the same order as written to i2s_nrfx_write().

**Impact**
Clearly audible audio glitches with 48 kHz audio where single i2s buffer holds 48 samples for each channel.

**Logs and console output**
```
2s_nrfx_write(const struct device*, void*, unsigned int) 37
[...]
9 978 343  0000560C   LDRB.W         R3, [R0, #228]                      if (drv_data->state == I2S_STATE_RUNNING &&
9 978 344  00005610   CMP            R3, #2
9 978 345  00005612   BEQ            0x0000565C ; <i2s_nrfx_write>+0xAC
9 978 346  0000565C   LDRB.W         R3, [R0, #232]   <----- drv_data->next_tx_buffer_needed reads false
  _isr_wrapper 8 
  nrfx_isr(const void*) 2 
  nrfx_i2s_0_irq_handler() 4 
  irq_handler(struct NRF_I2S_Type*, struct nrfx_i2s_cb_t*) 8 
  nrfy_i2s_events_process(struct nrfy_i2s_xfer_desc_t*, unsigned int, struct NRF_I2S_Type*) 3 
  __nrfy_internal_i2s_events_process(struct NRF_I2S_Type*, unsigned int, const struct nrfy_i2s_xfer_desc_t*) 21 
  irq_handler(struct NRF_I2S_Type*, struct nrfx_i2s_cb_t*) 47 
  data_handler0(const struct nrfy_i2s_buffers_t*, unsigned int) 5 
  data_handler(const struct device*, const struct nrfy_i2s_buffers_t*, unsigned int) 44 
  free_tx_buffer(struct i2s_nrfx_drv_data*, const void*) 3 
  k_mem_slab_free(struct k_mem_slab*, void*) 12 
  z_unpend_first_thread(struct _wait_q_t*) 12 
  z_priq_dumb_best(struct _dnode*) 7 
  z_unpend_first_thread(struct _wait_q_t*) 9 
  k_mem_slab_free(struct k_mem_slab*, void*) 11 
  free_tx_buffer(struct i2s_nrfx_drv_data*, const void*) 1 
  data_handler(const struct device*, const struct nrfy_i2s_buffers_t*, unsigned int) 14 
  get_next_tx_buffer(struct i2s_nrfx_drv_data*, struct nrfy_i2s_buffers_t*) 3 
  k_msgq_get(struct k_timeout_t, void*, struct k_msgq*) 3 
  z_impl_k_msgq_get(struct k_msgq*, void*, struct k_timeout_t) 20 
  get_next_tx_buffer(struct i2s_nrfx_drv_data*, struct nrfy_i2s_buffers_t*) 3 
  data_handler(const struct device*, const struct nrfy_i2s_buffers_t*, unsigned int) 10 
9 978 587  0000582C   CMP            R0, #0
9 978 588  0000582E   BNE            0x0000581A ; <data_handler>+0x15A
9 978 589  00005830   LDRB.W         R3, [R4, #228]                      } else if (drv_data->state == I2S_STATE_STOPPING) {
9 978 590  00005834   CMP            R3, #3
9 978 591  00005836   BNE            0x00005846 ; <data_handler>+0x186
9 978 592  00005846   MOVS           R3, #1          drv_data->next_tx_buffer_needed = true;
9 978 593  00005848   STRB.W         R3, [R4, #232] <----- ISR was one instruction too late
9 978 594  0000584C   B              0x0000577E ; <data_handler>+0xBE    return;
9 978 595  0000577E   ADD            SP, SP, #8
9 978 596  00005780   POP.W          {R4-R8, PC}
  data_handler0(const struct nrfy_i2s_buffers_t*, unsigned int) 1 
  irq_handler(struct NRF_I2S_Type*, struct nrfx_i2s_cb_t*) 3 
  nrfx_i2s_0_irq_handler() 1 
  nrfx_isr(const void*) 1 
  _isr_wrapper 3 
  z_arm_exc_exit 6 
  i2s_nrfx_write(const struct device*, void*, unsigned int) 17 
9 978 612  00005660   CMP            R3, #0 <----- Value fetched before ISR is compared (not current one)
9 978 613  00005662   BEQ            0x00005614 ; <i2s_nrfx_write>+0x64
9 978 614  00005614   ADD.W          R4, R0, #80                         int ret = k_msgq_put(&drv_data->tx_queue,
[...]  <----- 1st TX buffer pointer gets queued
__aeabi_uldivmod 5 
  __udivmoddi4 49 
  __aeabi_uldivmod 4 
  i2s_nrfx_write(const struct device*, void*, unsigned int) 2 
  k_msgq_put(struct k_timeout_t, const void*, struct k_msgq*) 3 
  z_impl_k_msgq_put(struct k_msgq*, const void*, struct k_timeout_t) 16 
  z_unpend_first_thread(struct _wait_q_t*) 12 
  z_priq_dumb_best(struct _dnode*) 7 
  z_unpend_first_thread(struct _wait_q_t*) 9 
  z_impl_k_msgq_put(struct k_msgq*, const void*, struct k_timeout_t) 2 
  __memcpy_ichk(unsigned int, const void* restrict, void* restrict) 4 
  memcpy 22 
  z_impl_k_msgq_put(struct k_msgq*, const void*, struct k_timeout_t) 19 
  i2s_nrfx_write(const struct device*, void*, unsigned int) 6 
  z_impl_i2s_buf_write(const struct device*, void*, unsigned int) 5 
  fill_i2s_buffer() 5 
  ring_buf_get_finish(struct ring_buf*, unsigned int) 27 
  fill_i2s_buffer() 9 
  ring_buf_get_claim(struct ring_buf*, unsigned char**, unsigned int) 26 
  fill_i2s_buffer() 3 
  i2s_buf_write(unsigned int, void*, const struct device*) 3 
  z_impl_i2s_buf_write(const struct device*, void*, unsigned int) 5 
  i2s_config_get(enum i2s_dir, const struct device*) 4 
  i2s_nrfx_config_get(const struct device*, enum i2s_dir) 8 
  z_impl_i2s_buf_write(const struct device*, void*, unsigned int) 10 
  k_mem_slab_alloc(struct k_mem_slab*, void**, struct k_timeout_t) 21 
  z_impl_i2s_buf_write(const struct device*, void*, unsigned int) 6 
  memcpy 774 
  i2s_write(unsigned int, void*, const struct device*) 6 <----- application supplies 2nd TX buffer
  i2s_nrfx_write(const struct device*, void*, unsigned int) 47
[...]
9 979 734  0000560C   LDRB.W         R3, [R0, #228]                      if (drv_data->state == I2S_STATE_RUNNING &&
9 979 735  00005610   CMP            R3, #2
9 979 736  00005612   BEQ            0x0000565C ; <i2s_nrfx_write>+0xAC
9 979 737  0000565C   LDRB.W         R3, [R0, #232]        <----- applications now reads true set in ISR above
9 979 738  00005660   CMP            R3, #0
9 979 739  00005662   BEQ            0x00005614 ; <i2s_nrfx_write>+0x64
9 979 740  00005664   MOVS           R3, #0          nrfx_i2s_buffers_t next = { .p_tx_buffer = mem_block };
9 979 741  00005666   STR            R3, [SP, #8]
9 979 742  00005668   STR            R3, [SP, #12]
9 979 743  0000566A   LDR            R2, [SP, #4]
9 979 744  0000566C   STR            R2, [SP, #12]
9 979 745  0000566E   STRB.W         R3, [R0, #232]  drv_data->next_tx_buffer_needed = false;
9 979 746  00005672   ADD            R1, SP, #8                                    if (!supply_next_buffers(drv_data, &next)) {
9 979 747  00005674   BL             supply_next_buffers ; 0x0000CD28   <----- 2nd buffers gets supplied to I2S, 1st buffer is queued
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.16.1
 - Commit SHA or Version used: 1fc16e65656e742fbe0adc89e0fdf685c51525a5

**Additional context**
`next_tx_buffer_needed` is not properly synchronized between application thread (calling `i2s_nrfx_write()`) and interrupt context. Application thread racer is:
https://github.com/zephyrproject-rtos/zephyr/blob/fc078f9d0239d6ba0cb2ea78ec0ab93fe80e63da/drivers/i2s/i2s_nrfx.c#L604-L624
ISR racer is:
https://github.com/zephyrproject-rtos/zephyr/blob/fc078f9d0239d6ba0cb2ea78ec0ab93fe80e63da/drivers/i2s/i2s_nrfx.c#L365-L369

The race condition is lost when ISR hits after the `if` condition is checked but before the data is put to `tx_queue`.

koffes, 2023-10-10:
Great catch! Roughly how often do these glitches appear?

tmon-nordic, 2023-10-10:
> Great catch! Roughly how often do these glitches appear?

Once every 160 seconds but it all depends on the relative clock ratios. This is quite sensitive to the crystal frequency which varies with temperature and I observed that the colder the nRF5340DK is the longer it takes before the issue hits.

tmon-nordic, 2023-10-10:
The out-of-order i2s data writes is just one of possible outcomes of this race condition. This is the outcome that I observed in my application because I have 3 buffers to basically guarantee that there'll always be at least one buffer available (as long as new data gets provided regularly) regardless of relative clock frequencies. 

In my case it was quite normal to have 2 `i2s_write()` in quick succession. However, if the application does not do subsequent `i2s_write()` calls then if the race was lost then I2S would due to being too late despite having the TX pointer in `tx_queue`. The application would really have plenty of time to supply the pointer (virtually the whole time the previous buffer took to consume) because when the race condition hit, the previous buffer has just started to go on the bus. In other words, the other outcome is premature i2s error state. If the application was just a tiny bit faster **or slower** (so it wouldn't lose the race) the error could be avoided.

PR #63790 (2023-10-11): drivers: i2s_nrfx: Fix write race condition
There is inherent race condition between i2s_nrfx_write() and I2S interrupt handler because I2S operates independently from the rest of the system. If software takes too long to supply next TX pointer then nRF I2S peripheral will simply resupply the previous buffer.

The race window is rather short. The failed race executes as follows:
  1. i2s_nrfx_write() checks state and loads next_tx_buffer_needed
  2. I2S interrupt handler executes and calls data_handler() which notices empty TX queue and therefore sets next_tx_buffer_needed
  3. i2s_nrfx_write( [... cut]
closed 2023-10-11 by carlescufi