zephyrproject-rtos/zephyr #63732: ESP32 example for deep_sleep doesn't build

LubomirBogdanov, 2023-10-10:
**Describe the bug**
The deep_sleep example in zephyrproject/zephyr/samples/boards/esp32/deep_sleep fails with "fatal error: zephyr/sys/poweroff.h: No such file or directory".

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? - ESP32 esp32c3_devkitm, but it also fails with esp32_devkitc_wroom.
 - What have you tried to diagnose or workaround this issue? - I have enabled "System Power Management", "CPU power management drivers", "Device Power Management", "Power domain", "Runtime device power management" and "Power domain drivers" in the menuconfig.

**To Reproduce**
Steps to reproduce the behavior:
1. cd ./zephyrproject/zephyr/
2. west build -b esp32_devkitc_wroom samples/boards/esp32/deep_sleep -t guiconfig
3. west build -b esp32_devkitc_wroom samples/boards/esp32/deep_sleep 

**Expected behavior**
The build should pass but instead it fails with "zephyrproject/zephyr/samples/boards/esp32/deep_sleep/src/main.c:8:10: fatal error: zephyr/sys/poweroff.h: No such file or directory"

**Impact**
Showstopper - we cannot build our device without power management, since it is a battery powered one.

**Environment (please complete the following information):**
 - Ubuntu 22.04 LTS 64-bit
 - Zephyr SDK
 -commit 7bfe0967879e380bffb850f5d02bc9119791da4a (HEAD -> main, origin/main, origin/HEAD)
Author: Kevin Wang <kevinwang821020@google.com>
Date:   Thu Jun 29 15:47:18 2023 +0800

**Additional context**
Maybe I'm missing some step during the build process?

sylvioalves, 2023-10-11:
@LubomirBogdanov I can't reproduce this issue anymore. Would you update your main and run `west update` so we can confirm this?

LubomirBogdanov, 2023-10-11:
I updated it, no change. I forgot to mention that I issued the command:

west blobs fetch hal_espressif

to install the ESP32's HAL.

LubomirBogdanov, 2023-10-11:
# The west manifest file for upstream Zephyr.
#
# The per-installation west configuration file, .west/config, sets the
# path to the project containing this file in the [manifest] section's
# "path" variable.
#
# You are free to create your own manifest files and put them in any
# repository you want, to create your own custom Zephyr installations.
# For example, you could create a manifest file in your own out of
# tree application directory, which would pull this zephyr repository
# in as an ordinary project.
#
# You can pass your manifest repositories to west init when creating a
# new Zephyr installation. See the west documentation for more
# information.

manifest:
  defaults:
    remote: upstream

  remotes:
    - name: upstream
      url-base: https://github.com/zephyrproject-rtos
    - name: babblesim
      url-base: https://github.com/BabbleSim

  group-filter: [-babblesim]

  #
  # Please add items below based on alphabetical order
  projects:
    - name: acpica
      revision: f16a0b4d0f0edd7b78a332fcf507be2187fac21e
      path: modules/lib/acpica
    - name: bsim
      repo-path: babblesim-manifest
      revision: 384a091445c57b44ac8cbd18ebd245b47c71db94
      path: tools/bsim
      groups:
        - babblesim
    - name: babblesim_base
      remote: babblesim
      repo-path: base.git
      path: tools/bsim/components
      revision: 19d62424c0802c6c9fc15528febe666e40f372a1
      groups:
        - babblesim
    - name: babblesim_ext_2G4_libPhyComv1
      remote: babblesim
      repo-path: ext_2G4_libPhyComv1.git
      path: tools/bsim/components/ext_2G4_libPhyComv1
      revision: 9018113a362fa6c9e8f4b9cab9e5a8f12cc46b94
      groups:
        - babblesim
    - name: babblesim_ext_2G4_phy_v1
      remote: babblesim
      repo-path: ext_2G4_phy_v1.git
      path: tools/bsim/components/ext_2G4_phy_v1
      revision: d47c6dd90035b41b14f6921785ccb7b8484868e2
      groups:
        - babblesim
    - name: babblesim_ext_2G4_channel_NtNcable
      remote: babblesim
      repo-path: ext_2G4_channel_NtNcable.git
      path: tools/bsim/components/ext_2G4_channel_NtNcable
      revision: 20a38c997f507b0aa53817aab3d73a462fff7af1
      groups:
        - babblesim
    - name: babblesim_ext_2G4_channel_multiatt
      remote: babblesim
[... 264 lines cut ...]
      path: modules/tee/tf-m/tf-m-tests
      groups:
        - tee
    - name: psa-arch-tests
      revision: 6a17330e0dfb5f319730f974d5b05f7b7f04757b
      path: modules/tee/tf-m/psa-arch-tests
      groups:
        - tee
    - name: uoscore-uedhoc
      revision: e8920192b66db4f909eb9cd3f155d5245c1ae825
      path: modules/lib/uoscore-uedhoc
    - name: zcbor
      revision: 67fd8bb88d3136738661fa8bb5f9989103f4599e
      path: modules/lib/zcbor
    - name: zscilib
      path: modules/lib/zscilib
      revision: 34c3432e81085bb717e4871d21ca419ae0058ec5
    - name: thrift
      path: modules/lib/thrift
      revision: 10023645a0e6cb7ce23fcd7fd3dbac9f18df6234

  self:
    path: zephyr
    west-commands: scripts/west-commands.yml
    import: submanifests

sylvioalves, 2023-10-11:
@LubomirBogdanov, the above confirms that you are currently on a branch which does not contain the patch to the issue you reported. I suggest updating zephyr main to the latest and then running `west update` to bring latest `hal_espressif` fixes. That will work.

closed 2023-10-11 by sylvioalves