zephyrproject-rtos/zephyr #63736: Avoid lis2dh interrupt multiple triggers

gwa-simon, 2023-10-10:
### Problem description

I am using the accelerometer lis3dh in a project that requires very low power consumption and I am having to high power when implementing anymotion interrupt. I notice that the driver [drivers/sensor/lis2dh](url) configures the interrupt pin as GPIO_INT_LEVEL_ACTIVE and turn it OFF while processing to avoid multiple triggers. Although there is a situation that multiple triggers still happen.

Lis3dh (or lis2dh) does not clear the interrupt immediately after INTx_SRC is read, but only at the next sample read. So lets say we implement a 50Hz sampling rate, it means readings every 20ms. 
The callback `lis2dh_thread_cb` can be executed much faster than 20 ms and it creates a loop that runs for 20 ms every time that the interrupt is triggered by the accelerometer (When lis2dh_thread_cb is finished the pin will still be high and a new call will be generated). 

Since the register INTx_SRC is read in the first loop, it will be cleared and `handler_anymotion` will be called only once. But the loop will still be running and consuming power, as can be seen in the picture below.

[image]

### Proposed change

I propose that the type of interrupt is changed from **GPIO_INT_LEVEL_ACTIVE** to **GPIO_INT_EDGE_RISING**, which will ensure that it is triggered only once. 

Since the interrupt is changed to EDGE type, the multiple calls for `setup_int1` and `setup_int2` can be removed and the interrupt can be initialized only once.

gwa-simon, 2023-10-10:
Actually, I think it would be better to be able to configure which type of interrupt is generated (GPIO_INT_EDGE, GPIO_INT_EDGE_RISING, GPIO_INT_EDGE_FALLING, GPIO_INT_LEVEL_ACTIVE, ...)

Turns out that for the model micro I am using (nrf52832), some edge interrupts cause a static 400 µA current. So being able to select the type might be interesting since different applications might require different setups.

Best regards

avisconti, 2023-10-12:
> Actually, I think it would be better to be able to configure which type of interrupt is generated (GPIO_INT_EDGE, GPIO_INT_EDGE_RISING, GPIO_INT_EDGE_FALLING, GPIO_INT_LEVEL_ACTIVE, ...)
>
> Turns out [...]

I'm investigating feasibility of adding this configuration in DT. But there are no other examples

avisconti, 2023-10-12:
@gwa-simon can you please take a look and test #63876? I just tested compilation, but I'd appreciate if you could help me testing it out... Thanks

gwa-simon, 2023-10-16:
Hi @avisconti, 

thank you for implementing the changes. It looks to work fine, but could you please add one extra type, GPIO_INT_EDGE. 

For some reason with the micro NRF52832, if I configure it directly with GPIO_INT_EDGE_RISING or GPIO_INT_EDGE_FALLING, there is a constant current of 400uA. But with GPIO_INT_EDGE it works fine.

avisconti, 2023-10-16:
> Hi @avisconti,
>
> thank you for implementing the changes. It looks to work fine, but could you please add one extra type, GPIO_INT_EDGE.
>
> For some reason with the micro NRF52832, if I configure it [...]

mmmh, sure. But it is strange.

avisconti, 2023-10-17:
> Hi @avisconti,
>
> thank you for implementing the changes. It looks to work fine, but could you please add one extra type, GPIO_INT_EDGE.
>
> For some reason with the micro NRF52832, if I configure it [...]

Maybe you can add your comments/thoughts directly on #63876. Thanks

closed 2023-11-06 by fabiobaltieri

avisconti, 2024-05-09:
@gwa-simon can you please take a look to what have been requested in #69487? I don't want that something would be broken

avisconti, 2024-05-17:
#71229 may potentially broke this issue. But the PR makes sense and should be approved. I kindly ask @gwa-simon to test his case against that PR once merged and re-open this issue if the case.

PR #63876 (2023-11-06): drivers/sensor: lis2dh: add INT1/INT2 gpio interrupt config
Add INT1/INT2 gpio interrupt configuration.
iPossible values:

      0 = GPIO_INT_EDGE_RISING
      1 = GPIO_INT_EDGE_FALLING
      2 = GPIO_INT_LEVEL_HIGH
      3 = GPIO_INT_LEVEL_LOW

Fixes #63736

EDIT:
Added also GPIO_INT_EDGE
commit 01dc3c48 (2023-11-06): drivers/sensor: lis2dh: add INT1/INT2 gpio interrupt config in DT

Add INT1/INT2 gpio interrupt configuration at Device Tree level.

Two new properties:

  - int1-gpio-config
  - int2-gpio-config

Possible values:

  0 = GPIO_INT_EDGE
  1 = GPIO_INT_EDGE_RISING
  2 = GPIO_INT_EDGE_FALLING
  3 = GPIO_INT_LEVEL_HIGH
  4 = GPIO_INT_LEVEL_LOW

Fixes #63736