zephyrproject-rtos/zephyr #63765: nxp: twister: does not detect NXP boards with LPC-Link

butok, 2023-10-10:
**Describe the bug**
Twister does not detect NXP boards with LPC-Link.

**To Reproduce**
Steps to reproduce the behavior:
1. Attach LPCXpresso55S69 (or any other LPCXpresso55Sxx board) with CMSIS-DAP to your PC.
2. Run in command line `python .\scripts\twister --generate-hardware-map map.yml` as described [here](https://docs.zephyrproject.org/latest/develop/test/twister.html#running-tests-on-hardware)
3. In a command line terminal, you can sees that the NXP device is marked as unsupported: 
```
C:\_ddm\ZEPHYR\zephyr_vs\zephyr>python .\scripts\twister --generate-hardware-map map.yml
ZEPHYR_BASE unset, using "C:\_ddm\ZEPHYR\zephyr_vs\zephyr"
Renaming output directory to C:\_ddm\ZEPHYR\zephyr_vs\zephyr\twister-out.5
INFO    - Using Ninja..
INFO    - Zephyr version: v3.5.0-rc2-8-gb122685eff5b
INFO    - Using 'zephyr' toolchain.
INFO    - Scanning connected hardware...
WARNING - Unsupported device (NXP): COM17 - LPC-LinkII UCom Port (COM17)
INFO    - Detected devices:

| Platform   | ID   | Serial device   |
|------------|------|-----------------|
```

**Expected behavior**
twister should detect NXP boards,

**Impact**
Cannot use twister to run tests on NXP CMSIS-DAP boards

**Environment (please complete the following information):**
OS: Windows
Toolchain: Zephyr SDK 0.16.1
Commit SHA: https://github.com/zephyrproject-rtos/zephyr/commit/b122685eff5b4f22a88359402a8b07b36c381d0c

henrikbrixandersen, 2023-10-11:
Twister currently supports a limited number of programmers for mapping hardware (see https://github.com/zephyrproject-rtos/zephyr/blob/1c2d326579ee8cea4c809eb737585537e3482677/scripts/pylib/twister/twisterlib/hardwaremap.py#L128). New ones can be added there, but I am wondering if it would be a better approach to delegate this "auto-detected" to the various West runners?

henrikbrixandersen, 2023-10-11:
That being said, I just tried this on my LPCxpresso55S16 under Linux. The board is detected (but as always, you'll need to manually edit the `map.yml` file and tell twister the board name): 

```
$ twister --generate-hardware-map map.yml --persistent-hardware-map
INFO    - Using Ninja..
INFO    - Zephyr version: v3.5.0-rc2-74-ge7bd10ae719d
INFO    - Using 'zephyr' toolchain.
INFO    - Scanning connected hardware...
WARNING - Unsupported device (None): /dev/ttyS0 - ttyS0
INFO    - Registered devices:

| Platform   | ID       | Serial device                                                                     |
|------------|----------|-----------------------------------------------------------------------------------|
| unknown    | PRARBQBQ | /dev/serial/by-id/usb-NXP_Semiconductors_LPC-LINK2_CMSIS-DAP_V5.361_PRARBQBQ-if01 |

```

butok, 2023-10-11:
> Twister currently supports a limited number of programmers for mapping hardware (see

Based on the code it should support PyOCD CMSIS-DAP. Right?
So the required runner/programmer is supported. 
So I have enabled PyOCD for LPC55S16 in its board.cmake and run the twister:
```
>python .\scripts\twister --generate-hardware-map map.yml
INFO    - Using Ninja..
INFO    - Zephyr version: v3.5.0-rc2-8-gb122685eff5b
INFO    - Using 'zephyr' toolchain.
INFO    - Scanning connected hardware...
WARNING - Unsupported device (NXP): COM15 - LPC-LinkII UCom Port (COM15)
INFO    - Detected devices:

| Platform   | ID   | Serial device   |
|------------|------|-----------------|
```
The generated .map file is empty.

My setup 
- Windows
- LPC55S16 with the default CMSIS-DAP Link2
- enabled pyocd for LPC55S16
The same behavior for all LPC55Sxx boards, the boards are not detected.

So something is broken with twister for Windows & pyocd & CMSIS-DAP.

henrikbrixandersen, 2023-10-11:
> Based on the code it should support PyOCD CMSIS-DAP. Right?

Right. It works for me on Linux.

butok, 2023-10-11:
> Right. It works for me on Linux.

We can take it as the twister issue report for Windows.

PR #66491 (2023-12-19): twister: fix hardware map hardware detection
NXP boards with CMSID-DAP are not detected by twister --generate-hardware-map, because serial device name 'mbed' is compared with upper case 'MBED' in a list of supported manufacturers. Fix it by making the comparison case-insensitive.
Tested using mimxrt1020_evk.

Fixes #63765
commit 096e9589 (2023-12-19): twister: fix hardware map hardware detection

NXP boards with CMSID-DAP are not detected by twister
--generate-hardware-map, because serial device name 'mbed' is compared
with upper case 'MBED' in a list of supported manufacturers.
Fix it by making the comparison case-insensitive.
Tested using mimxrt1020_evk.

Fixes #63765
closed 2023-12-19 by carlescufi