zephyrproject-rtos/zephyr #63770: drivers: flash: stm32-ospi-nor: LittleFS using Dual/Quad mode not working

nandojve, 2023-10-10:
**Describe the bug**
The `stm32-ospi-nor` does not work. It can be verified using the [subsys/fs/littlefs](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/subsys/fs/littlefs) sample.

**To Reproduce**
Steps to reproduce the behavior:

1. build/flash mentioned sample and check that it works

```
west build -b b_u585i_iot02a samples/subsys/fs/littlefs
west flash
```
```
I: OSPI flash config is OPI / DTR
I: Read SFDP from octoFlash
I: Read SFDP from octoFlash
I: littlefs partition at /lfs1
I: LittleFS version 2.5, disk version 2.0
I: FS at ospi-nor-flash@0:0x0 is 16384 0x1000-byte blocks with 512 cycle
I: sizes: rd 256 ; pr 256 ; ca 4096 ; la 256
I: Automount /lfs1 succeeded
*** Booting Zephyr OS build v3.5.0-rc2-92-gc5064cf70835 ***
Sample program to r/w files on littlefs
Area 0 at 0x0 on ospi-nor-flash@0 for 67108864 bytes
/lfs1 automounted
/lfs1: bsize = 256 ; frsize = 4096 ; blocks = 16384 ; bfree = 16381

Listing dir /lfs1 ...
[FILE] boot_count (size = 1)
[FILE] pattern.bin (size = 547)
/lfs1/boot_count read count:6 (bytes: 1)
/lfs1/boot_count write new boot count 7: [wr:1]
------ FILE: /lfs1/pattern.bin ------
07 aa aa aa aa aa aa aa 08 aa aa aa aa aa aa aa
09 aa aa aa aa aa aa aa 0a aa aa aa aa aa aa aa
0b aa aa aa aa aa aa aa 0c aa aa aa aa aa aa aa
0d aa aa aa aa aa aa aa 0e aa aa aa aa aa aa aa
0f aa aa aa aa aa aa aa 10 aa aa aa aa aa aa aa
11 aa aa aa aa aa aa aa 12 aa aa aa aa aa aa aa
13 aa aa aa aa aa aa aa 14 aa aa aa aa aa aa aa
15 aa aa aa aa aa aa aa 16 aa aa aa aa aa aa aa
17 aa aa aa aa aa aa aa 18 aa aa aa aa aa aa aa
19 aa aa aa aa aa aa aa 1a aa aa aa aa aa aa aa
1b aa aa aa aa aa aa aa 1c aa aa aa aa aa aa aa
1d aa aa aa aa aa aa aa 1e aa aa aa aa aa aa aa
1f aa aa aa aa aa aa aa 20 aa aa aa aa aa aa aa
21 aa aa aa aa aa aa aa 22 aa aa aa aa aa aa aa
23 aa aa aa aa aa aa aa 24 aa aa aa aa aa aa aa
25 aa aa aa aa aa aa aa 26 aa aa aa aa aa aa aa
27 aa aa aa aa aa aa aa 28 aa aa aa aa aa aa aa
29 aa aa aa aa aa aa aa 2a aa aa aa aa aa aa aa
2b aa aa aa aa aa aa aa 2c aa aa aa aa aa aa aa
2d aa aa aa aa aa aa aa 2e aa aa aa aa aa aa aa
2f aa aa aa aa aa aa aa 30 aa aa aa aa aa aa aa
31 aa aa aa aa aa aa aa 32 aa aa aa aa aa aa aa
33 aa aa aa aa aa aa aa 34 aa aa aa aa aa aa aa
35 aa aa aa aa aa aa aa 36 aa aa aa aa aa aa aa
37 aa aa aa aa aa aa aa 38 aa aa aa aa aa aa aa
39 aa aa aa aa aa aa aa 3a aa aa aa aa aa aa aa
3b aa aa aa aa aa aa aa 3c aa aa aa aa aa aa aa
3d aa aa aa aa aa aa aa 3e aa aa aa aa aa aa aa
3f aa aa aa aa aa aa aa 40 aa aa aa aa aa aa aa
41 aa aa aa aa aa aa aa 42 aa aa aa aa aa aa aa
43 aa aa aa aa aa aa aa 44 aa aa aa aa aa aa aa
45 aa aa aa aa aa aa aa 46 aa aa aa aa aa aa aa

47 aa aa aa aa aa aa aa 48 aa aa aa aa aa aa aa
49 aa aa aa aa aa aa aa 4a aa aa aa aa aa aa aa
4b aa b0 
I: /lfs1 unmounted
/lfs1 unmount: 0
```

2. change board devicetree to SPI mode and check that it works

```diff
diff --git a/boards/arm/b_u585i_iot02a/b_u585i_iot02a-common.dtsi b/boards/arm/b_u585i_iot02a/b_u585i_iot02a-common.dtsi
index b9b07348f3..8a02276d9d 100644
--- a/boards/arm/b_u585i_iot02a/b_u585i_iot02a-common.dtsi
+++ b/boards/arm/b_u585i_iot02a/b_u585i_iot02a-common.dtsi
@@ -136,8 +136,8 @@
                reg = <0>;
                ospi-max-frequency = <DT_FREQ_M(50)>;
                size = <DT_SIZE_M(512)>; /* 64 MBytes */
-               spi-bus-width = <OSPI_OPI_MODE>;
-               data-rate = <OSPI_DTR_TRANSFER>;
+               spi-bus-width = <OSPI_SPI_MODE>;
+               data-rate = <OSPI_STR_TRANSFER>;
                four-byte-opcodes;
                status = "okay";
```
```
I: OSPI flash config is SPI|DUAL|QUAD / STR
I: Read SFDP from octoFlash
I: Read SFDP from octoFlash
I: littlefs partition at /lfs1
I: LittleFS version 2.5, disk version 2.0
I: FS at ospi-nor-flash@0:0x0 is 16384 0x1000-byte blocks with 512 cycle
I: sizes: rd 256 ; pr 256 ; ca 4096 ; la 256
I: Automount /lfs1 succeeded
*** Booting Zephyr OS build v3.5.0-rc2-92-gc5064cf70835 ***
Sample program to r/w files on littlefs
Area 0 at 0x0 on ospi-nor-flash@0 for 67108864 bytes
/lfs1 automounted
/lfs1: bsize = 256 ; frsize = 4096 ; blocks = 16384 ; bfree = 16381

Listing dir /lfs1 ...
[FILE] boot_count (size = 1)
[FILE] pattern.bin (size = 547)
/lfs1/boot_count read count:5 (bytes: 1)
/lfs1/boot_count write new boot count 6: [wr:1]
------ FILE: /lfs1/pattern.bin ------
06 55 55 55 55 55 55 55 07 55 55 55 55 55 55 55
08 55 55 55 55 55 55 55 09 55 55 55 55 55 55 55
0a 55 55 55 55 55 55 55 0b 55 55 55 55 55 55 55
0c 55 55 55 55 55 55 55 0d 55 55 55 55 55 55 55
0e 55 55 55 55 55 55 55 0f 55 55 55 55 55 55 55
10 55 55 55 55 55 55 55 11 55 55 55 55 55 55 55
12 55 55 55 55 55 55 55 13 55 55 55 55 55 55 55
14 55 55 55 55 55 55 55 15 55 55 55 55 55 55 55
16 55 55 55 55 55 55 55 17 55 55 55 55 55 55 55
18 55 55 55 55 55 55 55 19 55 55 55 55 55 55 55
1a 55 55 55 55 55 55 55 1b 55 55 55 55 55 55 55
1c 55 55 55 55 55 55 55 1d 55 55 55 55 55 55 55
1e 55 55 55 55 55 55 55 1f 55 55 55 55 55 55 55
20 55 55 55 55 55 55 55 21 55 55 55 55 55 55 55
22 55 55 55 55 55 55 55 23 55 55 55 55 55 55 55
24 55 55 55 55 55 55 55 25 55 55 55 55 55 55 55
26 55 55 55 55 55 55 55 27 55 55 55 55 55 55 55
28 55 55 55 55 55 55 55 29 55 55 55 55 55 55 55
2a 55 55 55 55 55 55 55 2b 55 55 55 55 55 55 55
2c 55 55 55 55 55 55 55 2d 55 55 55 55 55 55 55
2e 55 55 55 55 55 55 55 2f 55 55 55 55 55 55 55
30 55 55 55 55 55 55 55 31 55 55 55 55 55 55 55
32 55 55 55 55 55 55 55 33 55 55 55 55 55 55 55
34 55 55 55 55 55 55 55 35 55 55 55 55 55 55 55
36 55 55 55 55 55 55 55 37 55 55 55 55 55 55 55
38 55 55 55 55 55 55 55 39 55 55 55 55 55 55 55
3a 55 55 55 55 55 55 55 3b 55 55 55 55 55 55 55
3c 55 55 55 55 55 55 55 3d 55 55 55 55 55 55 55
3e 55 55 55 55 55 55 55 3f 55 55 55 55 55 55 55
40 55 55 55 55 55 55 55 41 55 55 55 55 55 55 55
42 55 55 55 55 55 55 55 43 55 55 55 55 55 55 55
44 55 55 55 55 55 55 55 45 55 55 55 55 55 55 55

46 55 55 55 55 55 55 55 47 55 55 55 55 55 55 55
48 55 55 55 55 55 55 55 49 55 55 55 55 55 55 55
4a 55 af 
I: /lfs1 unmounted
/lfs1 unmount: 0

```

3. Now change to DUAL or QUAD mode and check that application does not works.

```diff
-               spi-bus-width = <OSPI_OPI_MODE>;
-               data-rate = <OSPI_DTR_TRANSFER>;
+               spi-bus-width = <OSPI_DUAL_MODE>;
# OR
+               spi-bus-width = <OSPI_QUAD_MODE>;
+               data-rate = <OSPI_STR_TRANSFER>;
```
```
I: OSPI flash config is SPI|DUAL|QUAD / STR
I: Read SFDP from octoFlash
I: Read SFDP from octoFlash
I: littlefs partition at /lfs1
I: LittleFS version 2.5, disk version 2.0
I: FS at ospi-nor-flash@0:0x0 is 16384 0x1000-byte blocks with 512 cycle
I: sizes: rd 256 ; pr 256 ; ca 4096 ; la 256
E: WEST_TOPDIR/modules/fs/littlefs/lfs.c:1234: Corrupted dir pair at {0x0, 0x1}
W: can't mount (LFS -84); formatting
W: WEST_TOPDIR/modules/fs/littlefs/lfs.c:1897: Superblock 0x0 has become unwritable
E: format failed (LFS -28)
E: fs mount error (-28)
E: Automount /lfs1 failed: -28
*** Booting Zephyr OS build v3.5.0-rc2-92-gc5064cf70835 ***
Sample program to r/w files on littlefs
Area 0 at 0x0 on ospi-nor-flash@0 for 67108864 bytes
/lfs1 automounted
E: mount point not found!!
FAIL: statvfs: -2
E: fs not mounted (mp == 0x20000034)
/lfs1 unmount: -22
```

**Expected behavior**
The `stm32-ospi-nor` must work in dual and quad mode.

**Impact**
This is critical because there are devices that only have one OCTOSPI with 4 lines and this slow down considerable the application speed/performance.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.16.2
 - Zephyr v3.5.0-rc2

FRASTM, 2023-10-11:
According to the b_u585i_iot02a board schematic [MB1551](https://www.st.com/content/ccc/resource/technical/layouts_and_diagrams/schematic_pack/group2/52/8f/38/da/dd/0f/40/56/mb1551-u585ai1-d01-schematic/files/mb1551-u585ai1-d01-schematic.pdf/jcr:content/translations/en.mb1551-u585ai1-d01-schematic.pdf), the octo flash is the [MX25LM51245GXDI00](https://www.mxic.com.tw/Lists/Datasheet/Attachments/8729/MX25LM51245G,%203V,%20512Mb,%20v1.1.pdf). 
I wonder if this serial NOR from Macronix  really supports DUAL and QUAD SPI modes:

Key Features
- • Protocol Support - Single I/O and Octa I/O
- • Support DTR (Double Transfer Rate) Mode
- • Support clock frequency up to 133MHz_

nandojve, 2023-10-12:
Hi @FRASTM ,

You are right about [MX25LM51245GXDI00](https://www.mxic.com.tw/Lists/Datasheet/Attachments/8729/MX25LM51245G,%203V,%20512Mb,%20v1.1.pdf) I tested these modes using the Kit but didn't checked if MX25 memory supports the Dual/Quad mode. I assumed that on the development board you could choose between then. 

Besides that, I first detected the issue using [W25Q01JVZEIQ](https://www.winbond.com/hq/product/code-storage-flash-memory/serial-nor-flash/?__locale=en&partNo=W25Q01JV).
  * SPI with Dual/Quad mode.
  * Support clock frequency up to 133MHz

I could validate this memory using SPI mode at 100MHz SDR (Single Data Rate) with LittleFS but when enabling the Dual/Quad mode it didn't worked. This is why I tested on the Kit and saw the described behaviour.

I was wondering if you can guide me to help diagnose the possible issue.

GeorgeCGV, 2023-10-12:
@nandojve many winbond flashes don't support writeoc `PP_1_4_4` (`0x38`) that is used by default for STR QUAD. Please set `writeoc = "PP_1_1_4"` in the device tree and try again.

nandojve, 2023-10-12:
Hi @GeorgeCGV ,

Thank you so much for the tips. I'll try and provide feedback.

closed 2023-10-12 by nandojve