zephyrproject-rtos/zephyr #63791: "Just works" Level 2 security

Pmr-precure, 2023-10-11:
Hi, I am not even sure its the correct place to ask this questions, but ill try. First of all, I don't have a lot of knowledge about firmware. We lost our firmware developer before he was done implementing security for our BLE devices. Everything is working as it should. But we miss an encrypted connection (level 2 security) to our GATT services. And our device doesn't have a display, so I've tried to implement a "Just works" pairing process. I've looked at many different samples and I've ended up with an implementation that almost works as intended. But I have 1 issue I can't seem to solve (and it seems to be only on android).

If I connect(first time), I get asked to pair. And this goes smooth. All the GATT services are running. No problem! But if I disconnect, and then reconnect, then it really struggles to setup the GATT connection again. Eventually it succeeds, sometimes it takes 2-3 reconnects attempts. Sometimes it takes between 5-10 mins before it succeeds.

This is the error from our mobile app. "Device '00000000-0000-0000-0000-fd0debeda8ca' disconnected while writing descriptor with 00002902-0000-1000-8000-00805f9b34fb"

I'm not using a devkit (I flash via Bluetooth) so I cant really show any logs. But here are some code samples of the implementation and config.

using nRF52840.

Best regards.

Pmr-precure, 2023-10-12:
I tried to remove the bt_conn_set_security() and decorate my gatt services like this:

BT_GATT_SERVICE_DEFINE(ble_data_service,
					   BT_GATT_PRIMARY_SERVICE(&ble_data_service_uuid),
					   BT_GATT_CHARACTERISTIC(&ble_data_char_uuid.uuid, BT_GATT_CHRC_READ | BT_GATT_CHRC_NOTIFY, BT_GATT_PERM_READ_ENCRYPT, ble_data_char_read, NULL, &ads1298_adv_data),
					   BT_GATT_CCC(ccc_cfg_changed, BT_GATT_PERM_WRITE_ENCRYPT), );

This also prompts the user for a pairing the first time and successfully start reading data, but after a reconnect it keeps getting disconnected, and after a lot of tries it succeeds.

Pmr-precure, 2023-10-12:
Setting them all the just READ fixes the reconnect issue, but then of cause its not encrypted.

theob-pro, 2023-10-12:
@Pmr-precure I believe the nordicsemi [devzone](https://devzone.nordicsemi.com/) will be better able to help your case.
It would help if you could gather some Bluetooth logs. You can use the nordicsemi Bluetooth sniffer if you don't have access to other HW: https://www.nordicsemi.com/Products/Development-tools/nrf-sniffer-for-bluetooth-le

Pmr-precure, 2023-10-12:
> @Pmr-precure I believe the nordicsemi [devzone](https://devzone.nordicsemi.com/) will be better able to help your case. It would help if you could gather some Bluetooth logs. You can use the [...]

I did make a post on devzone as well :D I'm kind of desperate..

XavierChapron, 2023-10-12:
Hello, did you try calling bt_conn_set_security() from a delayed kwork as proposed [here](https://github.com/zephyrproject-rtos/zephyr/issues/36600#issuecomment-1757260622)
I'm pretty sure that's a way to fix this issue.

Pmr-precure, 2023-10-12:
> Hello, did you try calling bt_conn_set_security() from a delayed kwork as proposed [here](https://github.com/zephyrproject-rtos/zephyr/issues/36600#issuecomment-1757260622) I'm pretty sure that's a way to fix this issue.

Hi man, i already tried that. right now im trying to get a hold on a dev kit so i can try SNIFF

Pmr-precure, 2023-10-12:
Problem solved, i forgot BT_GATT_CCC(ccc_cfg_changed, BT_GATT_PERM_READ_ENCRYPT | BT_GATT_PERM_WRITE_ENCRYPT ), ); I only had the READ part, so when i reconnected, it apparently disconnected me..

This can be closed.

closed 2023-10-12 by Pmr-precure