zephyrproject-rtos/zephyr #63804: west flash invocation in nRF Connect SDK in VS Code fails with KeyError: 'NRF52'

descartes, 2023-10-11:
**Describe the bug**
Flashing to u-Blox NINA-B1 dev board fails using the nRF Connect Visual Studio Code setup

**To Reproduce**
Click Flash in the nRFConnect VS Code extension

**Expected behavior**
Flash the board

**Impact**
Showstopper-ish for co-develop clients, I can construct the required command line to circumvent and have a hack to board.cmake.

**Logs and console output**
```
 *  Executing task: nRF Connect: Build: blinky/build (active) 

Building blinky
/bin/sh -c west build --build-dir /Users/nick/Documents/nRF/blinky/build /Users/nick/Documents/nRF/blinky

ninja: no work to do.
 *  Terminal will be reused by tasks, press any key to close it. 

 *  Executing task: nRF Connect: Flash: blinky/build (active) 

Flashing build to 483134503
/bin/sh -c west flash -d /Users/nick/Documents/nRF/blinky/build --skip-rebuild --dev-id 483134503

-- west flash: using runner nrfjprog
-- runners.nrfjprog: Flashing file: /Users/nick/Documents/nRF/blinky/build/zephyr/zephyr.hex
Traceback (most recent call last):
  File "/opt/nordic/ncs/toolchains/4ef6631da0/bin/west", line 8, in <module>
    sys.exit(main())
  File "/opt/nordic/ncs/toolchains/4ef6631da0/lib/python3.9/site-packages/west/app/main.py", line 866, in main
    app.run(argv or sys.argv[1:])
  File "/opt/nordic/ncs/toolchains/4ef6631da0/lib/python3.9/site-packages/west/app/main.py", line 111, in run
    self.run_command(argv)
  File "/opt/nordic/ncs/toolchains/4ef6631da0/lib/python3.9/site-packages/west/app/main.py", line 385, in run_command
    self.run_extension(args.command, argv)
  File "/opt/nordic/ncs/toolchains/4ef6631da0/lib/python3.9/site-packages/west/app/main.py", line 467, in run_extension
    self.cmd.run(args, unknown, self.topdir, manifest=self.manifest,
  File "/opt/nordic/ncs/toolchains/4ef6631da0/lib/python3.9/site-packages/west/commands.py", line 194, in run
    self.do_run(args, unknown)
  File "/opt/nordic/ncs/v2.4.1/zephyr/scripts/west_commands/flash.py", line 32, in do_run
    do_run_common(self, my_args, runner_args, domains=domains)
  File "/opt/nordic/ncs/v2.4.1/zephyr/scripts/west_commands/run_common.py", line 174, in do_run_common
    do_run_common_image(command, user_args, user_runner_args, d.build_dir)
  File "/opt/nordic/ncs/v2.4.1/zephyr/scripts/west_commands/run_common.py", line 250, in do_run_common_image
    runner.run(command_name)
  File "/opt/nordic/ncs/v2.4.1/zephyr/scripts/west_commands/runners/core.py", line 611, in run
    self.do_run(command, **kwargs)
  File "/opt/nordic/ncs/v2.4.1/zephyr/scripts/west_commands/runners/nrf_common.py", line 398, in do_run
    self.program_hex()
  File "/opt/nordic/ncs/v2.4.1/zephyr/scripts/west_commands/runners/nrf_common.py", line 260, in program_hex
    self.flush(force=False)
  File "/opt/nordic/ncs/v2.4.1/zephyr/scripts/west_commands/runners/nrf_common.py", line 188, in flush
    self.flush_ops(force=force)
  File "/opt/nordic/ncs/v2.4.1/zephyr/scripts/west_commands/runners/nrf_common.py", line 375, in flush_ops
    self.do_exec_op(self.ops.popleft(), force)
  File "/opt/nordic/ncs/v2.4.1/zephyr/scripts/west_commands/runners/nrfjprog.py", line 91, in do_exec_op
    self.check_call(cmd + ['-f', families[self.family]] + core_opt +
KeyError: 'NRF52'

 *  The terminal process terminated with exit code: 1. 
```

**Environment (please complete the following information):**
 - macOS 12.7 on Intel
 - Visual Studio Code 1.83.0
 - nRF Connect SDK 2.4.1 
All checked for updates

**Additional context**
This:

https://github.com/zephyrproject-rtos/zephyr/blob/8c6a3dbe37abd9b905ac89a375b97a811470dcbd/scripts/west_commands/runners/nrfjprog.py#L48

Is accessed here:

https://github.com/zephyrproject-rtos/zephyr/blob/8c6a3dbe37abd9b905ac89a375b97a811470dcbd/scripts/west_commands/runners/nrfjprog.py#L92

with a key of 'NRF52' rather than 'NRF52_FAMILY' which, as above fails.

Copy the hex path and using :

`nrfjprog --program  /Users/nick/Documents/nRF/blinky/build/zephyr/zephyr.hex --verify --reset`

works as expected - I tend to have only one board plugged in at a time so don't need to give it an id.

Slightly more convenient fix is comment out this line:

https://github.com/zephyrproject-rtos/zephyr/blob/8c6a3dbe37abd9b905ac89a375b97a811470dcbd/boards/arm/ubx_evkninab1_nrf52832/board.cmake#L6

And then clicking `Flash` just works.

A cursory search reveals all the u-Blox boards and others, including MicroBit v2, are likely affected.

I can PR a fix to comment out the line in `board.cmake` but that's a hack. I've tried to figure out the path that leads to the wrong key being passed over to nrfjprog but it's not clear what the flow is.

Can test and can try other things.

carlescufi, 2023-10-11:
> I can PR a fix to comment out the line in `board.cmake` but that's a hack. I've tried to figure out the path that leads to the wrong key being passed over to nrfjprog but it's not clear what the flow is.
> 
> Can test and can try other things.

Thanks @descartes for the report, this is my bug, I introduced this in a refactor I made a little while ago. I will send a PR fixing this and will ping you there.

carlescufi, 2023-10-11:
@descartes this has already been fixed in Zephyr. The issue is that NCS v2.4.2 does not include the fix, which you can find here:
https://github.com/zephyrproject-rtos/zephyr/commit/0ebc37a97252e4429e3c880f970d856b011aa618

You will need to cherry-pick that fix into your zephyr tree or wait until NCS 2.5.0.

Thanks again!

closed 2023-10-11 by carlescufi

descartes, 2023-10-12:
Cheers for this - project is using just the NINA-B1 board but just as easy to apply the universal fix to a few remote workstations.

Any timelines for an update to NCS?

carlescufi, 2023-10-12:
> Hopefully this issue will act as a placeholder for anyone searching - pending an update to NCS , would it be better to leave it open for others experiencing the same problem?

No, this GitHub project is for upsrtream Zephyr, so it only reflects the state of it, and the bug is fixed here. NCS is a separate project. However, users will be able to find this issue by googling even if it’s closed.

> Any timelines for an update to NCS?

I think a couple of weeks