zephyrproject-rtos/zephyr #63820: USB RNDIS Packet errors on HTTPS webserver

darshanmirajkar-eaton, 2023-10-11:
Observed RNDIS Packet errors on HTTPS webserver on Chrome browser. Tried with Other browsers Firefox and Edge. Works fine. Didn't observe any Packet errors. 

Error Logs
****************
[00:00:17.594,000] .[1;31m<err> usb_rndis: Error writing data to pkt: 0x2003415c.[0m
[00:00:17.601,000] .[1;31m<err> usb_rndis: Too small packet len 34.[0m
[00:00:17.607,000] .[1;31m<err> usb_rndis: Error parsing RNDIS header.[0m
[00:00:18.058,000] .[1;31m<err> usb_rndis: Error writing data to pkt: 0x2003415c.[0m
[00:00:18.065,000] .[1;31m<err> usb_rndis: Too small packet len 34.[0m
[00:00:18.072,000] .[1;31m<err> usb_rndis: Error parsing RNDIS header.[0m
Heartbeat Counter = 18
[00:00:18.986,000] .[1;31m<err> usb_rndis: Error writing data to pkt: 0x2003415c.[0m
[00:00:18.993,000] .[1;31m<err> usb_rndis: Wrong data packet type 0x11010000.[0m
[00:00:19.000,000] .[1;31m<err> usb_rndis: Error parsing RNDIS header.[0m
[00:00:19.007,000] .[1;31m<err> usb_rndis: Too small packet len 8.[0m
[00:00:19.013,000] .[1;31m<err> usb_rndis: Error parsing RNDIS header.[0m
Heartbeat Counter = 19
[00:00:20.179,000] .[1;31m<err> usb_rndis: Error writing data to pkt: 0x2003415c.[0m
[00:00:20.186,000] .[1;31m<err> usb_rndis: Wrong data packet type 0x11010000.[0m
[00:00:20.193,000] .[1;31m<err> usb_rndis: Error parsing RNDIS header.[0m
[00:00:20.200,000] .[1;31m<err> usb_rndis: Wrong data packet type 0x3039313a.[0m
[00:00:20.207,000] .[1;31m<err> usb_rndis: Error parsing RNDIS header.[0m
[00:00:20.213,000] .[1;31m<err> usb_rndis: Wrong data packet type 0x7265733a.[0m

palaniappanprasanna-eaton, 2023-10-13:
@jfischer-no any inputs or suggestions on this issue.

jfischer-no, 2023-10-16:
> @jfischer-no any inputs or suggestions on this issue.

Use the bug report template for bug reports. It is not clear which application in the tree is used to reproduce it, move it to discussion maybe it will evolve into something useful.