zephyrproject-rtos/zephyr #63830: RFC: Sensor Channel Identifiers

teburd, 2023-10-11:
## Introduction

Sensor channels today are solely identified by the type of information they contain. Many sensors provide multiple channels of the same measurement type.

Example devices of varying kinds

| Device       | Channels | Channel Type  | Product URL                                          |
|--------------|----------|---------------|------------------------------------------------------|
| TI FDC1004   | 4        | Capacitance   | https://www.ti.com/product/FDC1004                   |
| AD MAX6581   | 8        | Temperature   | https://www.analog.com/en/products/max6581.html      |
| AMS AS7261   | 6        | Light Spectra | https://ams.com/as7262                               |
| ST  VL53L5CX | 8x8      | ToF Array     | https://www.st.com/en/imaging-and-photonics-solutions/vl53l5cx.html |

### Problem description

Selecting channels simply by their type of measurement isn't flexible enough for such array like sensor devices that exist and are used widely in the real world.

### Proposed change

Better define sensor channels as a triple (type, optional subtype, optional idx)

### Proposed change (Detailed)

Sensor channels become defined as a uint64_t unioned with a struct to define a triple, though may continue to be represented with the existing enum with some careful usage.

``` c
enum sensor_channel {
  SENSOR_CHAN_ACCEL_X, /* Legacy channel types such as those describing a pseudo index or subtype come before the CHAN_NEW_START range */
  SENSOR_CHAN_ACCEL_XYZ,
  SENSOR_CHAN_MAX = INT16_MAX, /* Exists in tree today, convert to UINT16_MAX */
  SENSOR_CHAN_TYPE_NEW_START = SENSOR_CHAN_MAX+1
  SENSOR_CHAN_PRESS, /* Compatible channel types are moved into the non-legacy range */
  SENSOR_CHAN_HUMIDITY, 
  ...
  SENSOR_CHAN_TYPE_NEW_END = UINT32_MAX,
};

#define SENSOR_CHAN_LEGACY_END = UINT16_MAX; /* Ensure all existing enum values are understood

union sensor_chan {
   uint64_t chan_id;
   struct sensor_channel_triple {
      uint16_t chan_idx;
      uint16_t chan_subtype;
      uint32_t chan_type;
   };
};
```

In order to maintain most backwards compatibility the existing enum all values below
pow(2,16) are considered legacy and will be marked as deprecated, but continue to be supported.

A small translation function will be made available to driver authors and application authors
wishing to use the new union while mapping the legacy channels to new channel identifiers.

Perhaps an infallible mapping function pair would be enough with asserts, as using non legacy compatible values should be considered a programming error not a runtime error.

```c
union sensor_chan sensor_chan_from_legacy(enum sensor_channel legacy);
enum sensor_channel sensor_chan_to_legacy(union sensor_chan chan);
```

For an example mapping of how something like SENSOR_CHANNEL_ACCEL_XYZ might be mapped...

``` c
union sensor_chan accel_xyz_0 = {
  .chan_subtype = SENSOR_CHAN_SUBTYPE_XYZ,
  .idx = 0
  .chan_type = SENSOR_CHAN_TYPE_ACCEL,
};
```

In order to again not break backwards compatibility... sensor_fetch/sensor_get/sensor_set_attr/sensor_get_attr will maintain the legacy enum sensor_channel parameters for the time being. Noting that values greater than pow(2,16) are in fact the newer channel identifier type.

We could wrap these functions with a macro so that the type of the channel parameter does not matter to start and both types might be accepted. E.g.

``` c
#define sensor_attr_set(dev, chan, attr, val) \
  do {
     enum sensor_chan chan0 = (enum sensor_chan)chan;
     sensor_attr_set(dev, chan, attr, val);
  } while(0)
```

By doing so both the *new* and *old* channel identifier could be accepted. The driver is
in charge of understanding how to deal with both and should be straightforward.

All newer APIs, e.g. sensor_read and the sensor_stream API would be updated to only support the newer channel identifier type while again mapping to the legacy type for backwards compatibility with the 100+ drivers in tree. A deprecation path would follow.

microbuilder, 2023-10-12:
If index here for the spectrometer examples is analogous to the wavelength in nm, a 16-bit integer should be sufficient for most use cases from UV to IR, but is there a mechanism to programmatically determine what index values are exposed by the driver? Ex: 285, 315, 360, 405, 452, 505, 580, 670, 745, 780 (random numbers) might be a sensible UV to visible sensor range in nm, but how do I know what's available as a developer?

Otherwise, would a linear index 0..9 make more sense with per channel/index attributes to indicate the center wavelength in nm, potentially with more precision since it's unlikely to be a clean integer? I could then just scan the index range until I hit an exception, and probably also set a precise value like 524.6 nm.

Sorry if the answer is obvious ... on my phone and I've been away from the sensor API for a long time.

teburd, 2023-10-12:
> If index here for the spectrometer examples is analogous to the wavelength in nm (500 nm here), a 16-bit integer should be sufficient for most use cases from UV to IR, but is there a mechanism to programmatically determine what index values are exposed by the driver? Ex: 285, 315, 360, 405, 452, 505, 580, 670, 745, 780 (random numbers) might be a sensible UV to visible to IR sensor range in nm, but how do I know what's available as a developer?
> 
> Otherwise, would a linear index 0..9 make more sense with per channel/index attributes to indicate the center wavelength in nm, potentially with more precision since it's unlikely to be a clean integer? I could then just scan the index range until I hit an exception, and probably also set a precise value like 524.6 nm via an attribute?

yeah I changed it in realizing this, the index is just the channel index into the sensor channel array now. So an attribute could be looked at to for the wavelength.

I think some of them also let you maybe select which wavelength in some channels to use? I swore I saw one that did that. Which makes this even more dubious to try and use the index as a wavelength.

> 
> Sorry if the answer is obvious ... on my phone and I've been away from the sensor API for a long time.

No, you were spot on, and I've corrected it.

jilaypandya, 2023-10-26:
Hi @teburd, This is a very relevant use-case for one of the sensors i am working on. +1 from me and would be happy to collaborate on this.

teburd, 2023-11-06:
Related things
- Related PR https://github.com/zephyrproject-rtos/zephyr/issues/62944
- Related Issue https://github.com/zephyrproject-rtos/zephyr/issues/61163
- Related PR https://github.com/zephyrproject-rtos/zephyr/pull/61271
- Related PR https://github.com/zephyrproject-rtos/zephyr/pull/60833
- Related PR https://github.com/zephyrproject-rtos/zephyr/pull/31056

henrikbrixandersen, 2023-11-07:
> Hi @teburd, This is a very relevant use-case for one of the sensors i am working on. +1 from me and would be happy to collaborate on this.

Same here. As mentioned in the Sensors WG meeting yesterday, my only concern is adding a new API and adding "legacy" wrappers instead of just modifying the old API. At minimum, I think we need to have a migration plan for all in-tree drivers to use the new sensor channel specifiers.

One option could be to go with the legacy wrappers for the initial PR and create a meta issue for assigning/tracking driver conversions.

kartben, 2023-11-13:
Another candidate would be for particulate matter and particle count sensors, where the subtype could be used to specify the particle size, among the common values 1.0, 2.5, etc.

teburd, 2023-11-13:
> Another candidate would be for particulate matter and particle count sensors, where the subtype could be used to specify the particle size, among the common values 1.0, 2.5, etc.

The more I keep looking at this, the more I wonder if what we want is to put channel descriptors into DT and solely use an index into this channel descriptor array as the identifier, with of course some functions to find the correct channel index.

```c
uint32_t pm25_idx = DT_SENSOR_CHANNEL_IDX(mysensornode, SENSOR_CHAN_PM, SENSOR_CHAN_ST_2_5, 0);
```

Kind of thing perhaps...

This way we can at runtime also iterate over all potential channels a sensor has for things like the sensor shell, or maybe a sensor tracing app (dump the data to a visualizer on a host PC)

I think there's some nice reasons to go this route and maybe worth revamping the RFC a little to account for the need to introspect channels at runtime.

dolence, 2024-03-21:
Since it was pointed as something relevant to this, I would like to explain my use case.

I have a custom board with two DS2482-800, an 8 channels IC used to handle the communication with multiple DS18B20 and other one-wire devices. On my project I will  have up to 32 sensors per channel on a long cable (one sensor each 1.5m), to the max of 16*32=512 sensors. Another board will periodically send a packet through Modbus or LoRaWAN asking for a temperature reading on all the active channels, one by one.

This is how I have things set at this moment:

```
&i2c2 {
	eeprom0: eeprom@50 {
		compatible = "atmel,at24";
		reg = <0x50>;
		status = "okay";
		size = <256>;
		pagesize = <8>;
		address-width = <8>;
		timeout = <5>;
	};

	w1@19 {
		compatible = "maxim,ds2482-800";
		reg = <0x19>;

		#address-cells = <1>;
		#size-cells = <0>;

		w1_a_ch0: ch@0 {
			compatible = "maxim,ds2482-800-channel";
			reg = <7>;
		};
		w1_a_ch1: ch@1 {
			compatible = "maxim,ds2482-800-channel";
			reg = <6>;
		};
		w1_a_ch2: ch@2 {
			compatible = "maxim,ds2482-800-channel";
			reg = <5>;
		};
		w1_a_ch3: ch@3 {
			compatible = "maxim,ds2482-800-channel";
			reg = <4>;
		};
		w1_a_ch4: ch@4 {
			compatible = "maxim,ds2482-800-channel";
			reg = <3>;
		};
		w1_a_ch5: ch@5 {
			compatible = "maxim,ds2482-800-channel";
			reg = <2>;
		};
		w1_a_ch6: ch@6 {
			compatible = "maxim,ds2482-800-channel";
			reg = <1>;
		};
		w1_a_ch7: ch@7 {
			compatible = "maxim,ds2482-800-channel";
			reg = <0>;
		};
	};

	w1@18 {
		compatible = "maxim,ds2482-800";
		reg = <0x18>;

		#address-cells = <1>;
		#size-cells = <0>;

		w1_b_ch8: ch@8 {
			compatible = "maxim,ds2482-800-channel";
			reg = <7>;
		};
		w1_b_ch9: ch@9 {
			compatible = "maxim,ds2482-800-channel";
			reg = <6>;
		};
		w1_b_ch10: ch@10 {
			compatible = "maxim,ds2482-800-channel";
			reg = <5>;
		};
		w1_b_ch11: ch@11{
			compatible = "maxim,ds2482-800-channel";
			reg = <4>;
		};
		w1_b_ch12: ch@12 {
			compatible = "maxim,ds2482-800-channel";
			reg = <3>;
		};
		w1_b_ch13: ch@13 {
			compatible = "maxim,ds2482-800-channel";
			reg = <2>;
		};
		w1_b_ch14: ch@14 {
			compatible = "maxim,ds2482-800-channel";
			reg = <1>;
		};
		w1_b_ch15: ch@15 {
			compatible = "maxim,ds2482-800-channel";
			reg = <0>;
		};
	};
};
```

Shell devices list ouput
[image]

w1 seach on a short cable with 7 devices
[image]

w1 search on a much longer cable with 32 devices
[image]

Since I don't know how many channels will be in use or how many sensors will be connected to these channels (and their ROM ID) at the building time, I will need to save somewhere the ROM of each sensor alongside the channel and position on cable. It could be done inside the sensor own two register (1 byte each) or to an external eeprom.

My problem relies on how to set this to make use of sensors API. 512 node entries for each sensor would be impractical for obvious reasons. 
This is something that would work, for reading only one sensor at a time (https://github.com/zephyrproject-rtos/zephyr/discussions/55743#discussioncomment-5422388) but I would lose the ability to multidrop and convert an entire channel at once.

I would like to hear your opinions and would be nice to know if my case could be take in account or if it has relevance to this discussion.

teburd, 2024-05-17:
We have channel specifiers in now with #71093 now with the new read+decode API and I'm considering this complete

closed 2024-05-17 by teburd

teburd, 2024-05-22:
A small addendum here, I think its still great if we could have a channel descriptor of some kind available, but doing it in DT turns out to be somewhat painful, either we have an array with modulo dependent data (e.g. position 0 is type, position 1 is index, etc) or better would be to have a sub node representing what really is a subset of sensor channels typically, e.g. accel x,y,z and gyro x,y,z with their own independent properties and configuration (this is often possible).

So that's where #71235 comes in and hopefully resolves the issue of describing sensors in a way that could be converted to C code useful to zephyr.

dolence, 2024-05-22:
I still have a problem understanding how would a fetch an entire bus of sensors wired to the channels of a DS2482-800. I have many sensors per channel and need to send a skip rom command to convert all and after the conversion ends I would need to read one by one by rom ID.

teburd, 2024-05-22:
assuming you describe the device layout property with dt as you noted above you could setup a single reader that asks for all the temperature sensors hanging off the i2c to 1 wire converter. It would require a device driver that understands your setup though.

E.g. something like...

```c
SENSOR_DT_READ_IODEV(w1, DT_NODELABEL(w1),
    { SENSOR_CHAN_TEMP, 0 },
    { SENSOR_CHAN_TEMP, 1 },
    ...
    { SENSOR_CHAN_TEMP, 8 },
);
SENSOR_DT_READ_IODEV(w2, DT_NODELABEL(w2),
    { SENSOR_CHAN_TEMP, 0 },
    { SENSOR_CHAN_TEMP, 1 },
    ...
    { SENSOR_CHAN_TEMP, 8 },
);
...

struct rtio_iodev *temp_devices[TEMP_DEVS] = LISTIFY(...);

RTIO_MEMPOOL_DEFINE(r, TEMP_DEVS, TEMP_DEVS, 64, 64);

int main() {
    struct rtio_sqe *sqe;
    struct rtio_cqe *cqe;

    while(true) {
        for (int i = 0; i < TEMP_DEVS; i++) {
            sqe = rtio_sqe_acquire(r);
            rtio_prep_read_with_pool(sqe, w1, w1);
        }

        /* wait for all temp sensors to read, may happen concurrently if multiple i2c buses are used */
        rtio_submit(r, TEMP_DEVS); 
        
        for (int i = 0; i < TEMP_DEVS; i++) {
            /* get the completion queue entry for each read request to the sensors */
            cqe = rtio_cqe_consume(r);

	    /* Cache the data from the CQE */
	    rc = cqe->result;
	    userdata = cqe->userdata;
	    rtio_cqe_get_mempool_buffer(ctx, cqe, &buf, &buf_len);

	    /* Release the CQE */
	    rtio_cqe_release(ctx, cqe);

	    /* Decode the buffer with decoder->decode() into a vector of fixed point values, optional! */
            /* decoders[i]->decode(...);
	    
	    /* Release the memory */
	    rtio_release_buffer(ctx, buf, buf_len);
        }

        /* Poll once a second */
        k_msleep(1000);

    }
}
```

dolence, 2024-05-22:
> assuming you describe the device layout property with dt as you noted above you could setup a single reader that asks for all the temperature sensors hanging off the all of the i2c to 1 wire converters. It would require a device driver that understands your setup though (the w1, w2, etc in this case)
> 
> E.g. something like...
> 
> ```c
> SENSOR_DT_READ_IODEV(w1, DT_NODELABEL(w1),
>     { SENSOR_CHAN_TEMP, 0 },
>     { SENSOR_CHAN_TEMP, 1 },
>     ...
>     { SENSOR_CHAN_TEMP, 8 },
> );
> SENSOR_DT_READ_IODEV(w2, DT_NODELABEL(w2),
>     { SENSOR_CHAN_TEMP, 0 },
>     { SENSOR_CHAN_TEMP, 1 },
>     ...
>     { SENSOR_CHAN_TEMP, 8 },
> );
> ...
> 
> /* put all the iodev pointers in an array for ease of use */
> struct rtio_iodev *temp_devices[TEMP_DEVS] = LISTIFY(...);
> 
> /* put a matching decoder for each sensor into an array for ease of use */
> struct sensor_decoder *decoders[TEMP_DEVS] = LISTIFY(...);
> 
> /* create an async executor context with a mempool and an exactly matching number of inflight requests/completions as the number of sensors */
> RTIO_MEMPOOL_DEFINE(r, TEMP_DEVS, TEMP_DEVS, 64, 64);
> 
> 
> int main() {
>     struct rtio_sqe *sqe;
>     struct rtio_cqe *cqe;
>     uint8_t *buf;
>     size_t buf_len;
> 
>     while(true) {
>         for (int i = 0; i < TEMP_DEVS; i++) {
>             sqe = rtio_sqe_acquire(r);
>             rtio_prep_read_with_pool(sqe, temp_devs[i], temp_devs[i]);
>         }
> 
>         /* wait for all temp sensors to read, may happen concurrently if multiple i2c buses are used */
>         rtio_submit(r, TEMP_DEVS); 
>         
>         for (int i = 0; i < TEMP_DEVS; i++) {
>             /* get the completion queue entry for each read request to the sensors */
>             cqe = rtio_cqe_consume(r);
> 
> 	    /* contains the result of the read request, 0 success, -errno for error as usual */
> 	    rc = cqe->result;
> 
> 	    /* is a pointer to one of the temp devs, letting you determine *which* sensor the completion came from */
> 	    userdata = cqe->userdata; 
> 
>             /* since we used the builtin memory pool to stash the readings, we need to get the pointer to the associated buffer */
> 	    rtio_cqe_get_mempool_buffer(ctx, cqe, &buf, &buf_len);
> 
> 	    /* Release the CQE */
> 	    rtio_cqe_release(ctx, cqe);
> 
> 	    /* Decode the buffer with decoder->decode() into a vector of fixed point values, optional! */
>             /* decoders[i]->decode(...);
> 	    
> 	    /* Release the memory back to the mempool */
> 	    rtio_release_buffer(ctx, buf, buf_len);
>         }
> 
>         /* Poll once a second */
>         k_msleep(1000);
> 
>     }
> }
> ```
>  
> 
> 

As someone who is beginning on my journey through learning Zephyr and RTOS I must say this is really nice. I didn't knew about RTIO API and it seemed to be exactly what I need. Thank you for providing this example. I have a lot more to learn now. May I ask some questions? What you mean by "a device driver that understands my setup"? I guess ds2482/w1 doesn't support RTIO API by default, is this the case? Would you please elaborate a little more the concept behind it, considering I'm still getting the concept behind this?

teburd, 2024-05-22:
> > assuming you describe the device layout property with dt as you noted above you could setup a single reader that asks for all the temperature sensors hanging off the all of the i2c to 1 wire converters. It would require a device driver that understands your setup though (the w1, w2, etc in this case)
> 
> As someone who is beginning on my journey through learning Zephyr and RTOS I must say this is really nice. I didn't knew about RTIO API and it seemed to be exactly what I need. Thank you for providing this example. I have a lot more to learn now. May I ask some questions? What you mean by "a device driver that understands my setup"? I guess ds2482/w1 doesn't support RTIO API by default, is this the case? Would you please elaborate a little more the concept behind it, considering I'm still getting the general idea behind this?

Absolutely, the RTIO API from a sensor driver needs to implement the submit and decoder APIs (decoder is specific to sensors, for interpreting the readings).

The idea here is that the driver gets submit called whenever requests (read requests in this case) come in, the read request comes with a buffer you can directly read into.

For sensors the idea would be to directly read the register readings into the buffer without manipulation, allowing for the raw data to be directly used by the application if desired. A small prefix or suffix can be used to stash some data for the decoder (the decoder is stateless!).

so for example..

```c

void ds2482_consume(const struct device *dev);

void ds2482_read_complete(....)
{
   /* get completions from our driver's internal async context (data->r);

   /* note that our current request is *done* /
   rtio_iodev_sqe_ok(data->r, data->sqe);

   /* continue working through the queue */
   ds2482_consume(dev);
}

void ds2482_start(const struct device *dev) 
{
   struct ds2482_data *data = dev->data;

   /* start async i2c request with rtio */
   struct rtio_mpsc_node *node = rtio_mpsc_pop(&data->io_q);
   struct rtio_iodev_sqe *sqe = CONTAINER_OF(struct rtio_iodev_sqe, q, node);
   
   __ASSERT(sqe->op == RTIO_SQE_OP_RX, "ds2482 only accepts read requests");

   /* the iodev associated with the sqe can be looked at to determine the channels to read here, not shown though */
   uint8_t *buf;
   size_t buf_len;
   
   /* get a reference to the read buffer (or allocate if using pools) from the incoming request
    * this can validate that the buffer is large enough for our needs 
    */
   int rc = rtio_sqe_rx_buf(sqe, min_buf_len, max_buf_len, &buf, &buf_len);

   __ASSERT(rc == 0, "expected a valid read buffer");

   /* we have our io task descriptor (sqe), our array of channels (from the iodev) request the appropriate reads from i2c peripheral asynchronously */
   struct rtio_sqe *i2c_sqe;

    i2c_sqe = rtio_sqe_acquire(data->r);
    /* write the register value to read... */
    rtio_sqe_prep_write(i2c_sqe,  i2c_iodev, RTIO_PRIO_NORM, (uint8_t*)&reg_addr, sizeof(reg_addr));
    /* mark the request with a flag signaling the next one is part of this transaction */
    i2c_sqe->flags |= RTIO_SQE_TRANSACTION;

    i2c_sqe = rtio_sqe_acquire(data->r);
    /* note we read *directly* into the buffer provided by the application here */
    rtio_sqe_prep_read(i2c_sqe, i2c_iodev, buf, buf_len);
    /* we want the next sqe after this one to be chained to it (requires this one completes first).
    i2c_sqe->flags |= RTIO_SQE_CHAIN;
   
    i2c_sqe = rtio_sqe_acquire(data->r);
    /* finally we want a callback when the transaction completes */
    rtio_sqe_prep-callback(i2c_sqe, ds2482_read_complete);
 
    /* don't wait here, submit and return */
    rtio_submit(data->r, 0);
}

void ds2482_consume(const struct device *dev)
{
 /* attempt to kick off async interrupt driven tasks if not already doing things 
    * since the tasks are asynchronous, this really amounts to a few register writes
    * typically and should be just fine in a spin lock
    */
   k_spinlock_key_t key = k_spin_lock(&data->lock);

   if (data->sqe != NULL) {
      ds2482_start(dev);
   }

   k_spin_unlock(&data->lock, key);
}

void ds2482_submit(const struct device *dev, struct rtio_iodev_sqe *sqe) 
{
    struct ds2482_data *data = dev->data;

   /* put sqe into atomic queue, no lock needed here, the "multiple producer" side */
   rtio_mpsc_push(&data->io_q, &sqe->q);

  
}

dolence, 2024-05-22:
Thank you very much for your time. This is a very nice explanation! Now I have plenty of things to work with.

bryceschober, 2024-05-23:
The comments in this thread sound like they could be fleshed out a bit more and become a good reference example for RTIO usage. It seems like there are important exemplary details here that aren't in the [existing `samples/subsys/rtio/sensor_batch_processing` sample](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/subsys/rtio/sensor_batch_processing), which doesn't even have a README...

teburd, 2024-05-23:
> The comments in this thread sound like they could be fleshed out a bit more and become a good reference example for RTIO usage. It seems like there are important exemplary details here that aren't [...]

Fair! I have on my TODO list updating the docs... which I was working on and then got side tracked trying to make building docs not take so long :computer: :sloth: