zephyrproject-rtos/zephyr #63831: fdtable entries are not initialized properly with CONFIG_POSIX_API on

mengyaliu, 2023-10-12:
When CONFIG_POSIX_API & CONFIG_FDTABLE options are both enabled, the stdin/stdout/stderr entries of fdtable are not initialized correctly. The initialization of member 'lock' of these three entries is missed.

1. Most of fdtable entries are initialized in function z_reserve_fd() from file zephyr/lib/os/fdtable.c
```
  int z_reserve_fd(void)
  {
          int fd;
 
          (void)k_mutex_lock(&fdtable_lock, K_FOREVER);
 
          fd = _find_fd_entry();
          if (fd >= 0) {
                  /* Mark entry as used, z_finalize_fd() will fill it in. */
                  (void)z_fd_ref(fd);
                  fdtable[fd].obj = NULL;
                  fdtable[fd].vtable = NULL;
                  k_mutex_init(&fdtable[fd].lock);
                  k_condvar_init(&fdtable[fd].cond);
          }
 
          k_mutex_unlock(&fdtable_lock);
 
          return fd;
 }
```
2. When CONFIG_POSIX_API enabled, the first three entries in fdtable will be initialized statically as below, but member 'lock' is missed.
```
   static struct fd_entry fdtable[CONFIG_POSIX_MAX_FDS] = {
   #ifdef CONFIG_POSIX_API
           /*
            * Predefine entries for stdin/stdout/stderr.
            */
           {
                   /* STDIN */
                   .vtable = &stdinout_fd_op_vtable,
                   .refcount = ATOMIC_INIT(1)
           },
          {
                   /* STDOUT */
                   .vtable = &stdinout_fd_op_vtable,
                   .refcount = ATOMIC_INIT(1)
           },
           {
                   /* STDERR */
                   .vtable = &stdinout_fd_op_vtable,
                   .refcount = ATOMIC_INIT(1)
           },
```
3. We can clearly find the fdtable content in gdb console when running a test, the 'lock' member of these entries are not initialized, waitq should not be all zero.
```
(gdb) p fdtable[0]
$2 = {obj = 0x0, vtable = 0x101ad7c0 <stdinout_fd_op_vtable>, refcount = 1, lock = {wait_q = {waitq = {{head = 0x0, next = 0x0}, {
          tail = 0x0, prev = 0x0}}}, owner = 0x0, lock_count = 0, owner_orig_prio = 0}, cond = {wait_q = {waitq = {{head = 0x0,
          next = 0x0}, {tail = 0x0, prev = 0x0}}}}}
(gdb) p fdtable[1]
$3 = {obj = 0x0, vtable = 0x101ad7c0 <stdinout_fd_op_vtable>, refcount = 1, lock = {wait_q = {waitq = {{head = 0x0, next = 0x0}, {
          tail = 0x0, prev = 0x0}}}, owner = 0x0, lock_count = 0, owner_orig_prio = 0}, cond = {wait_q = {waitq = {{head = 0x0,
          next = 0x0}, {tail = 0x0, prev = 0x0}}}}}
(gdb) p fdtable[2]
$4 = {obj = 0x0, vtable = 0x101ad7c0 <stdinout_fd_op_vtable>, refcount = 1, lock = {wait_q = {waitq = {{head = 0x0, next = 0x0}, {
          tail = 0x0, prev = 0x0}}}, owner = 0x0, lock_count = 0, owner_orig_prio = 0}, cond = {wait_q = {waitq = {{head = 0x0,
          next = 0x0}, {tail = 0x0, prev = 0x0}}}}}
```

**Environment (please complete the following information):**
 - OS: docker image: ghcr.io/zephyrproject-rtos/zephyr-build
 - Toolchain: /opt/toolchains/zephyr-sdk-0.16.1/x86_64-zephyr-elf/
 - Commit SHA: 5cf5282e60ba0a9df91690a4a08a5900bca7b818

cfriedt, 2023-10-12:
Thanks for spotting this - I'll see if I can whip up a PR quickly.

mengyaliu, 2023-10-12:
> Thanks for spotting this - I'll see if I can whip up a PR quickly.

You're welcome, and appreciate your prompt reply. 👍

PR #63895 (2023-10-13): posix: fdtable: ensure stdin, stdout, and stderr are initialized
Ensure that stdin, stdout, and stderr are initialized statically.

Previously, the mutex and condition variable were uninitialized.

Fixes #63831
closed 2023-10-13 by cfriedt