zephyrproject-rtos/zephyr #63845: Change 0-length arrays to flexible arrays

Thalley, 2023-10-12:
**Is your enhancement proposal related to a problem? Please describe.**
Searching for ` \w+\[0\];\n\}` in Zephyr yields 195 results (as of 8c6a3dbe37abd9b905ac89a375b97a811470dcbd) of arrays defied with `[0]`, which is not a valid C array as per https://www.open-std.org/jtc1/sc22/WG14/www/docs/n1256.pdf section `6.7.5.2 Array declarators` which state:

> In addition to optional type qualifiers and the keyword static, the [ and ] may delimit an expression or *. If they delimit an expression (which specifies the size of an array), the expression shall have an integer type. If the expression is a constant expression, **it shall have a value greater than zero**

The `[0]` is in fact a compiler extension, and since it is used to denote an element with a variable length (as per https://gcc.gnu.org/onlinedocs/gcc/Zero-Length.html), we should be able to replace all `[0]` with `[]`. 

Alternatively, we could also look into bounded flexible arrays that they Linux kernel is currently working on implementing: https://people.kernel.org/kees/bounded-flexible-arrays-in-c

**Describe the solution you'd like**
Replace all `[0]` with `[]` to be compliant with C99 and newer standards, instead of relying on a compiler extension. 

**Describe alternatives you've considered**
Bounded flexible arrays may be a better solution overall, and would replace most current non-bounded flexible arrays. 

**Additional context**
The majority of these `[0]` arrays seems to be in the Bluetooth subsystem.

Thalley, 2024-05-09:
There is a recent warning added to GCC that can help check if there are any flexible array members in the middle of a struct, which is a big no-go: https://gcc.gnu.org/onlinedocs/gcc-14.1.0/gcc/Warning-Options.html#index-Wflex-array-member-not-at-end

Thalley, 2024-05-09:
See also https://embeddedor.com/slides/2024/eo/eo2024.pdf

ceolin, 2024-06-28:
For coincidence I was looking into it. I was looking for ways to improve buffer checking to avoid overflows and that is something that can help. There are some really neat compiler features, unfortunately some of them are not available in our SDK. But they rely on proper arrays declaration.

I am working on it.

ceolin, 2024-06-28:
Among all usages, this one is the weirdest and it seems very error  prone https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/bluetooth/tester/src/btp/btp_mesh.h#L117-L118

ceolin, 2024-06-28:
https://github.com/zephyrproject-rtos/zephyr/pull/75192 to start :)

Thalley, 2024-06-29:
> Among all usages, this one is the weirdest and it seems very error prone https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/bluetooth/tester/src/btp/btp_mesh.h#L117-L118

That is actively disallowed in C. Empty/zero-length arrays are only ever allowed if it is the last value in a struct IIRC (and then that struct shall always be the last field if embedded in other structs). 

Luckily it's just for a test application, but should definitely be fixed (ping @sjanc )

sjanc, 2024-07-01:
yeah, I'll fix that

sjanc, 2024-07-01:
fix https://github.com/zephyrproject-rtos/zephyr/pull/75242

Thalley, 2025-09-15:
The `counted_by` attribute has been adopted in both [GCC](https://gcc.gnu.org/onlinedocs/gcc/Common-Variable-Attributes.html#index-counted_005fby-variable-attribute) and [clang](https://clang.llvm.org/docs/AttributeReference.html#counted-by-counted-by-or-null-sized-by-sized-by-or-null)