zephyrproject-rtos/zephyr #63885: drivers: uart: stm32: Power management is broken after recent commit

coran21, 2023-10-12:
Power management for UART is broken after merging PR #63158 made by @erwango .
The code on line 1272 in `drivers/serial/uart_stm32.c` is actually never called if you don't use  `CONFIG_UART_ASYNC_API`.

```
if (pm_device_runtime_is_enabled(dev)) {
	(void)pm_device_runtime_put(dev);
}

```
 I believe that this code should be somewhere around line 1233.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?  
    STM32L4R5 
 - What have you tried to diagnose or workaround this issue?
   Measured power consumption and checked clocks for all used peripherals.

**To Reproduce**
Steps to reproduce the behavior:
1. use UART with `CONFIG_UART_INTERRUPT_DRIVEN` and without `CONFIG_UART_ASYNC_API`

**Expected behavior**
The power consumption of my system will be the same as before merging this PR. 

**Impact**
UART clock runs all the time and power consumption is about 0.5mA higher

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows) :  Linux
 - Toolchain (e.g Zephyr SDK, ...)  :  Zephyr SDK
 - Commit SHA or Version used :  3.5.0

Are these changes actually tested by someone?

erwango, 2023-10-13:
@fabiobaltieri FYI, working on a fix, but might be required to revert

erwango, 2023-10-13:
Unfortunately, today I'll miss time and HW to find a proper solution which fixes both issues.
Considering the release timeline, I'm pushing #63907 (revert of https://github.com/zephyrproject-rtos/zephyr/pull/63158) as a lesser evil, which can be merged once confirmed by @coran21 that it is fixing his issue.
I'll work on a follow up fix, but I can't commit it can be fixed in due time for the release

PR #63907 (2023-10-13): Revert "drivers: serial: stm32: Make it compatible with runtime PM ....
This reverts commit 1c2d326579ee8cea4c809eb737585537e3482677. which was limited to CONFIG_UART_ASYNC_API=y case and causing regression otherwise.

Fixes #63885
commit 179d435a (2023-10-13): Revert "drivers: serial: stm32: Make it compatible with runtime PM ....

This reverts commit 1c2d326579ee8cea4c809eb737585537e3482677.
which was limited to CONFIG_UART_ASYNC_API=y case and causing regression
otherwise.

Fixes #63885
closed 2023-10-13 by fabiobaltieri