zephyrproject-rtos/zephyr #63894: LoRa : Lower bandwidth settings are not listed/supported

kokroo, 2023-10-13:
https://github.com/zephyrproject-rtos/zephyr/blob/38e86c8e86f8bc9b0c0794edc26151f7907145cf/include/zephyr/drivers/lora.h#L34C1-L34C1

LoRa physical layer supports bandwidths lower than 125 kHz. 62.5 kHz for example. I am not sure why they are not supported here. I am developing some firmware on an NRF board and got stuck here trying to set the bandwidth lower.

henrikbrixandersen, 2023-10-13:
It is difficult for me to tell whether you are reporting a bug or asking a question. If you are reporting a bug, please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

kokroo, 2023-10-14:
> It is difficult for me to tell whether you are reporting a bug or asking a question. If you are reporting a bug, please use our bug template when reporting bugs. You need to edit this issue to [...]

@henrikbrixandersen Thank you for the guidance. Actually I'm not sure if it's a bug or a question at this point. Perhaps the Zephyr devs ommited those bandwidths for some reason? Otherwise, LoRa does support bandwidth as low as 7 kHz as well. If anyone who worked on the LoRa parts can comment on this, I can definitely edit the issue and add relevant details based on their comment.

JordanYates, 2023-10-14:
While the physical layer may support it, `loramac-node` (which is the base of our LoRa support) only exposes the 125kHz, 250kHz and 500kHz bandwidths.

https://github.com/Lora-net/LoRaMac-node/blob/6384af8f4c4a57016c83f3d34430f0f68855e695/src/radio/radio.h#L210-L249

This is therefore not a Zepyhr limitation, but inherited from upstream. If this is an important issue for you, I would suggest contributing support in that project, which can then be integrated here.

kokroo, 2023-10-14:
@JordanYates Thank you so much, I'll take it up there and come back to this issue once it's fixed upstream.

JordanYates, 2023-10-14:
I'll close this for now, feel free to re-open once loramac-node supports other settings

closed 2023-10-14 by JordanYates