zephyrproject-rtos/zephyr #63901: Sparc window flush code uses incorrect stack pointer?

keith-packard, 2023-10-13:
I'm building some stand-alone sparc test code for picolibc and found this code for flushing the register windows:

https://github.com/zephyrproject-rtos/zephyr/blob/38e86c8e86f8bc9b0c0794edc26151f7907145cf/arch/sparc/core/window_trap.S#L95

According to the sparc architecture reference, the trap unconditionally switches to a new register window which leaves the window-relative stack pointer (%o6 aka %sp) with whatever value happened to be in that window at the time the trap was entered. That means this function is saving some registers to a random location in the stack, which probably isn't what was intended.

I think this code should be using the *previous* window stack pointer (%i6, aka %fp) instead. Simply flipping all of the %sp to %fp seems like the right plan.

But, my knowledge of sparc architecture is really (really) stale, having last worked on this architecture around 1992...

julius-barendt, 2023-10-17:
I can confirm that this implementation is buggy/not correct behavior, seems like we're not testing longjmp()/setjmp() enough :smile:. Will look into how to best fix it

Did you find this via a specific test that triggers it? Could be a good addition to the test-suite

keith-packard, 2023-10-18:
I copied this code into picolibc and am now using the fixed version in CI testing there. I haven't gotten sparc9 code running yet; that would be a nice addition.

commit fd4e6649 (2023-10-25): SPARC: Update the Flush windows software trap

This commit re-implements the SPARC V8 ABI "Flush windows" software
trap. The trap is generated by C++ compilers for exceptions and also by
the C standard library function longjmp().

There were two issues with the previous implementation:
1. It did reads and writes via the stack pointer of the trap window,
   which is not defined.
2. It executed with traps enabled but without the processor run-time
   state set to safely handle traps. In particular there was no valid
   stack for trap processing. Even though interrupt priority was set to
   highe [... cut]
closed 2023-10-25 by MaureenHelm