zephyrproject-rtos/zephyr #63943: net: ip: tcp_conn use-after-free

chih-hung-yu, 2023-10-14:
**Describe the bug**
When `connect()` is called to make a TCP connection to the server, and the server does not listen on the required TCP port, there are two conditions that cause tcp_conn to be used after free.
1. The server responds with RST, tcp_conn is freed due to connection reset by peer.
Execution flow:
a. <B>main thread</B> calls net_tcp_connect() and wait for `k_sem_take(&conn->connect_sem, timeout)`
b. <B>rx_q[0] thread:</B> tcp_in() -> /* Valid RST received. */ -> tcp_conn_close() -> tcp_conn_unref() -> `k_mem_slab_free(&tcp_conns_slab, (void *)conn)`
c.  <B>main thread:</B> `k_sem_take(&conn->connect_sem, timeout)` returns -EAGAIN, but <B>at this time `conn` has already been memset() and freed</B>, and `conn` will also be used in the proceeding code. 

2. The server does not respond, tcp_conn is freed due to retransmission timed out.
Execution flow:
a. <B>main thread</B> calls net_tcp_connect() and wait for `k_sem_take(&conn->connect_sem, timeout)`
b. <B>tcp_work thread:</B> tcp_send_process() -> tcp_conn_close() -> tcp_conn_unref() -> `k_mem_slab_free(&tcp_conns_slab, (void *)conn)`
c.  <B>main thread:</B> `k_sem_take(&conn->connect_sem, timeout)` returns -EAGAIN, but <B>at this time `conn` has already been memset() and freed</B>, and `conn` will also be used in the proceeding code. 

**To Reproduce**
Run the following code with native_posix execution.
```
#include <zephyr/net/socket.h>

int main(void)
{
    int fd, ret;
    struct sockaddr_in servaddr;
    struct in_addr inaddr;

    fd = zsock_socket(AF_INET, SOCK_STREAM, 0); 

    inet_pton(AF_INET, "192.0.2.2", (void *)&inaddr);

    servaddr.sin_family = AF_INET;
    servaddr.sin_addr = inaddr;
    servaddr.sin_port = htons(8088);

    ret = zsock_connect(fd, (const struct sockaddr *)&servaddr, sizeof(servaddr));

    if (ret == 0) {
        zsock_close(fd);
    } else {
        printk("zsock_connect()=%d errno=%d %s\n", ret, errno, strerror(errno));
    }

    return 0;
}
```

**Expected behavior**
1. tcp_conn use-after-free fixed.
2. On reception of RST, connect() should return -1 immediately with errno setting to ECONNREFUSED.

**Environment**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit: https://github.com/zephyrproject-rtos/zephyr/commit/433a85e7bf32ffc77ba194bb903bcca84c44c047

jukkar, 2023-10-16:
@chih-hung-yu I was not able to reproduce this error with your example application. Could you share your configuration file, perhaps I am just missing some option that you have enabled?

chih-hung-yu, 2023-10-16:
> @chih-hung-yu I was not able to reproduce this error with your example application. Could you share your configuration file, perhaps I am just missing some option that you have enabled?

My prj.conf (based on samples/net/eth_native_posix.conf)
```
CONFIG_NETWORKING=y
CONFIG_NET_UDP=y
CONFIG_NET_TCP=y
CONFIG_NET_IPV6=y
CONFIG_NET_IPV4=y
#CONFIG_NET_DHCPV4=y

CONFIG_ENTROPY_GENERATOR=y
CONFIG_TEST_RANDOM_GENERATOR=y
CONFIG_INIT_STACKS=y

CONFIG_NET_LOG=y
CONFIG_LOG=y

CONFIG_NET_STATISTICS=y

CONFIG_NET_PKT_RX_COUNT=32
CONFIG_NET_PKT_TX_COUNT=32
CONFIG_NET_BUF_RX_COUNT=32
CONFIG_NET_BUF_TX_COUNT=32

CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=3
CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT=4
CONFIG_NET_MAX_CONTEXTS=10

CONFIG_NET_SHELL=y

CONFIG_NET_CONFIG_SETTINGS=y
CONFIG_NET_CONFIG_NEED_IPV6=y
CONFIG_NET_CONFIG_NEED_IPV4=y
CONFIG_NET_CONFIG_MY_IPV6_ADDR="2001:db8::1"
CONFIG_NET_CONFIG_PEER_IPV6_ADDR="2001:db8::2"
CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.0.2.1"
CONFIG_NET_CONFIG_PEER_IPV4_ADDR="192.0.2.2"

CONFIG_NET_L2_ETHERNET=y

CONFIG_ETH_NATIVE_POSIX=y
CONFIG_ETH_NATIVE_POSIX_RANDOM_MAC=y
#CONFIG_ETH_NATIVE_POSIX_MAC_ADDR="00:00:5e:00:53:2a"

#
# above copied from samples/net/eth_native_posix.conf
# following added by chih-hung-yu
#
CONFIG_NATIVE_UART_0_ON_STDINOUT=y
CONFIG_NET_SOCKETS=y
CONFIG_NET_TCP_INIT_RETRANSMISSION_TIMEOUT=1000
CONFIG_NET_TCP_RETRY_COUNT=10
```

<B>My test procedure:</B>
1. print debug messages in subsys/net/ip/tcp.c
[image]
[image]

2. One terminal runs net-tools/net-setup.sh to create `zeth` interface with IP address `192.0.2.2`
```
root@debian:/zephyr/zephyrmaster/tools/net-tools# ./net-setup.sh 
Using ./zeth.conf configuration file.
Creating zeth
```

3.  Another terminal runs sample code.
```
root@debian:/zephyr/tcp_block_connect# west build -b native_posix --pristine
root@debian:/zephyr/tcp_block_connect# build/zephyr/zephyr.exe
WARNING: Using a test - not safe - entropy source

[00:00:00.110,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:00.110,000] <inf> net_config: IPv6 address: 2001:db8::1
<tcp_in> uptime=180: Valid RST received
<tcp_conn_unref> uptime=180: thread='rx_q[0]' conn=0x5660b7c8 has been memset() and freed
<net_tcp_connect> uptime=180: thread='main' k_sem_take()=-11 conn=0x5660b7c8 conn->context=(nil)
zsock_connect()=-1 errno=110 Connection timed out
uart:~$ 
```

And we can see from debug message that when `main thread` is waked up from `k_sem_take(&conn->connect_sem, timeout)`, `conn` has already been memset() and freed.  Timeout for `k_sem_take(&conn->connect_sem, timeout)` is 3000 ms, but it did not wait for 3000 ms but returned -EAGAIN immediately after that `conn` was freed in `tcp_conn_unref`.

jukkar, 2023-10-16:
I am still not able to reproduce this. This is all I get after enabling TCP debugging (`CONFIG_NET_TCP_LOG_LEVEL_DBG=y`):

```
[00:00:00.110,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:00.110,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:00.110,000] <dbg> net_tcp: tcp_conn_ref: (main): conn: 0x80897f8, ref_count: 1
[00:00:00.110,000] <dbg> net_tcp: tcp_conn_alloc: (main): conn: 0x80897f8
[00:00:00.110,000] <dbg> net_tcp: net_tcp_recv: (main): context: 0x80808c0, cb: 0x804c819, user_data: (nil)
[00:00:00.110,000] <dbg> net_tcp: net_tcp_connect: (main): context: 0x80808c0, local: 0.0.0.0, remote: 192.0.2.2
[00:00:00.110,000] <dbg> net_tcp: net_tcp_connect: (main): conn: 0x80897f8 src: 192.0.2.1, dst: 192.0.2.2
[00:00:00.110,000] <dbg> net_tcp: tcp_in: (main):  [LISTEN Seq=1707693388 Ack=0]
[00:00:00.110,000] <dbg> net_tcp: tcp_out_ext: (main): SYN Seq=1707693388 Len=0
[00:00:00.110,000] <dbg> net_tcp: tcp_send_process_no_lock: (main): SYN Seq=1707693388 Len=0 
[00:00:00.110,000] <dbg> net_tcp: tcp_send: (main): SYN Seq=1707693388 Len=0
[00:00:00.110,000] <dbg> net_tcp: tcp_in: (main): LISTEN->SYN_SENT
[00:00:00.180,000] <dbg> net_tcp: tcp_in: (rx_q[0]): ACK,RST Seq=0 Ack=1707693389 Len=0 [SYN_SENT Seq=1707693389 Ack=0]
[00:00:00.180,000] <dbg> net_tcp: tcp_conn_close_debug: (rx_q[0]): conn: 0x80897f8 closed by TCP stack (tcp_in():3017)
[00:00:00.180,000] <dbg> net_tcp: tcp_conn_close_debug: (rx_q[0]): SYN_SENT->CLOSED
[00:00:00.180,000] <dbg> net_tcp: tcp_conn_unref: (rx_q[0]): conn: 0x80897f8, ref_count=1
[00:00:00.180,000] <dbg> net_tcp: net_tcp_connect: (main): conn: 0x80897f8, ret=-110
zsock_connect()=-1 errno=110 Connection timed out
```
Could you generate TCP debug log when the issue happens?

jukkar, 2023-10-16:
So if I am understanding you correctly, the example application is not able to trigger this condition?

chih-hung-yu, 2023-10-17:
Yes, my example application is not able to trigger Segmentation fault condition. I used it to trace kernel code and found there is 'use-after-free' vulnerability. If the timing to free the TCP 'conn' for the closing of a not-yet-established connection is well handled (as discussed in https://github.com/zephyrproject-rtos/zephyr/pull/63984), the problem mentioned here may be resolved.

chih-hung-yu, 2023-10-18:
@jukkar The 'use-after-free' vulnerability mentioned here is resolved in https://github.com/zephyrproject-rtos/zephyr/pull/63984 since in this PR a not-yet-established TCP `conn` is not freed until `close(fd)` is called by the application.

But the blocking `connect()` still returns -1 with errno set to `ETIMEDOUT` when RST is received, is it possible to set errno to `ECONNREFUSED`.

closed 2023-10-24 by chih-hung-yu