zephyrproject-rtos/zephyr #63952: net: ip: Segmentation fault when select() after non-blocking connect()

chih-hung-yu, 2023-10-14:
**Describe the bug**
When non-blocking `connect()` is called to make a TCP connection to the server, `connect()` may return -1 with errno set to EINPROGRESS in the first place, and then we can use select() to wait for the socket being set to be writable which indicates connection established. If the server does not listen on the required TCP port, there are two conditions that cause <B>Segmentation fault</B> due to NULL pointer access.

1. The server responds with RST, tcp_conn is freed due to connection reset by peer.
Execution flow:
a. <B>main thread</B> calls net_tcp_connect() which returns -EINPROGRESS immediately due to K_NO_WAIT timeout
b. <B>rx_q[0] thread:</B> tcp_in() -> /* Valid RST received. */ -> tcp_conn_close() -> tcp_conn_unref() {`conn->context->tcp = NULL;`}
c.  <B>main thread:</B> zsock_select() -> zsock_poll_prepare_ctx() -> net_tcp_conn_sem_get(ctx) <B>{ this function will return ctx->tcp->connect_sem, but ctx->tcp has already been set to NULL in step b., so Segmentation fault is caused due to NULL pointer access }</B>

2. The server does not respond, tcp_conn is finally freed due to retransmission timed out.
Execution flow:
a. <B>main thread</B> calls net_tcp_connect() which returns -EINPROGRESS immediately due to K_NO_WAIT timeout
b. <B>tcp_work thread:</B> tcp_send_process() -> tcp_conn_close() -> tcp_conn_unref() {`conn->context->tcp = NULL;`}
c.  <B>main thread:</B> zsock_select() -> zsock_poll_prepare_ctx() -> net_tcp_conn_sem_get(ctx) <B>{ this function will return ctx->tcp->connect_sem, but ctx->tcp has already been set to NULL in step b., so Segmentation fault is caused due to NULL pointer access }</B>

Relative issue https://github.com/zephyrproject-rtos/zephyr/issues/63943 also caused by already freed tcp_conn.

**Workaround this issue**
Add the following code in zsock_poll_prepare_ctx(), and zsock_select() will return -1 with errno set to ECONNRESET when tcp_conn was freed. But if tcp_conn was freed due to retransmission timed out, it still returns -1 with errno set to ECONNRESET.
[image]

**To Reproduce**
Run the following code with native_posix execution.
```
#include <zephyr/net/socket.h>
#include <fcntl.h>

int main(void)
{
    int fd, flags, ret;
    struct sockaddr_in servaddr;
    struct in_addr inaddr;
    struct timeval tv;
    zsock_fd_set wfds;

    fd = zsock_socket(AF_INET, SOCK_STREAM, 0);

    flags = fcntl(fd, F_GETFL, 0);
    fcntl(fd, F_SETFL, flags | O_NONBLOCK);

    inet_pton(AF_INET, "192.0.2.2", (void *)&inaddr);

    servaddr.sin_family = AF_INET;
    servaddr.sin_addr = inaddr;
    servaddr.sin_port = htons(8088);

    ret = zsock_connect(fd, (const struct sockaddr *)&servaddr, sizeof(servaddr));

    if (ret == 0) {
        printk("connection established\n");
        zsock_close(fd);
    } else {
        printk("zsock_connect()=%d errno=%d %s\n", ret, errno, strerror(errno));
        /* wait for connection established */
        while (1) {
            tv.tv_sec = 5;
            tv.tv_usec = 0;
            FD_ZERO(&wfds);
            FD_SET(fd, &wfds);
            /* connection established if socket is writable */
            ret = zsock_select(fd + 1,  NULL, &wfds, NULL, &tv);
            if (ret == 0) {
                printk("zsock_select()=0 timeout\n");
            } else if (ret < 0) {
                printk("zsock_select()=%d errno=%d %s\n", ret, errno, strerror(errno));
                break;
            } else {
                printk("connection established\n");
                zsock_close(fd);
                break;
            }
        }
    }

    return 0;
}
```

**Expected behavior**
Segmentation fault does not occur and select() returns -1 when TCP connection fails.

**Logs and console output**
```
root@debian:/zephyr/tcp_connect# valgrind build/zephyr/zephyr.exe
==477789== Memcheck, a memory error detector
==477789== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==477789== Using Valgrind-3.16.1 and LibVEX; rerun with -h for copyright info
==477789== Command: build/zephyr/zephyr.exe
==477789== 
WARNING: Using a test - not safe - entropy source

[00:00:00.110,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:00.110,000] <inf> net_config: IPv6 address: 2001:db8::1
zsock_connect()=-1 errno=115 Operation now in progress
zsock_select()=0 timeout
uart:~$ ==477789== Thread 3:
==477789== Invalid read of size 4
==477789==    at 0x138A6F: is_condition_met (poll.c:67)
==477789==    by 0x138A6F: register_events (poll.c:246)
==477789==    by 0x138C71: z_impl_k_poll (poll.c:304)
==477789==    by 0x115E73: k_poll (kernel.h:1460)
==477789==    by 0x115E73: zsock_poll_internal (sockets.c:1789)
==477789==    by 0x1164EB: z_impl_zsock_select (sockets_select.c:141)
==477789==    by 0x111CC8: zsock_select (socket_select.h:39)
==477789==    by 0x111CC8: _posix_zephyr_main (main.c:37)
==477789==    by 0x135F00: bg_thread_main (init.c:377)
==477789==    by 0x11235F: z_thread_entry (thread_entry.c:48)
==477789==    by 0x11AD99: posix_arch_thread_entry (thread.c:56)
==477789==    by 0x11AFA9: posix_thread_starter (posix_core.c:293)
==477789==    by 0x496E0B3: start_thread (pthread_create.c:477)
==477789==    by 0x4A88385: clone (clone.S:108)
==477789==  Address 0x44 is not stack'd, malloc'd or (recently) free'd
==477789== 
==477789== 
==477789== Process terminating with default action of signal 11 (SIGSEGV)
==477789==  Access not within mapped region at address 0x44
==477789==    at 0x138A6F: is_condition_met (poll.c:67)
==477789==    by 0x138A6F: register_events (poll.c:246)
==477789==    by 0x138C71: z_impl_k_poll (poll.c:304)
==477789==    by 0x115E73: k_poll (kernel.h:1460)
==477789==    by 0x115E73: zsock_poll_internal (sockets.c:1789)
==477789==    by 0x1164EB: z_impl_zsock_select (sockets_select.c:141)
==477789==    by 0x111CC8: zsock_select (socket_select.h:39)
==477789==    by 0x111CC8: _posix_zephyr_main (main.c:37)
==477789==    by 0x135F00: bg_thread_main (init.c:377)
==477789==    by 0x11235F: z_thread_entry (thread_entry.c:48)
==477789==    by 0x11AD99: posix_arch_thread_entry (thread.c:56)
==477789==    by 0x11AFA9: posix_thread_starter (posix_core.c:293)
==477789==    by 0x496E0B3: start_thread (pthread_create.c:477)
==477789==    by 0x4A88385: clone (clone.S:108)
==477789==  If you believe this happened as a result of a stack
==477789==  overflow in your program's main thread (unlikely but
==477789==  possible), you can try to increase the size of the
==477789==  main thread stack using the --main-stacksize= flag.
==477789==  The main thread stack size used in this run was 8388608.
==477789== 
==477789== HEAP SUMMARY:
==477789==     in use at exit: 3,008 bytes in 11 blocks
==477789==   total heap usage: 17 allocs, 6 frees, 5,074 bytes allocated
==477789== 
==477789== LEAK SUMMARY:
==477789==    definitely lost: 0 bytes in 0 blocks
==477789==    indirectly lost: 0 bytes in 0 blocks
==477789==      possibly lost: 1,088 bytes in 8 blocks
==477789==    still reachable: 1,920 bytes in 3 blocks
==477789==         suppressed: 0 bytes in 0 blocks
==477789== Rerun with --leak-check=full to see details of leaked memory
==477789== 
==477789== For lists of detected and suppressed errors, rerun with: -s
==477789== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
Segmentation fault
```

**Environment**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit: https://github.com/zephyrproject-rtos/zephyr/commit/433a85e7bf32ffc77ba194bb903bcca84c44c047

jukkar, 2023-10-16:
This I could replicate, thanks for the good example program.

jukkar, 2023-10-16:
@chih-hung-yu thanks for the report. Your workaround is working fine so I used it in the fix. Also added unit tests in order to check the return codes are correct.

jukkar, 2023-10-17:
@chih-hung-yu Robert and I have been pondering how to fix this and the current proposal can be found at #63984. Could you try it and see if it fixes your issue. Note that the code does not return ECONNRESET to the application and you would need to check the error condition from either select() exceptfs set or in poll() POLLERR which is returned to the caller. See for example the unit test in that PR how user could detect the connection closure.
The idea of this proposal is that the `ctx->tcp` pointer is valid during the socket() -> close() calls, so there should never be a null tcp pointer and the null access issue should be handled properly.

chih-hung-yu, 2023-10-18:
I modified my sample code to select() exceptfds and select() did return with FD_ISSET(fd, &efds) immediately when RST is received or when connection timeout. Though for the application it is not a portable method to use exceptfds to detect connection failure, but Zephyr does set ZSOCK_POLLERR event to signal socket error when sock_is_error(ctx), and in this way we do not know the error reason.

According to POSIX, 'exceptfds' is used to signal the availability of out-of-band (OOB) data, which is not supported by Zephyr currently. 
http://man.he.net/man2/select_tut
```
exceptfds
              This set is watched for "exceptional conditions".  In  practice,
              only  one such exceptional condition is common: the availability
              of out-of-band (OOB) data for reading from a  TCP  socket.  
```

jukkar, 2023-10-18:
I have to agree with you that we should behave more like Linux with the select() call.

commit fd1c226c (2023-10-24): net: tcp: Increment ref count in initial SYN

Increase reference count already when initial SYN is sent.
This way the tcp pointer in net_context is fully valid for
the duration of the connection.

Fixes #63952
commit f9f78b44 (2023-11-06): net: tcp: Increment ref count in initial SYN

Increase reference count already when initial SYN is sent.
This way the tcp pointer in net_context is fully valid for
the duration of the connection.

Fixes #63952

(cherry picked from commit fd1c226cd806ec8c8250bddc5def11a114d712be)
closed 2023-10-24 by cfriedt