zephyrproject-rtos/zephyr #63966: net: ip: Does not reset expired TCP session

chih-hung-yu, 2023-10-16:
**Describe the bug**
A TCP client can be programmed to use a fixed TCP source port. When we perform a power cycle to a target running the fixed-source-port TCP client that has already connected to the TCP server, the target will make a new TCP session with the same TCP source port, but the old TCP session is still kept by the TCP server which replies with the old ACK number.

<B>For Linux TCP client</B>, RST is issued if the TCP server ACKed unseen segment, and then the TCP server can accept new SYN with the same TCP source port.
(At packet No. 93, the TCP client restarts and sends a new SYN.)
[image]

If the Linux TCP client issues random-source-port SYN after a power cycle and hence the new connection is established immediately, the TCP server still issues TCP Keep-Alive for the old session, and the Linux TCP client will reset the old connection when it receives  TCP Keep-Alive.
(At packet No. 107, the TCP client restarts and sends a new SYN.)
[image]

<B>For Zephyr TCP client</B>, ACK is ignored if the TCP server ACKed unseen segment, and the Zephyr TCP client keeps retransmitting SYN, the connection cannot be established until the old TCP session kept by the TCP server reaches TCP-Keep-Alive timeout.
(At packet No. 49, the TCP client restarts and sends a new SYN.)
[image]

**To Reproduce**
1. Power the Zephyr TCP client up to make a TCP connection to the TCP server.
2. Power the Zephyr TCP client down, and then power the Zephyr TCP client up again.
```
#include <zephyr/net/socket.h>

int main(void)
{
    int fd, ret;
    struct sockaddr_in sin, servaddr;
    struct in_addr inaddr;

    fd = zsock_socket(AF_INET, SOCK_STREAM, 0);

    sin.sin_family = AF_INET;
    sin.sin_addr.s_addr = INADDR_ANY;
    sin.sin_port = htons(4044);
    zsock_bind(fd, (const struct sockaddr *)&sin, sizeof(sin));

    inet_pton(AF_INET, "10.123.4.223", (void *)&inaddr);

    servaddr.sin_family = AF_INET;
    servaddr.sin_addr = inaddr;
    servaddr.sin_port = htons(502);

    ret = zsock_connect(fd, (const struct sockaddr *)&servaddr, sizeof(servaddr));
    printk("zsock_connect()=%d\n", ret);

    return 0;
}
```

**Expected behavior**
Zephyr will reset the expired TCP session kept by the peer when the peer replies with an unexpected ACK number.

**Environment**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit: https://github.com/zephyrproject-rtos/zephyr/commit/433a85e7bf32ffc77ba194bb903bcca84c44c047

rlubos, 2023-10-17:
@chih-hung-yu Please give https://github.com/zephyrproject-rtos/zephyr/pull/64029 a try.

chih-hung-yu, 2023-10-18:
> @chih-hung-yu Please give #64029 a try.

 For fixed-source-port Zephyr TCP client, the behaviour is now the same as Linux TCP client.
[image]

 For random-source-port Zephyr TCP client, RST is sent for unexpected TCP Keep-Alive packet, and an extra ICMP Port unreachable is also sent.
[image]

rlubos, 2023-10-18:
> For random-source-port Zephyr TCP client, RST is sent for unexpected TCP Keep-Alive packet, and an extra ICMP Port unreachable is also sent.

Zephyr stack currently sends ICMP Destination Unreachable for any IP-based protocol, so it indeed duplicates with the RST behavior introduced. What's the expected behavior for TCP, shall it reply with RST only, and forfeit error reporting at the ICMP level?

Also @jukkar, regarding the Kconfig option to disable RST sending for unbound ports - shall we still send ICMP error in case RST's are disabled, or just silently ignore the packet?

In any case, adjusting this behavior is just a matter of minor tweaks in the code at this point, we just need to agree on the expected behavior.

PR #64029 (2023-10-20): net: tcp: Send RST reply for unexpected TCP packets 
This PR improves the handling of unexpected packets within Zephyr's TCP stack. The TCP stack will now send RST packet in a response to unexpected TCP packets, to address issues mostly related to ungraceful TCP connection shutdown (for example due to reboot):
1) Invalid ACK value received during handshake (connection still open
   on the peer side),
2) Unexpected data packet on a listening port (accepted connection
   closed),
3) SYN received on a closed port.

For this, a new helper function was introduced which allows to send an RST packet  [... cut]
closed 2023-10-20 by carlescufi