zephyrproject-rtos/zephyr #63972: Allow import of optional projects from manifest file

pdgendt, 2023-10-16:
**Is your feature request related to a problem? Please describe.**
When using out-of-tree applications, I would like to add optional projects from the application's manifest file.
For example, I use `nanopb` which is an optional package in `zephyr`.

**Describe the solution you'd like**
It would be nice to support `project-filter: [+nanopb]` directly from the manifest file.

Currently I have to configure a workspace with:

```shell
west config manifest.project-filter -- +nanopb
```

While my `west.yml` already has it in the `name-allowlist`:

```yaml
manifest:
  self:
    path: app

  remotes:
    - name: upstream
      url-base: https://github.com/zephyrproject-rtos

  projects:
    - name: zephyr
      revision: main
      remote: upstream
      import:
        name-allowlist:
          - cmsis
          - hal_atmel
          - nanopb
          - picolibc
```

**Describe alternatives you've considered**
I can add `group-filter: [+optional]` to the manifest file, but this can lead to more projects being updated than necessary.

nordicjm, 2023-10-16:
>  I can add group-filter: [+optional] to the manifest file, but this can lead to more projects being updated than necessary.

How? You have an allow list

pdgendt, 2023-10-16:
> How? You have an allow list

In this particular case, yes. But I think it would be nice to have using `import: true` too.