zephyrproject-rtos/zephyr #63977: Windows GSG: ruamel.yaml wheel not available for Python 3.12

wmamills, 2023-10-16:
I am following the Getting Started Guide for v3.5-rc3 on a fresh Windows machine. 
As of Oct 15, I get python 3.12 when using the GSG instructions. (choco line says "python").  
At the moment there are no wheel candidates for ruamel.yaml for 3.12 on WIndows.
The machine tries to build the native components of that package but cannot as the machine does not have the expected compiler.

If I change the choco line to install "python311" instead this problem goes away.
(Along with the windows-curses issues that is already reported in https://github.com/zephyrproject-rtos/windows-curses/pull/44 and expected to be fixed before the v3.5 release.)

I assume you will arrange to have a ruamel.yaml wheel available or change the GSG to use phython311.
(Instructing the user to install the native compiler seems like overkill for this problem.)

henrikbrixandersen, 2023-10-25:
This is the second step for addressing https://github.com/zephyrproject-rtos/zephyr/issues/63977.

kartben, 2023-10-25:
Thanks for reporting this @wmamills. We definitely want the Getting Started to work out of the box so your feedback is very helpful. As you can see in recent comments, we will be looking at hardening our CI so that to get a better chance to catch potential issue with 3.12. 
In the meantime, I think the Windows wheels are now available (https://pypi.org/project/ruamel.yaml.clib/#files) -- would you mind trying again and reporting back?

Thanks!

henrikbrixandersen, 2023-10-26:
We now test against Python 3.12 in our Continuous Integration on both MS Windows, macos, and Linux.

stephanosio, 2023-11-01:
[windows-curses 2.3.2 has been released with the wheels for Python 3.12.](https://github.com/zephyrproject-rtos/windows-curses/releases/tag/v2.3.2)

closed 2023-11-01 by stephanosio