zephyrproject-rtos/zephyr #63980: precision is limited on lsm6dsl sensor measurements.

Alessio-Parmeggiani, 2023-10-16:
I noticed that in the driver for the lsm6dsl sensor the precision of the measurement is limited to 3 decimal numbers, when the datasheet specifies that it could be much higher. In fact the sensitivity is 0.004375 dps/LSB or 0.0000763582 rad/LSB.

In particular in the function to convert the measurement to the sensor_value struct the following function is applied:
[code](https://github.com/zephyrproject-rtos/zephyr/blob/137097f5c3153cefb9f585aa7b2d88bf09a384aa/drivers/sensor/lsm6dsl/lsm6dsl.c#L517C1-L527C2)
`val->val2 = (((int32_t)(dval * 1000)) % 1000) * 1000;` limiting the value to 3 decimal digits.

The same applies also to the acceleration measurements because a similar function is used. [Code](https://github.com/zephyrproject-rtos/zephyr/blob/137097f5c3153cefb9f585aa7b2d88bf09a384aa/drivers/sensor/lsm6dsl/lsm6dsl.c#L467C1-L478C2)

**Possible solution**
A possible solution that would allow 6 decimal digits is the following:
`val->val2= (((int32_t)(dval * 100000)) % 100000);`

**Other minor related precision error**
In the same conversion function there is also a conversion from degrees to radians but the constant used to perform the conversion doesn't have enough decimal digits to allow a precise conversion. For example I measured a sensitivity of 0.00893 instead of the nominal 0.00875 on the gyroscope. By using the measurement in degree I get the nominal value.

avisconti, 2024-03-21:
@Alessio-Parmeggiani Sorry for the long delay in asnwering. If you are still interested can you possibly submit a PR for this enhancement? Thanks

Alessio-Parmeggiani, 2024-03-26:
> @Alessio-Parmeggiani Sorry for the long delay in asnwering. If you are still interested can you possibly submit a PR for this enhancement? Thanks

Sure, here it is #70718

PR #72367 (2024-05-08): drivers/sensor: lsm6dsl: use sensor APIs to convert to units
Make use of sensor_ug_to_ms2() and sensor_10udegrees_to_rad() APIs to convert from raw values into proper units (i.e. m/s^2 for accel and rad/s for gyro). It also improves measurement precision.

Fixes #63980

Mimic what already done for other sensors in #71223
Replaces: #70718
commit dd42501a (2024-05-08): drivers/sensor: lsm6dsl: use sensor APIs to convert to units

Make use of sensor_ug_to_ms2() and sensor_10udegrees_to_rad() APIs
to convert from raw values into proper units (i.e. m/s^2 for accel
and rad/s for gyro). It also improves measurement precision.

Fixes #63980
closed 2024-05-08 by nashif