zephyrproject-rtos/zephyr #64021: docs: rv32m1_vega: reset button image is unreadable

p-woj, 2023-10-17:
**Describe the bug**
In the [one-time board setup](https://docs.zephyrproject.org/latest/boards/riscv/rv32m1_vega/doc/index.html#one-time-board-setup-for-booting-ri5cy-or-zero-riscy) section, under the line saying

> The reset button is at top left, as shown in the following figure.

there is an image ([ri5cy_boot.jpg](https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/riscv/rv32m1_vega/doc/ri5cy_boot.jpg)) embedded which is 4x2 pixels and thus not readable.

This was introduced in https://github.com/zephyrproject-rtos/zephyr/pull/49024. It might affect other images, I haven't verified this - only noticed it on this one.

**To Reproduce**
Visit the [one-time board setup](https://docs.zephyrproject.org/latest/boards/riscv/rv32m1_vega/doc/index.html#one-time-board-setup-for-booting-ri5cy-or-zero-riscy) section of the VEGAboard docs and notice that the image showing the reset button is not readable.

**Expected behavior**
The image is readable.

**Impact**
Have to consult the [last good version](https://github.com/zephyrproject-rtos/zephyr/blob/1dc4b1dd2f179ca1153f4c0412356e1ae79eb055/boards/riscv/rv32m1_vega/doc/ri5cy_boot.jpg) in the history.

Pinging @kartben - apologies if it's unnecessary.

kartben, 2023-10-17:
Ouch. Thanks for digging in the history behind this one. Hopefully it's the only problematic one (as per `find . -iname "*.jpg" -size -2k`). Would you be able to submit a PR restoring the previous image?
Thanks!

p-woj, 2023-10-17:
Sure, I opened #64026 which restores the version as of 1dc4b1dd2f179ca1153f4c0412356e1ae79eb055 which is the last readable one

PR #64026 (2023-10-17): doc: riscv: rv32m1_vega: Restore reset button image
This image was accidentally shrunk too much in e81e92dbb9092a3d67ec7e17f0c2190239635a50. Restore the previous version.

Fixes #64021
commit 6e39775e (2023-10-17): doc: riscv: rv32m1_vega: Restore reset button image

This image was accidentally shrunk too much in
e81e92dbb9092a3d67ec7e17f0c2190239635a50.
Restore the previous version.

Fixes #64021
closed 2023-10-17 by jhedberg