zephyrproject-rtos/zephyr #64023: hello_world no longer usable on USB CDC interface devices

nordicjm, 2023-10-17:
**Describe the bug**
Due to how USB CDC works, it needs to enumerate before a PC can open the port, which takes time. As a result, it needs to be possible to delay the output of the hello world application on these devices so that this is visible, this (previously) was accomplished by enabling logging with a delay and forwarding printk to logging. However, with commit f623fcf3987cb35d73e1e5537822da5af865269f this is now broken, because hello_world uses printf calls which do not get routed to the log functions.

**To Reproduce**
Build hello_world with: `cmake -GNinja -DBOARD=nrf52840dk_nrf52840 -DSNIPPET=cdc-acm-console -DCONFIG_USB_DEVICE_INITIALIZE_AT_BOOT=y -DLOG=y -DLOG_PROCESS_THREAD_STARTUP_DELAY_MS=3000` then flash to an nrf52840dk and boot the board with dedicated USB plugged in - hello world is not printed

**Expected behavior**
hello world to be printed

**Impact**
Sample is useless for USB CDC boards

**Environment (please complete the following information):**
 - Commit SHA or Version used: 9e9f439522c1f0f0ef58d562283f7c979954bb86

57300, 2023-10-19:
> As a result, it needs to be possible to delay the output of the hello world application on these devices so that this is visible, this (previously) was accomplished by enabling logging with a delay and forwarding printk to logging.

Would adding a `CONFIG_BOOT_DELAY` work just as well? I see that the USB device can be initialized in `POST_KERNEL`, so the boot delay would come after that.

nordicjm, 2023-10-19:
> Would adding a `CONFIG_BOOT_DELAY` work just as well? I see that the USB device can be initialized in `POST_KERNEL`, so the boot delay would come after that.

No, there already is a delay for USB CDC/logging but that is ignored because `printf()` does not use the logging system, it uses poll out directly. And if you delayed the whole application's startup, it would be the same problem, the application would start, wait, then init so USB CDC would start and the `printf()` call would output before the PC even detected you had a USB CDC device connected

gmarull, 2023-10-23:
> No, there already is a delay for USB CDC/logging but that is ignored because `printf()` does not use the logging system, it uses poll out directly. And if you delayed the whole application's [...]

could printf have a path via logging (enabled using a Kconfig option)?

closed 2023-11-13 by carlescufi