zephyrproject-rtos/zephyr #64030: fs_open error log

clamattia, 2023-10-17:
The documentation suggests to use `fs_open` with `flags=0` and presumably `fs_opendir` to detect existence/non-existence of a file.
In case the file does not exists the `fs.c` logs an error: [LOG_ERR("file open error (%d)", rc)](https://github.com/zephyrproject-rtos/zephyr/blob/b732c6d564e542fd25b63041ec713346e01711aa/subsys/fs/fs.c#L166). Since this is the intended outcome it surely is not an error.

There will be false positives in the log, if as suggested by the documentation `fs_open` is used to detect existence/non-existence of a file.

Ideally `fs_` would:
- [ ] Either not log an error if the file/directory does not exist in such circumstance
- [ ] Or provide an additional API function to detect existence/non-existence of files/directories
- [ ] If instead an alternative exists (for example `fs_stat`), the documentation may be updated to say so clearly.

I am opening the issue, because I do not know how to filter out real errors from the false positive.
I would imagine something like:
```
	rc = mp->fs->open(zfp, file_name, flags);
        bool false_positive = (rc == NOT_EXISTS && flags == 0);
	if (rc < 0 && !false_positive) {
		LOG_ERR("file open error (%d)", rc);
        }
	if (rc < 0) {
		zfp->mp = NULL;
		return rc;
	}
```

- [ ] Ideally something similar could be done for `fs_opendir`

Thanks in advance.

nordicjm, 2023-10-17:
`fs_open` shouldn't be used for that, use `fs_stat`: https://docs.zephyrproject.org/latest/services/file_system/index.html#c.fs_stat

clamattia, 2023-10-17:
> `fs_open` shouldn't be used for that, use `fs_stat`: https://docs.zephyrproject.org/latest/services/file_system/index.html#c.fs_stat

Thanks for the quick reply. Maybe the documentation (`If flags are set to 0 the function will attempt to open an existing file with no read/write access; this may be used to e.g. check if the file exists.`) could be adjusted then, to reference `fs_stat` instead?

Maybe the documentation of `fs_stat` could explicitly mention existence checks?

de-nordic, 2023-10-17:
> Thanks for the quick reply. Maybe the documentation (`If flags are set to 0 the function will attempt to open an existing file with no read/write access; this may be used to e.g. check if the file [...]

Yeah, we should mention that it will log error.

clamattia, 2023-10-19:
Please feel free to change the area to `documentation` or let me know, if I should make a separate issue.

de-nordic, 2023-10-19:
> Please feel free to change the area to `documentation` or let me know, if I should make a separate issue.

I will change the doc, no need for another issue.

PR #64146 (2024-02-07): fs: Clarify description of fs_open when no access bits given
Added note warnign that fs_open should not be used for checking whether file exists or not and fs_stat should be used instead.

Fixes #64030
commit 07c1e49f (2024-02-07): fs: Clarify description of fs_open when no access bits given

Add warning that file opened without R/W flags will have no read/write
access.
Remove suggestion for using fs_open to check if file exists.
Clarify -ENOENT return reason.

Fixes #64030
closed 2024-02-07 by dleach02