zephyrproject-rtos/zephyr #64038: Zephyr application ABI

mkschreder, 2023-10-17:
**Is your feature request related to a problem? Please describe.**

Currently Zephyr has source code compatibility with applications that use zephyr, but this is insufficient to be able to define bindings on top of Zephyr for supporting applications written in other languages that would run on zephyr. 

While Zephyr does define many system calls, there currently does not seem to exist a well defined application interface that does not require the application to be compiled as part of the same build process as the rest of the system. 

This problem occurs when trying to interface rust and also erlang atomvm to zephyr. 

**Describe the solution you'd like**

I would like a well defined application interface similar to linux system calls but possibly more "flat" to avoid incuring unnecessary overhead by too many high level abstractions.

Not sure if aiming for full posix compliance would solve enough issues. It would at least be good to be able to link seamlessly to code that is built separately outside of zephyr such as when building rust code to object code that is then linked to zephyr. 

Possibly if there is some easy way to generate bindings for other languages solely by extracting and post processing system call definitions then this may be a workable solution. Not sure how to proceed here though so thankful for some suggestions. 

**Describe alternatives you've considered**

- Writing wrappers for some zephyr functions that are to be exposed to "application" code written in a different language: requires way too much work and doesn't work very well since some zephyr stuff is implemented using macros.

**Additional context**

- Rust Zephyr: https://github.com/tylerwhall/zephyr-rust/issues/37#issuecomment-1763549397
- AtomVM Erlang runtime: https://github.com/atomvm/AtomVM/issues/802

pillo79, 2023-10-17:
#57086 very recently added the [Linkable Loadable Extensions (LLEXT)](https://docs.zephyrproject.org/latest/services/llext/index.html) subsystem, and it is now in the just released Zephyr 3.5 (albeit still at a very experimental stage). 
Would this eventually fit your use case?

teburd, 2023-10-18:
Even #57086 would also benefit from a stable ABI at least to some extent so that modules could work across some subset of zephyr versions.

I don't think we can assume Zephyr will have a stable syscall interface at any point in the future as its not the primary interface. We would need instead to have C function and object symbols remain stable. That is a difficult task.

mkschreder, 2023-10-18:
@pillo79 looks very promising. Thanks for mentioning that.