zephyrproject-rtos/zephyr #64069: on pinctrl and gpio relationship

yf13, 2023-10-18:
**Is your enhancement proposal related to a problem? Please describe.**

The [Pin control vs. GPIO[¶](https://docs.zephyrproject.org/latest/hardware/pinctrl/index.html#pin-control-vs-gpio) might missed an important point, it says now like:

` Some functionality covered by a pin controller driver overlaps with GPIO drivers. For example, pull-up/down resistors can usually be enabled by both the pin control driver and the GPIO driver. In Zephyr context, the pin control driver purpose is to perform peripheral signal multiplexing and configuration of other pin parameters required for the correct operation of that peripheral. Therefore, the main users of the pin control driver are SoC peripherals. In contrast, GPIO drivers are for general purpose control of a pin, that is, when its logic level is read or controlled manually.`

**Describe the solution you'd like**

I am wondering if  we can say that GPIO is just another type of peripheral like those I2C or SPI or UART controllers. They all compete for pins so that to reach external world of the chip. From this regard, Zephyr pinctrl class models the multiplexing and pin config functions and it has little overlapping with GPIO controller class.

In general, the high level overview of pinctrl is excellant documentation practice. Zephyr probably should require all drivers have such conceptual overview document so that users can quickly learn if a driver is of interest or not.

henrikbrixandersen, 2023-10-18:
There is a difference between GPIO and other peripherals when it comes to pinctrl. GPIO drivers are responsible for setting up pins (including pin multiplexing) for GPIO usage _without_ devicetree pinctrl nodes, whereas other peripherals rely on devicetree pinctrl nodes for describing the required multiplexing, pull-ups, -downs, etc.

This is what the paragraph quoted above tries to convey.

henrikbrixandersen, 2023-10-18:
CC: @gmarull