zephyrproject-rtos/zephyr #64098: imgtool 1.9.0 Generates Invalid Hex File

legoabram, 2023-10-18:
**Describe the bug**
When using `imgtool` 1.9.0, the generated `zephyr.signed.confirmed.hex` file places the `image_ok` indicator in the wrong place when using the  `CONFIG_MCUBOOT_GENERATE_CONFIRMED_IMAGE` configuration. The bin is built correctly. Upgrading to `imgtool` 1.10.0 resolves the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Force install the broken version `pip install 'imgtool==1.9.0'`
2. Set `CONFIG_MCUBOOT_GENERATE_CONFIRMED_IMAGE=y`
3. Run Build
4. See incorrect location for `01` at `CF20`
```
...
:02000004080FE3
:10CF200001FFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0F
:10CF3000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF01
:10FFF00020002DE15D29410B8D77679C110F1F8A31
:00000001FF
```

**Expected behavior**
5. Force install latest `pip install 'imgtool>1.9.0'` or `pip install 'imgtool==1.10.0'`
6. Rerun build
7. See correct location for `01` at `FFC0`
```
...
:02000004080FE3
:10FFC00001FFFFFFFFFFFFFFFFFFFFFFFFFFFFFF3F
:10FFD000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF31
:10FFF00020002DE15D29410B8D77679C110F1F8A31
:00000001FF
```

**Impact**
While not a showstopper, I spent way too long trying to figure out why OTA wasn't working since I was using the `.hex` files to program the chip in my debugger. It was causing a lockup where the image would be reverted on the next run, and is not recoverable.

**Logs and console output**
N/A

**Environment (please complete the following information):**
 - OS:  Windows 11
 - Toolchain: GCC arm-none-eabi 12.2.1
 - Commit: Zephyr 3.5.0-rc1

bbilas, 2023-10-18:
It looks the same as https://github.com/zephyrproject-rtos/zephyr/issues/46414

legoabram, 2023-10-18:
> It looks the same as #46414

Yep. That's when it was fixed in the imgtool branch. But the fix (https://github.com/mcu-tools/mcuboot/pull/1445) was merged after the Mar 2, 2022 release of 1.9.0.

PR #64099 (2024-02-12): scripts/requirements: bump imgtool to 2.0.0
Resolves incorrectly located `image_ok` tag in generated hex files when  CONFIG_MCUBOOT_GENERATE_CONFIRMED_IMAGE is used.

Fixes #64098
commit 5dec4fca (2024-02-12): scripts/requirements: bump imgtool to 2.0.0

Resolves incorrectly located `image_ok` tag in generated hex files when
CONFIG_MCUBOOT_GENERATE_CONFIRMED_IMAGE is used.

Fixes #64098
PR #68905 (2024-03-05): [Backport v3.5-branch] scripts/requirements: bump imgtool to 2.0.0
Backport 5dec4fcab8bee81d1557fd6693c5f536e99060af from #64099.

Fixes #64098
commit 0de70854 (2024-03-05): scripts/requirements: bump imgtool to 2.0.0

Resolves incorrectly located `image_ok` tag in generated hex files when
CONFIG_MCUBOOT_GENERATE_CONFIRMED_IMAGE is used.

Fixes #64098

(cherry picked from commit 5dec4fcab8bee81d1557fd6693c5f536e99060af)
closed 2024-02-12 by henrikbrixandersen