zephyrproject-rtos/zephyr #64110: SW ISR table gets mangled when enabling SHARED_INTERRUPTS

davedesro, 2023-10-19:
## Describe the bug
tl;dr `zephyr_pre0.elf` does not have `z_shared_isr()` in its image, but should. See bottom for my work-around. Does the linker need modification?  

When enabling the new [shared interrupts feature](https://github.com/zephyrproject-rtos/zephyr/pull/61422) (Great feature!! long overdue!! thank you!!) in [v3.5.0-rc3](https://github.com/zephyrproject-rtos/zephyr/tree/v3.5.0-rc3), The interrupt pointers set in `build/zephyr/isr_tables.c` (generated by [gen_isr_tables.py](https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/build/gen_isr_tables.py)) are wrong.

After a bit of `objdump`'ing, I found that the linker for `build/zephyr/zephyr_pre0.elf` removes [`z_shared_isr()`](https://github.com/zephyrproject-rtos/zephyr/blob/v3.5.0-rc3/arch/common/shared_irq.c#L17) function since it's not used (due to a combo of `-ffunction-sections` at compile time, and `-Wl,--gc-sections` at link time). It's not used because `zephyr_pre0.elf` does not have `isr_tables.c` in it.

... however, `build/zephyr/zephyr.elf` _does_ have `build/zephyr/isr_tables.c`, so `z_shared_isr()` is part of the final image.

This causes a shift of all ISR addresses when comparing `zephyr_pre0.elf` to `zephyr.elf`.  When loaded, hang is bound to happen on the first couple of interrupts.
 

## To Reproduce
Using the environment below,
1. set `CONFIG_SHARED_INTERRUPTS=y` and `CONFIG_ADC_STM32_SHARED_IRQ=n`
2. On NUCLEO_G474RE, enable `&adc1` and `&adc2` to exercise the new feature.
3. flash it. You should see a hang pretty quickly.. likely on the first interrupt.

## Impact
Without a fix, the feature is not usable at least with the setup below.

## Environment (please complete the following information):**
* Hardware: NUCLEO_G474RE
* SDK: 0.16.3
* Relevent config: CONFIG_SHARED_INTERRUPTS=y
* Zephyr tag v3.5.0-rc3
* Fedora Linux OS

## Additional context
I'm working around this issue by modifying the linker.

**Modify the linker to keep `z_shared_isr()` no matter what**
```
> cat app/shared_irq.ld                                                                                                                       
SECTION_PROLOGUE(.text.z_shared_isr,,)                                                                                                                                                                                                  
{
    KEEP(*(.text.z_shared_isr))
} GROUP_ROM_LINK_IN(0, FLASH)
```
**Add new `.ld` file to the build**
```
> git diff app/CMakeLists.txt
diff --git a/app/CMakeLists.txt b/app/CMakeLists.txt
index 38c15f6..0196245 100644
--- a/app/CMakeLists.txt
+++ b/app/CMakeLists.txt
@@ -9,3 +9,6 @@ add_dependencies(app app_version)

 add_subdirectory(src)
 add_subdirectory_ifdef(CONFIG_SHELL src/cli)
+
+zephyr_linker_sources(SECTIONS shared_irq.ld)
+
```

@LaurentiuM1234 @carlescufi

henrikbrixandersen, 2023-10-19:
CC: @LaurentiuM1234

davedesro, 2023-10-19:
Hi @LaurentiuM1234. I am happy for the PR.. and it couldn't have come at a better time for our project :)

Yes, I have a good understand why the address shift is happening.

1. `zephyr_pre0.elf` is linked. This does _not_ have the final interrupt vector table
2. `objcopy` is run to extract the `.intList` section, which contains all interrupts
3. `scripts/build/gen_isr_tables.py is run, output is `isr_tables.c`, which contains the final  "hw" vector table and "sw" vector table
4. `zephyr.elf` is linked including `isr_tables.c`

The problem is that the linker optimizes the `zephyr_pre0.elf` image by removing `z_shared_isr()` since the function is not used.. however, `zephyr.elf` _does_ use this function via the new `isr_tables.c` file. 

This difference causes a shift in all symbol addresses by the size of the `z_shared_isr()` function... `isr_tables.c` has pointers for `zephyr_pre0.elf`. The build system assumes the ISR addresses are the same between both `.elf` files, when in fact they are not.

Anyways.. @LaurentiuM1234  I will submit a PR. Besides yourself, lemme know who else to include.

thanks!

carlescufi, 2023-10-23:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/64152

closed 2023-10-23 by carlescufi

davedesro, 2023-10-23:
Hi @carlescufi I see 3.5.0 was released on Friday, but this got merged afterwards. 
Is this something that can be cherry picked for the `v3.5-branch`?

henrikbrixandersen, 2023-10-23:
> Hi @carlescufi I see 3.5.0 was released on Friday, but this got merged afterwards. Is this something that can be cherry picked for the `v3.5-branch`?

I have added the backport label for v3.5.x.