zephyrproject-rtos/zephyr #64120: Who is PDF documentation for?

fabiobaltieri, 2023-10-19:
## Introduction

We currently [generate the documentation](https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/workflows/doc-build.yml) in both HTML and PDF format. While the HTML version is actively used, built on every PR and pushed regularly, the PDF version is [not built on pull requests](https://github.com/zephyrproject-rtos/zephyr/pull/53004) and it occasionally breaks in a way that is hard to detect as the build step is [configured to not fail](https://github.com/zephyrproject-rtos/zephyr/pull/59430) to avoid blocking the HTML push which is considered more important.

### Problem description

The PDF build is a bit of a minor maintainence headache, the output does not really get validated as people normally refer to the HTML version, it breaks in weird ways (once it hits the limit of the GitHub machine specs generating it, now it cannot process emojis 😱).

The PDF version of the doc has been introduced in https://github.com/zephyrproject-rtos/zephyr/pull/37088, the PR does not specify the motivations for having it (could you add on this @gmarull?), and I'm having a hard time figuring out what gap a 3570 page PDF closes compared to the HTML version (I genuinly hope no one is actually printing it) and justify the maintainence effort to keep the build alive.

### Proposed change

Drop it! (yes I've already tried in https://github.com/zephyrproject-rtos/zephyr/pull/59404)

fabiobaltieri, 2023-10-19:
cc @carlescufi @gmarull @kartben @nashif

nordicjm, 2023-10-19:
Probably people that want to use it offline or from things like a network drive, far faster to host a pdf on a server somewhere that people can access in a PDF reader or web browser to use - it is lazy load. The html documentation on the other hand has thousands of files, not so good for a small local server and it requires a web server i.e. it does not work if you just open the index file in a web browser

fabiobaltieri, 2023-10-19:
> The html documentation on the other hand has thousands of files, not so good for a small local server and it requires a web server i.e. it does not work if you just open the index file in a web browser

You can download the html documentation artefacts (https://github.com/zephyrproject-rtos/zephyr/suites/17224064730/artifacts/984313781), extract it locally and open the index file on a browser with no web server. Looks like it works just fine to me, how about we upload that instead of the pdf and replace the pdf link with one to download that file?

nordicjm, 2023-10-19:
> Looks like it works just fine to me

If it does, you should worry about your web browser's (lack of) security. Try using the search feature

fabiobaltieri, 2023-10-19:
> If it does, you should worry about your web browser's (lack of) security. Try using the search feature

Oh, the missing result snippets? Fair enough, at least you get the page links though.

gmarull, 2023-10-19:
I vote to drop PDF, it's unusable, and I doubt there'll be anybody with enough courage to fix it.

nordicjm, 2023-10-19:
I would suggest sending an email on the users and dev mailing lists to ask if anyone uses it since that'll reach a far wider audience that this issue

nashif, 2023-10-19:
Déjà vu https://github.com/zephyrproject-rtos/zephyr/pull/59404#pullrequestreview-1486947001

carlescufi, 2023-10-23:
I am quite sure that @dcpleung uses the .pdf.

fabiobaltieri, 2023-10-23:
> Nowadays I just grep and read RST directly. :D

Yeah same, `git grep`... anyway

> Though, when PDF creation was first introduced (IIRC, 5+ years ago), it was about:
> ...

Ok, I was thinking about uploading a zip file of the snapshot instead of the pdf, so you would still have a downloadable offline copy pinned to every version. How does that sound? It's still many files instead of a single file but the rest should be covered.

kartben, 2023-10-23:
> Ok, I was thinking about uploading a zip file of the snapshot instead of the pdf, so you would still have a downloadable offline copy pinned to every version. How does that sound? It's still many files instead of a single file but the rest should be covered.

Re: many files, see #63923 as maybe another option

dcpleung, 2023-10-23:
I was thinking the other day whether we could do HTML to PDF instead of via LaTeX. This should avoid lots of issues with RST->LaTeX->PDF.

fabiobaltieri, 2023-10-23:
> I was thinking the other day whether we could do HTML to PDF instead of via LaTeX. This should avoid lots of issues with RST->LaTeX->PDF.

Is that a thing? Would be surprised if ou could get anything even remotely usable by passing through the html files.

dcpleung, 2023-10-23:
> > I was thinking the other day whether we could do HTML to PDF instead of via LaTeX. This should avoid lots of issues with RST->LaTeX->PDF.
> Is that a thing? Would be surprised if ou could get anything even remotely usable by passing through the html files.

You can probably do it via CUPS printing to a virtual PDF printer. Presentation is probably not great but should be usable IF you really want a PDF.

fabiobaltieri, 2023-10-23:
> You can probably do it via CUPS printing to a virtual PDF printer. Presentation is probably not great but should be usable IF you really want a PDF.

Well I'm trying to answer that "if". I'll send an email in the next few days, see if anyone is actually depending hard on it.

kartben, 2023-10-23:
FWIW the PDF is actually not in a terrible shape these days, so I'm not sure why we would even need to explore a CUPS hack, quite frankly. Yes, it's huge, but it looks as it's by design, since it's meant to capture as much as possible of what one might want to access and search offline. And it is indeed quite "searchable".
There are a couple dozen very broken SVG images, mainly ones drawn using draw.io, but this can be fixed with some time.

The real outstanding issue IMO is that we should fix CI so that folks (starting with me, I guess) get notified when PDF build fails, rather than finding out sometimes weeks later.

fabiobaltieri, 2023-10-23:
> The real outstanding issue IMO is that we should fix CI so that folks (starting with me, I guess) get notified when PDF build fails, rather than finding out sometimes weeks later.

Sure we could throw people at the problem, but it would also be nice to have one less moving part to think about, no point maintaining it if we can replace it with downloadable archive of the html doc.

scapri, 2023-11-01:
I had no idea there was a pdf of the documentation!  I've just downloaded the 3.4.0 PDF, and it is a thing of beauty!  This is so nice - a single file that I can copy from my desktop to my laptop quickly, and be able to have all of the documentation available.  Searching and clicking on the pdf links is so very fast and convenient.  And to have a single pdf file so that I can know exactly which version I'm searching (for various customers, we use different versions of Zephyr depending on the phase of the project) is perfect - somehow, I often end up in the wrong version when searching online.

Ok, I'm sorry - I'm sure your intent in creating this issue was not to get more people using the PDF, but now that I know this is available, I would be so sad to lose it now.  I can understand the desire to avoid maintaining complexity if there is no value in it, but if my 2 cents/€0.02 mean anything (and maybe it doesn't mean anything), I would love to see the PDF stay around.

fabiobaltieri, 2023-11-06:
Hi! Been a little over a week since https://lists.zephyrproject.org/g/devel/topic/does_anyone_use_the_pdf/102295499.

Thanks everyone for the feedback, if I can summarize: few people prefer the format, it's convenient to have it in a single file and use it on PDF reading device rather than a browser. Also: it looks pretty, and multi screen Linux support is still a mess (I agree with both).

Interesting to learn that Noridc dropped the PDF generation, was not aware of that.

Apart from convenience and preferences, the only fundamental blocker I think I've seen mentioned is using the file on an eBook reader. That's the point I was mostly interested to, I should have probably rephrased the question as: "what can you do with a PDF that you can't with an HTML archive".

With that in mind, I think it's fair to keep it for now and re-evaluate next time the generation fails. In the interim we need to figure how to get some better visibility into the status of the PDF generation workflow so at least we catch breakages as early as possible.

closed 2023-11-06 by fabiobaltieri