zephyrproject-rtos/zephyr #64148: Code relocation for functions

meijemac, 2023-10-19:
**Is your enhancement proposal related to a problem? Please describe.**
I have some code that needs to be relocated to another memory section. I've found the **zephyr_code_relocate** sample application, which seems to do the job. However, I do not need to reallocate the complete file, but only a few functions.

- We do not want to change the file structure. So moving the functions to be reallocated to another file is not an option.
- We also use source code not owned by us. So also in this case moving the functions to be reallocated to another file is not an option.

**Describe the solution you'd like**
I would like to have something similar to the current solution:

zephyr_code_relocate(FILES src/*.c LOCATION SRAM2)
zephyr_code_relocate(FUNC foo FILE src/foo.c LOCATION SRAM2)

**Describe alternatives you've considered**
NA

**Additional context**
NA

tejlmand, 2023-10-25:
the problem with purely relocating a function is that you have no knowledge of how the function has been optimized and placed by the compiler.

In this case you would need to compile the source using `-ffunction-sections` (gcc), but in that case care must be taken to only apply this on the source files in question.

From GCC manual: https://gcc.gnu.org/onlinedocs/gcc-13.2.0/gcc/Optimize-Options.html
> -ffunction-sections
> -fdata-sections
>
>    Place each function or data item into its own section in the output file if the target supports arbitrary sections. The name of the function or the name of the data item determines the section’s name in the output file.
>
>    Use these options on systems where the linker can perform optimizations to improve locality of reference in the instruction space. Most systems using the ELF object format have linkers with such optimizations. On AIX, the linker rearranges sections (CSECTs) based on the call graph. The performance impact varies.
>
>    Together with a linker garbage collection (linker --gc-sections option) these options may lead to smaller statically-linked executables (after stripping).
>
>    On ELF/DWARF systems these options do not degenerate the quality of the debug information. There could be issues with other object files/debug info formats.
>
>    Only use these options when there are significant benefits from doing so. **When you specify these options, the assembler and linker create larger object and executable files and are also slower.** These options affect code generation. **They prevent optimizations by the compiler and assembler using relative locations inside a translation unit since the locations are unknown until link time.** An example of such an optimization is relaxing calls to short call instructions.

also note that when using the `-ffunction-sections`, then each function is placed in its own section, in which case this issue covers this specific use-case: https://github.com/zephyrproject-rtos/zephyr/issues/64149

carlescufi, 2025-02-14:
Implemented in #74402

closed 2025-02-14 by carlescufi