zephyrproject-rtos/zephyr #64171: OpenOCD not working for windows

MishalJasmine, 2023-10-20:
Hi .. I am using Zephyr RTOS to build a firmware for Raspberry Pi Pico and using RaspberryPi Debug Probe for debug purpose. Although the build is happening succesfully, I am facing issues with the flashing. 

Things I tried:

1. I followed the instructions mentioned [here](https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html#uploading-new-programs-to-your-pico). Installed the OpenOCD using MSYS2 but I am getting this error.

`C:\Users\protocentral\protocentral_healthypi5_zephyr>west flash                                                              
-- west flash: rebuilding                                                                                                    
[1/10] Performing build step for 'second_stage_bootloader'                                                             
ninja: no work to do.                                                                                                        
[3/3] Completed 'second_stage_bootloader'                                                                                    
-- west flash: using runner openocd                                                                                       
FATAL ERROR: required program C:/Program Files/OpenOCD/bin/openocd.exe not found; install it or add its location to PATH 
`

2. I installed OpenOCD using the instructions given [here](https://docs.zephyrproject.org/latest/develop/flash_debug/host-tools.html#flash-debug-host-tools). But I am getting this error

` C:\Users\protocentral\protocentral_healthypi5_zephyr>west flash
-- west flash: rebuilding
[1/10] Performing build step for 'second_stage_bootloader'
ninja: no work to do.
[3/3] Completed 'second_stage_bootloader'
-- west flash: using runner openocd
-- runners.openocd: Flashing file: C:/Users/protocentral/protocentral_healthypi5_zephyr/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.12.0 (2023-10-02) [https://github.com/sysprogs/openocd]
Licensed under GNU GPL v2
libusb1 09e75e98b4d9ea7909e8837b7a3f00dda4589dc3
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
set_adapter_speed_if_not_set
Can't find interface/cmsis-dap.cfg
FATAL ERROR: command exited with status 1: 'C:/Program Files/OpenOCD/bin/openocd.exe' -s 'C:/Users/protocentral/protocentral_healthypi5_zephyr/boards/arm/healthypi5_rp2040\support' -s C:/Users/protocentral/zephyr-sdk-0.16.3/sysroots/AMD64-pokysdk-linux/usr/share/openocd/scripts -f 'C:/Users/protocentral/protocentral_healthypi5_zephyr/boards/arm/healthypi5_rp2040\support\openocd.cfg' -c 'source [find interface/cmsis-dap.cfg]' -c 'transport select swd' -c 'source [find target/rp2040.cfg]' -c 'set_adapter_speed_if_not_set 20000' '-c init' '-c targets' -c 'reset init' -c 'flash write_image erase C:/Users/protocentral/protocentral_healthypi5_zephyr/build/zephyr/zephyr.hex' -c 'reset run' -c shutdown
`

what is it that I am doing wrong?

`

MishalJasmine, 2023-12-24:
i followed the instructions given [here](https://docs.zephyrproject.org/latest/develop/flash_debug/host-tools.html#openocd-debug-host-tools) and I was able to proceed with the flashing of the firmware.

yonsch, 2023-12-24:
> i followed the instructions given [here](https://docs.zephyrproject.org/latest/develop/flash_debug/host-tools.html#openocd-debug-host-tools) and I was able to proceed with the flashing of the firmware.

@MishalJasmine So can this issue be closed?

MishalJasmine, 2023-12-24:
Yes the issue can be closed.

closed 2023-12-24 by yonsch