zephyrproject-rtos/zephyr #64172: Bluetooth: Host: Unable to easily determine if the GATT MTU has been exchanged

Thalley, 2023-10-20:
**Is your enhancement proposal related to a problem? Please describe.**
An application is currently not able to easily determine if the GATT MTU has been exchanged or not. 

In the case that the support maximum supported MTU on any of the devices is 23, then `bt_gatt_get_mtu` will always return 23 (`BT_ATT_DEFAULT_LE_MTU`), so that cannot be used. 

The `bt_gatt_cb.att_mtu_updated` callback is called on connection via `bt_att_connected` as well, so just checking if that callback has been won't work either. 

It looks like the only way to determine if the MTU has been exchanged is by checking if `bt_gatt_cb.att_mtu_updated` has been called twice. 

This makes it's a bit tricky for the application to know whether it should attempt to update the MTU (if it's a GATT client). 

**Describe the solution you'd like**
Either modify or add a new callback to more easily determine if the MTU has been exchanged, or add something like `bool bt_gatt_mtu_is_exchanged(struct bt_conn *conn)` (which would just check the `BT_CONN_ATT_MTU_EXCHANGED` flag). 

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

ubieda, 2024-01-18:
Some follow-up points on the use of `att_mtu_updated` and `bt_gatt_exchange_mtu`:

**- API applicability based on GATT role:** `att_mtu_updated` gets called both as GATT Server as well as GATT Client; whereas, `bt_gatt_exchange_mtu` only corresponds to being used as a GATT Client, from which the MTU exchange is initiated.
**- MTU Exchange Result:** `bt_gatt_exchange_mtu` does include the err-code that comes from the ATT request, whereas `att_mtu_updated` is called as a result of a successful MTU exchange.
**- One-MTU-exchange-per-connection rule:** Per the BT spec, there should be only one MTU exchange per connection, which corresponds to the `bt_gatt_exchange_mtu` implementation returning `-EALREADY` when the exchange has been requested (even if it’s in progress). That being the case, there wouldn't be a change on the handling of `bt_gatt_exchange_mtu()` return-codes, as only one MTU exchange request should be applicable.

As a result:
- If the MTU exchange is successful, both `att_mtu_updated` and handler provided through `bt_gatt_exchange_mtu` will be notified with the resulting MTU size.
- If the MTU exchange is unsuccessful (`ATT_EXCHANGE_MTU_REQ` is replied with `ATT_ERROR_RSP`), `bt_gatt_exchange_mtu` will contain the resulting `err`, whereas `att_mtu_updated` is not triggered.

Based on this information: do we want to provide means on `att_mtu_updated` to inform listeners that a MTU exchange took place and failed?

CC: @carlescufi @jori-nordic.

ubieda, 2024-01-19:
> - Add a note to the migration document documenting the change in behavior to att_mtu_updated

Count I get a pointer where the migration document is? I'm not familiarized with it :smiling_face_with_tear:

Thalley, 2024-01-23:
> Based on this information: do we want to provide means on `att_mtu_updated` to inform listeners that a MTU exchange took place and failed?

Hmm, this is a good point. If there are multiple applications on a device, then if the MTU exchange fails, then only the caller will be aware of it, and since the core spec states

> This request shall only be sent once during a connection by the client.

Then it cannot be retried. That will of course become obvious if they try, where the have the return value

Thalley, 2024-01-25:
I would OK with closing it. We don't really have any use case for it, as https://github.com/zephyrproject-rtos/zephyr/issues/68070 better fits what LE Audio needs (as of today).

closed 2024-02-13 by jori-nordic