zephyrproject-rtos/zephyr #64182: Convert classic GitHub projects to new project experience

henrikbrixandersen, 2023-10-20:
The following GitHub Zephyr projects still use the classic GitHub project experience:

- [ ] [Retention system (and retained memory)](https://github.com/zephyrproject-rtos/zephyr/projects/42) (@nordicjm)
- [ ] [mcumgr](https://github.com/zephyrproject-rtos/zephyr/projects/37) (@nordicjm)
- [ ] [USB Device rework](https://github.com/zephyrproject-rtos/zephyr/projects/28) (@jfischer-no)
- [ ] [Security features/improvements/documentation](https://github.com/zephyrproject-rtos/zephyr/projects/24) (@ceolin)

These should be converted to the new GitHub project experience to take advantage of the improved integration. Please see https://docs.github.com/en/issues/planning-and-tracking-with-projects/creating-projects/migrating-from-projects-classic for conversion instructions.

nordicjm, 2023-10-23:
The new projects are abysmal from my perspective, there is a lack of information on them like who opened the issue and the page runs so incredibly slow when dragging a card that it freezes for periods of time.

henrikbrixandersen, 2023-10-23:
> The new projects are abysmal from my perspective, there is a lack of information on them like who opened the issue and the page runs so incredibly slow when dragging a card that it freezes for periods of time.

I hadn't noticed that the submitter name is not shown on the new project experience boards, that's a pity - but hardly a show-stopper for migrating?

I have no issues with browser freezing or slowing down when dragging (Firefox + Safari). Anybody else seeing this? Did you submit a GitHub support issue for this?

nordicjm, 2023-10-23:
The issue is well known, apparently, and nothing has been done to address it: https://github.com/orgs/community/discussions/50088

henrikbrixandersen, 2023-10-23:
> The issue is well known, apparently, and nothing has been done to address it: https://github.com/orgs/community/discussions/50088

Did you report a bug on https://support.github.com/contact/bug-report ?

nordicjm, 2023-10-23:
> > The issue is well known, apparently, and nothing has been done to address it: https://github.com/orgs/community/discussions/50088
> 
> Did you report a bug on https://support.github.com/contact/bug-report ?

Not previously but have just opened a support ticket on it

ceolin, 2023-10-24:
@henrikbrixandersen I just closed the security one. We already have one security project using the new model.

henrikbrixandersen, 2023-10-24:
> Lets just set a deadline and disable classic in the org by that deadline. Most projects have moved on already. The remaining classic projects seem to be subsystem specific and should not gate the whole the project.

Sounds good to me. How about end of 2023? That gives people ~2 months for migration the last few classic projects.

nordicjm, 2023-10-25:
New projects are pretty much unusable from the project page for me so if there is a forced migration to version 2 then is there someone I can ping to mass move cards i.e. around release time?

henrikbrixandersen, 2023-11-11:
> Not previously but have just opened a support ticket on it

@nordicjm Any feedback on the ticket yet?

nordicjm, 2023-11-13:
> Sorry about the delay in getting back to you on this. I have escalated this to the team for investigation. We'll keep you updated to any developments.

> Note that while this support ticket may automatically close, that the internal issue will remain open and under investigation.

henrikbrixandersen, 2024-01-02:
@nordicjm Any news regarding the support ticket?

nordicjm, 2024-01-02:
> @nordicjm Any news regarding the support ticket?

Nothing from github no

henrikbrixandersen, 2024-07-25:
Please note:

> Projects (classic) will be sunset on August 23, 2024
> We encourage you to migrate this project to the new Projects experience. Read more about this sunset in our changelog. Note: Projects must be enabled to migrate.

henrikbrixandersen, 2024-09-26:
Closing this as complete since the classic projects were automatically converted when the feature was sunset.

closed 2024-09-26 by henrikbrixandersen