zephyrproject-rtos/zephyr #64195: BT: tests/bsim/bluetooth/audio/test_scripts/bap_broadcast_audio_assistant broken in main

aescolar, 2023-10-20:
**Describe the bug**
The BT Audio test 
tests/bsim/bluetooth/audio/test_scripts/bap_broadcast_audio_assistant
is broken in main, on an nrf52_bsim. It blocks CI.

**To Reproduce**
Steps to reproduce the behavior:
1. WORK_DIR=${ZEPHYR_BASE}/bsim_out  nice tests/bsim/bluetooth/audio/compile.sh 
2. tests/bsim/bluetooth/audio/test_scripts/bap_broadcast_audio_assistant
3. See error

**Expected behavior**
Passing testcase

**Impact**
Main development broken when this test is triggered

**Logs and console output**
https://github.com/zephyrproject-rtos/zephyr/actions/runs/6588609341/job/17901270642?pr=63803#step:12:137

**Environment (please complete the following information):**
 - OS: Linux, Ubuntu 22.04 / CI
 - Toolchain (e.g Zephyr SDK, ...): Host gcc (11 or 12)
 - Commit SHA or Version used: Main as of now 02deea0e806a40a33eb70edf4b810221c14a0cad

aescolar, 2023-10-20:
Lowering priority as the test has been disabled by now

Thalley, 2023-10-23:
Will look at this today

closed 2023-11-06 by carlescufi