zephyrproject-rtos/zephyr #64206: Unable to Obtain Thread IDs for all Zephyr k_thread_create() APIs in application

bob2oneil, 2023-10-20:
For the Native-POSIX board target, I need to obtain the thread ID for all threads in the application, both those that result from kernel configuration and ones that I create in the application using the **K_THREAD_DEFINE** macro, or the default thread that results from the main() application.  The intention here is to be able to increase the thread priorities on a commodity Linux workstation so that the application proper has a more pronounced Linux scheduler priority.

For the main() thread, I use the **pthread_self()** API to obtain the handle that can successfully increase its thread priority via the standard POSIX **pthread_setschedparam()** API.  Using the API, this thread when viewed using the **top** or **htop** application shows a higher priority than the Zephyr default of level 20 for all threads.

This ID would presumably be the return value from **k_thread_create()**, which I am unable to access.

I intend to use thread callback mechanism via the **k_thread_foreach()** API to iterate through all application threads.
While this callback mechanism is called for each application thread, I am unaware of how to obtain the thread ID
for the Zephyr created threads that is compatible with the **pthread_setschedparam()** API similar to the way the **pthread_self()**
API works for the main application thread.

Obtaining this handle via the callback mechanism is not apparent through an API from the k_thread pointer.

My code looks something like the following, what is missing the Thread ID obtained in the callback.

I need an ID of the data type **pthread_t** to use with the standard Linux API **pthread_setschedparam()**.

The use of **k_thread** pointer provided as an argument to the callback does not work as the thread does not have an enhanced Linux host thread priority.

```
k_thread_foreach(App::threadCallback, (void *)0);

    
void App::threadCallback(const struct k_thread *cthread, void *user_data)
{
    char state_str[32];

    struct k_thread *thread = (struct k_thread *)cthread;

    const char *tname = k_thread_name_get(thread);

    printk("%s%p %-10s\n",
           (thread == k_current_get()) ? "*" : " ",
           thread,
           tname ? tname : "NA");

    printk("\tID: %p, options: 0x%x, priority: %d timeout: %d\n",
           thread,
           thread->base.user_options,
           thread->base.prio,
           (int64_t)thread->base.timeout.dticks);

    printk("\tstate: %s, entry: %p\n",
           k_thread_state_str(thread, state_str, sizeof(state_str)),
           thread->entry.pEntry);

#ifdef CONFIG_ARCH_POSIX
    if (isRootUser())
    {
        int policy = SCHED_FIFO;
        uint32_t sched_priority = zephyr_to_posix_priority(thread->base.prio, &policy);

        pthread_t threadid = *(pthread_t *)thread;

        printk("Changing thread '%s' ID %ld Zephyr priority %d to Linux priority %d, policy %d\n", tname, threadid, thread->base.prio, sched_priority, policy);

        SysUtils::setThreadScheduleAndPriority(threadid, sched_priority, policy);
    }
#endif
}
```

aescolar, 2023-10-21:
Not a bug, but a question.
In very short: the Zephyr API does not know and cannot know anything about the fact that each Zephyr thread is mapped to an underlying Linux pthread. So there is no way to get the pthread number from a kthread structure.
In any case, what you are doing is not advisable (trying to raise priorities of a Zephyr thread in Linux), but you can do that easily when creating the process (calling the zephyr.exe executable), or later by using [renice](https://unix.stackexchange.com/questions/294299/how-to-renice-all-threads-and-children-of-one-process-on-linux).
This seems to be a continuation of https://github.com/zephyrproject-rtos/zephyr/issues/58031#issuecomment-1570726586

bob2oneil, 2023-10-22:
Thanks for the quick.  I am using the pthread_self() API to obtain the pthread info for the main application thread as I stated, but this does not do my any good when trying to promote the dozen or so threads created within Zephyr.  We will disagree on the danger of promotion of threads into real-time behavior, this is done frequently for developments for embedded Linux and I have done so personally successfully on several project, as well as OS tuning.  I need near real-time behavior from a Zephyr port to a th Native-POSIX (Linux) board target, and am finding it does not behave real-time and appears to have OS suspensions entire which causes problems for me.  I will not near term have access to any underlying hardware, thus the necessity to get near real-time behavior when operating under standard Linux.  The reason for the "bug" category, is nobody was offering up any comments on a post that I have in June, so the "bug" category served the purpose of getting some eyes on the issue.  If what I am requiring simply can not be done, that is fine, I just wanted to pursue all possible options.  I had a suspicion from my logging that the k_thread_create ID might not correlate to an ID that the Linux operating system could use, but I needed to confirm that.

closed 2023-10-22 by bob2oneil