zephyrproject-rtos/zephyr #64211: There is no support to clear the _kernel.cpu[1].current information after PSCI CPU off is called.

SumitJain3083, 2023-10-21:
On multicore , after PSCI cpu_off is called, _kernel.cpus.current is still holding the thread info which turned it off and while calling CPU ON on that core fatal exception is observed.

Below is how my pseudo code looks like

`
void thread_2_code()
{
  pm_cpu_off();
}

void thread_1_code()  // Running on 0th Core.
{
   wait_for_cpu_off();
  pm_cpu_on(mpid_of_1st_core,__start);
}

void create_threads()
{
   create_thread_1;
   create_thread_2;
   pin_thread_1_on_core_0;
   pin_thread_2_on_core_1;
  start_thread_2;
}

`

the _kernel.cpu[1].current after swithcing off the core still shows thread_2_code running .

Is this way of calling PSCI CPU OFF correct ? 
what all needs to be taken care off before calling CPU OFF on other core , so that if sometime later CPU ON is called it works fine ?

SumitJain3083, 2023-10-21:
@ceolin @gmarull , Can you please help to answer above queries.

MaureenHelm, 2023-10-24:
@povergoing

povergoing, 2023-10-25:
@carlocaione Have we ever supported the `pm_cpu_off`? I think this is more like a missing feature instead of a bug? As I am afraid that there might be many changes to support the core power off. Apart from the `pm_cpu_off`, we need also to handle the case without PSCI call.

SumitJain3083, 2023-10-25:
@povergoing as i see , PSCI is supported in zephyr kernel. But seems like the other things required to support PSCI is not there.

povergoing, 2023-10-25:
> @povergoing as i see , PSCI is supported in zephyr kernel. But seems like the other things required to support PSCI are not there.

Apologize that I didn't make it clear, but in terms of power control, `pm_cpu_on` works with both PSCI and non-PSCI, so the `pm_cpu_off` should work in the same way which results in not just a simple fix. Anyway, let me have a look.

carlocaione, 2023-10-25:
I think we simply never cared about "rebooting" a core so far. Moving this to low priority.

SumitJain3083, 2023-10-25:
@carlocaione if possible can you please share some guide (if i want do some custom implementation) for rebooting a core then  how to implement such feature. 

will it be possible with the current way of kernel handling the scheduler and GIC ?

carlocaione, 2023-10-25:
It should be possible but nobody cared so far and I think there would be several fixes here and there. I don't have any guide but it should not be impossible to achieve.

SumitJain3083, 2023-10-30:
@carlocaione  thanks for the info. 
So just to turn on the core after CPU_OFF has been called, does it required to be turned on with "__start" of the kernel , like it is done in SMP init. 

Like below pseudo code
Core 1 executing : 
pm_cpu_off()

core 0 executing : 

wait_for_cpu_off();
pm_cpu_on(<mpid_of_1st_core> , &__start);

carlocaione, 2023-10-30:
First of all you should use PSCI to turn the CPU on. The problem I see is that Zephyr right now is assuming that all the cores are started by the first core in a sequential way at boot, so it requires a synchronisation point at https://github.com/zephyrproject-rtos/zephyr/blob/18167754665b6ce3a1b80d9c07d4e4008a3eb298/arch/arm64/core/smp.c#L92-L119 so this could be definitely more difficult than I was anticipating.

SumitJain3083, 2023-10-30:
@carlocaione  i tried to turnoff the core and then turn it on with pm_cpu_on( <core_mpid , &__reset ). I believe the code you pointed out will be running only on primary code ?

Since the mpid of the secondary core is already know the __reset function will call 

void z_arm64_secondary_start(void) function.

The only problem it faces is the fn(arg) if run again , fn is NULL so it creates fatal error. Can you please make 
fn as global and not loose the variable value ? this should help to call smp_init_top always even when z_arm64_secondary_start is called twice for a single core ?

`
void z_arm64_secondary_start(void)
{
	int cpu_num = arm64_cpu_boot_params.cpu_num;
	arch_cpustart_t fn;
`

SumitJain3083, 2023-10-30:
```c
arch_cpustart_t temp_copy=NULL;

/* the C entry of secondary cores */
void z_arm64_secondary_start(void)
{
	int cpu_num = arm64_cpu_boot_params.cpu_num;
	arch_cpustart_t fn;
	void *arg;

	__ASSERT(arm64_cpu_boot_params.mpid == MPIDR_TO_CORE(GET_MPIDR()), "");

	/* Initialize tpidrro_el0 with our struct _cpu instance address */
	write_tpidrro_el0((uintptr_t)&_kernel.cpus[cpu_num]);

	z_arm64_mm_init(false);
#ifdef CONFIG_SMP
	arm_gic_secondary_init();

	irq_enable(SGI_SCHED_IPI);
#ifdef CONFIG_USERSPACE

	irq_enable(SGI_MMCFG_IPI);
#endif
#ifdef CONFIG_FPU_SHARING

	irq_enable(SGI_FPU_IPI);
#endif
#endif

	fn = arm64_cpu_boot_params.fn;
	if(temp_copy==NULL)
	{
		temp_copy = arm64_cpu_boot_params.fn;
	} 
	arg = arm64_cpu_boot_params.arg;

	dsb();

	/*
	 * Secondary core clears .fn to announce its presence.
	 * Primary core is polling for this. We no longer own
	 * arm64_cpu_boot_params afterwards.
	 */
	arm64_cpu_boot_params.fn = NULL;

	dsb();
	sev();

	temp_copy(arg);
}
``` 

Something like above

carlocaione, 2023-11-03:
@SumitJain3083 please, submit a PR for that if you want that reviewed and tested in CI

SumitJain3083, 2023-11-09:
@carlocaione  Raised the PR https://github.com/zephyrproject-rtos/zephyr/pull/64999 , can you please help to review.

andyross, 2023-11-09:
FWIW: note that @dcpleung has some changes to the CPU lifecycle APIs in #64755 (for the benefit of intel_adsp, which has some... idiosyncratic CPU idle code) that seems relevant to this issue.  Probably worth checking that out and making sure that whatever work is requried here will be compatible.

closed 2024-01-24 by github-actions[bot]