zephyrproject-rtos/zephyr #64216: bt: hci: ipm_stm32wb: doesn't apply STM's unique/factory ID MAC address

lochej, 2023-10-21:
**Describe the bug**
When trying the bluetooth examples for STM32WB55RG on two different nucleo_wb55rg boards, I notices that the MAC address was the same for both:

```
 02:80:E1:00:00:00
```

**To Reproduce**
```
#build a Bluetooth example, peripheral_hr for example:
cd ~/zephyrproject/zephyr/samples/bluetooth/peripheral_hr/
west build -p always -b nucleo_wb55rg && west flash
```
The 2 boards prints the same MAC address:
Board 1:
```
*** Booting Zephyr OS build zephyr-v3.5.0-233-g9ac852549610 ***                                                                                                                                                                       
[00:00:00.216,000] <inf> bt_hci_core: Identity: 02:80:E1:00:00:00 (public)
[00:00:00.216,000] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0026, manufacturer 0x0030
[00:00:00.217,000] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0x2126
Bluetooth initialized
Advertising successfully started
```
Board 2:
```
*** Booting Zephyr OS build zephyr-v3.5.0-233-g9ac852549610 ***                                                                                                                                                                       
[00:00:00.216,000] <inf> bt_hci_core: Identity: 02:80:E1:00:00:00 (public)
[00:00:00.216,000] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0026, manufacturer 0x0030
[00:00:00.217,000] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0x2126
Bluetooth initialized
Advertising successfully started
```

**Possible root of the issue**
The hci_core->common_init() function issues a HCI reset, which I believe disgards the previous configuration done by ipm_stm32wb->bt_ipm_open()->bt_ipm_ble_init()->bt_imp_set_addr() by resetting the M0 coprocessor. I don't know if all the settings set prior to hci_core->common_init() are affected, but it could be the case. 

**Possible fix**
Add BT_QUIRK_NO_RESET to the bt_hci_driver in ipm_stm32wb.c
Is there a way to add this quirk using another method that I'm not aware of ?

```
static const struct bt_hci_driver drv = {
	.name           = "BT IPM",
	.bus            = BT_HCI_DRIVER_BUS_IPM,
	.open           = bt_ipm_open,
#ifdef CONFIG_BT_HCI_HOST
	.close          = bt_ipm_close,
#endif
	.send           = bt_ipm_send,
	.quirks = BT_QUIRK_NO_RESET, //---> Add this driver quirk
};

static int _bt_ipm_init(void)
{
...
}
```

New output with this change, the MAC address are unique for 2 different boards running the same sample (peripheral_hr).

Board 1:
```
*** Booting Zephyr OS build zephyr-v3.5.0-233-g9ac852549610 ***
[00:00:00.215,000] <inf> bt_hci_core: Identity: 80:E1:26:1B:A1:E9 (public)
[00:00:00.215,000] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0026, manufacturer 0x0030
[00:00:00.215,000] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0x2126
Bluetooth initialized
Advertising successfully started
```
Board 2:
```
*** Booting Zephyr OS build zephyr-v3.5.0-233-g9ac852549610 ***
[00:00:00.215,000] <inf> bt_hci_core: Identity: 80:E1:26:01:AB:7C (public)
[00:00:00.215,000] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0026, manufacturer 0x0030
[00:00:00.215,000] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0x2126
Bluetooth initialized
Advertising successfully started
```

**Expected behavior**
A config is available to select the public static mac address generated by STMicroElectronics when the IPM hci driver is loaded.

**Impact**
All stm32wb samples have the same MAC address which prevents differentiating them.

**Environment (please complete the following information):**
 - OS: Xubuntu 22.04 x64
 - Version: Zephyr RTOS V3.5.0-233

lochej, 2023-10-22:
Related issue to MAC address setting on the WB55, 
I've tried setting up my company's public MAC address on the nucleo_wb55rg board without success yet:

Tentative was the following:

I tried modifying my public MAC address using and the settings API in the "samples/bluetooth/peripheral"  example with small modifications:
```
#Disable the PRIVACY config to use the public MAC address instead
CONFIG_BT_PRIVACY=n
```

Use the CONFIG_BT_SETTING to load "bt/id" key and initialized the bluetooth driver:
then add the public address setting update function using the settings API:
```
void setting_load_public_MAC_addr(void)
{
	//Makeen Energy range:
	//28:36:13 (hex)
	//A0:00:00 to AF:FF:FF
	bt_addr_le_t pub_addr;
	
	pub_addr.type=BT_ADDR_LE_PUBLIC;
	pub_addr.a.val[0]=0xEF;
	pub_addr.a.val[1]=0xBE;
	pub_addr.a.val[2]=0xA0;
	pub_addr.a.val[3]=0x13;
	pub_addr.a.val[4]=0x36;
	pub_addr.a.val[5]=0x28;

	//BT_ADDR_SET_STATIC(&pub_addr.a);

	/* Save the BT address for next reboot */
	settings_save_one("bt/id",&pub_addr,sizeof(pub_addr));
}

static void bt_ready(void)
{
...
	if (IS_ENABLED(CONFIG_SETTINGS)) {
		settings_load();
		setting_load_public_MAC_addr(); //--> make sure the MAC setting is the one we want
	}
...
	printk("Advertising successfully started\n");
}
```

The sample outputs the correctly setup address 28:36:13:A0:BE:EF

```
*** Booting Zephyr OS build zephyr-v3.5.0-233-g9ac852549610 ***
[00:00:00.019,000] <inf> fs_nvs: 2 Sectors of 4096 bytes
[00:00:00.019,000] <inf> fs_nvs: alloc wra: 0, f30
[00:00:00.019,000] <inf> fs_nvs: data wra: 0, 100
[00:00:00.039,000] <inf> bt_hci_core: No ID address. App must call settings_load()
Bluetooth initialized
[00:00:00.040,000] <err> settings: set-value failure. key: bt/irk error(-2)
[00:00:00.042,000] <inf> bt_hci_core: Identity: 28:36:13:A0:BE:EF (public)
[00:00:00.042,000] <inf> bt_hci_core: HCI: version 1.0b (0x00) revision 0xa075, manufacturer 0x0030
[00:00:00.042,000] <inf> bt_hci_core: LMP: version 1.0b (0x00) subver 0x2175
Advertising successfully started
Indicate VND attr 0x8016fd0 (UUID 12345678-1234-5678-1234-56789abcdef1)
```

But unfortunately, a scan with NRF Connect shows the following MAC:
```
Zephyr Peripheral Sample Long
02:80:E1:00:00:00
```

Furthermore, even if I apply the BT_QUIRKS_NO_RESET that fixed loading the public address derived from STM's chip unique ID I still get STM's public address and not my company's one.
```
Zephyr Peripheral Sample Long
80:E1:26:1B:A1:E9
```

My guess is that the HCI command to update the loaded identity address is not being sent to the M0 core, therefore not updating the MAC address that shows up in NRF Connect.

jlh-makeen, 2024-05-25:
It's been a while since I posted this issue and I've not worked with wb55 lately. 

Has anyone been able to reproduce this issue with latest main ?

erwango, 2024-05-27:
No changes were done to solve this issue. It is still valid.

erwango, 2024-07-30:
^^@asm5878

asm5878, 2024-08-02:
Issue is clear, STM32WB HCI driver must be reworked in order to give an unique address using  "setup" function
Assigning a public address is clearly feasible.
A solution will be provided by mid-September

jlh-makeen, 2024-08-03:
> Issue is clear, STM32WB HCI driver must be reworked in order to give an unique address using "setup" function Assigning a public address is clearly feasible. A solution will be provided by mid-September

Amazing to see you'll be able to fix this issue. I didn't really have much time or BLE projects active to focus on this issue... 

I hope you'll be able to address both ST's public mac setting and "proprietary" public mac setting through the settings API :-) 

I don't remember the hci api having a setup function at the time of testing. Maybe I just missed it, but it's cool if the issues can be fixed by implementing it 👍

ericmechin, 2025-08-20:
Hi,

The issue is solved with the PR [https://github.com/zephyrproject-rtos/zephyr/pull/93186](url).

Best regards.

ericmechin, 2025-08-21:
Hi,

You can set a custom/vendor mac address with an HCI vendor specific command, like with the following function example:
static void set_public_addr(void)
{
	struct net_buf *buf;
	struct aci_set_ble_addr *param;
        uint8_t addr[6] = { 0x55, 0x44, 0x33, 0x22, 0x11, 0x00 };
	int err;

	buf = bt_hci_cmd_create(BT_OP(BT_OGF_VS, 0xFC0C), sizeof(param));
	if (!buf) {
		printk("Unable to allocate command buffer\n");
		return;
	}

	param = net_buf_add(buf, sizeof(*param));
	param->config_offset = 0;
	param->length = 6;
	param->value[0] = addr[0];
	param->value[1] = addr[1];
	param->value[2] = addr[2];
	param->value[3] = addr[3];
	param->value[4] = addr[4];
	param->value[5] = addr[5];

	err = bt_hci_cmd_send_sync(BT_OP(BT_OGF_VS, 0xFC0C),
				   buf, NULL);
	if (err) {
		printk("Set public addr err: %d\n", err);
		return;
	}
}

This function can be called in main() just after bt_enable() function call.

Best regards.