zephyrproject-rtos/zephyr #64238: Bluteooth: False comment: all cmd_complete have status field

alwa-nordic, 2023-10-23:
**Describe the bug**

The following code is documenting a false assumption that all command complete events have a status field. This is not true for at least one command, HCI_Disconnect. Core 5.4, 4.E.7.1.6 says there are no return parameters.

https://github.com/zephyrproject-rtos/zephyr/blob/4e946262cbcf712e01c9d9d631d3f843789e6c91/subsys/bluetooth/host/hci_core.c#L2335-L2338

alwa-nordic, 2023-10-25:
You are right. I will close this ticket.

Still, it seems like this comment is relying on happenstance. Do you know if there has been any effort to promote the 'status' field into the 'Command complete' event itself, if there really is consensus that every 'Command complete' has a status.

closed 2023-10-25 by alwa-nordic

jhedberg, 2023-10-25:
> You are right. I will close this ticket.
> 
> Still, it seems like this comment is relying on happenstance. Do you know if there has been any effort to promote the 'status' field into the 'Command complete' event itself, if there really is consensus that every 'Command complete' has a status.

I'm not aware of such efforts in the CSWG, i.e. this seems like an unwritten rule. Maybe @holtmann has some info if this has ever been discussed in the SIG?

Btw, also Linux makes the same assumption, so if a command ever deviates from this it's likely to cause breakage in other systems as well:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/net/bluetooth/hci_event.c#n4258