zephyrproject-rtos/zephyr #64281: Freeze pip tooling as part of a release

dkalowsk, 2023-10-23:
## Introduction

The Zephyr kernel has dependencies on external build tooling to work correctly.  As it is today, the version of those tools are not clearly defined, making it easy for a release to not build and to not have a reproducible build.  With some minor tweaks to the release process this can be mitigated.

### Problem description

Once a Zephyr release is made, it is unclear what versions of additional tooling are necessary to build the release.  This becomes a problem for any LTS or release that is older than $CURRENT-2 for support

### Proposed change

During the release process, adding the tested/supported versions of the additional `pip` tooling to narrow potential deltas and clarify versions needed.

## Detailed RFC

Today the build dependencies are for the most part installed through the `scripts/requirements.txt` settings, which are executed with a `pip install -r scripts/requirements.txt`.  This process will install the versions of tooling as defined.  When not defined, the tooling pulled is the latest version available.  This process is an excellent process to make sure that Zephyr compiles and functions with the most recent versions of the tooling.

To provide longer lasting releases though, the ambiguity of the versions being installed becomes a challenge.  Especially on anything with LTS.  Leaving it to end users to develop the necessary setups for tracking and maintaining tooling dependencies isn't ideal when the release package can make it very clear.

When the final Zephyr release candidate is generated, before tagging the release, adding a commit that freezes the versions of all the tools in `requirements.txt` makes it significantly easier to support an LTS and releases older than $CURRENT-2.

### Proposed change (Detailed)

Repeatable builds in firmware are rather essential from having patching to properly recovering a binary released out to the world.  Doing so often runs developers down the path of creating a Virtual Machine for builds that is only updated on some agreed upon cadence, a containerized build environment, or moving to build environments such as `conan` or `bazel`.

All of these have some deficiency that the Zephyr release process can close off.  For example, the best of the containerized environments I've found from Jonathan Beri (https://github.com/beriberikix/zephyr-docker/blob/168968f0fe3b44b861a34e6f6b097ae22bdddd01/Dockerfile#L65) still does a blind pull of the requirements.txt.  If the container were to be re-built sometime after an initial release, it's possible to silently break reproducible builds.

The proposal is to run a set such as `pip freeze` or `pipreqs` to generate the tested against and known good combinations of tooling for that release.

Taking a look at requirements-base.txt today, some of the tooling does already have the values set:
```
# used by various build scripts
pyelftools>=0.27

# used by dts generation to parse binding YAMLs, also used by
# twister to parse YAMLs, by west, zephyr_module,...
PyYAML>=5.1
```

while others have nothing defined:
```
# YAML validation. Used by zephyr_module.
pykwalify

# used by west_commands
canopen
packaging
progress
psutil
pylink-square
pyserial
requests

# for ram/rom reports
anytree
```

An argument can be made that these are not all essential to building.  I'd counter with them being part of the regular tooling used by the Zephyr community to varying degrees.  See recent issues where updates to `anytree` broke builds for LTS and past releases: https://github.com/zephyrproject-rtos/zephyr/issues/61946 and https://github.com/zephyrproject-rtos/zephyr/issues/60213

### Concerns and Unresolved Questions

The unknown here is how this will impact folks working between main development and multiple release branches.

## Alternatives

None that I can think.

henrikbrixandersen, 2023-10-24:
_If_ the Python package versions are to be frozen at release time, we'd need some way of unfreezing them again for the next release cycle. This could be as simple as reverting the freeze commit, but it's not a very clean process imho.

Another thing to consider - if we start freezing Python package versions in the release branches, we will also need to start caring about security fixes to those Python packages. We would need to update the package versions on supported release branches whenever a security flaw is discovered in one of the frozen packages.

I am not sure where this proposal is best discussed. We can take it up in the release meeting, but perhaps the Process WG/Security WG should also have a say in this?

pdgendt, 2023-10-24:
A new file could be added for each tag, that is done at release time?

```shell
python -m pip freeze > scripts/release/requirements-3.5.0.txt
```

henrikbrixandersen, 2023-10-24:
> A new file could be added for each tag, that is done at release time?
> 
> 
> 
> ```shell
> 
> python -m pip freeze -r scripts/requirements.txt > scripts/requirements-3.5.0.txt
> 
> ```

That could solve the unfreezing issue, but we'd likely still need to take action on security issues in frozen packages for supported releases.

pdgendt, 2023-10-24:
> That could solve the unfreezing issue, but we'd likely still need to take action on security issues in frozen packages for supported releases.

If only the tooling needs an update, I think the file should be replaced with a validated new version. This allows the git history to track security issues solved with newer versions.

fabiobaltieri, 2023-10-24:
> A new file could be added for each tag, that is done at release time?
> 
> ```shell
> python -m pip freeze -r scripts/requirements.txt > scripts/requirements-3.5.0.txt
> ```

This brings in all the packages on the local system, would not ask anyone to run this on their own local checkout. How about a GitHub action that generates a commit adding this automatically from CI when the release is tagged? You could make it react to the tag and target the `v*-branch`, then once the release is tagged we'd just have a PR generated automatically with a snapshot of the packages that were used by the CI builder at that time.

pdgendt, 2023-10-24:
> This brings in all the packages on the local system, would not ask anyone to run this on their own local checkout. How about a GitHub action that generates a commit adding this automatically from CI [...]

Sure a CI action could be used to automate it. A release manager could also use a virtual environment to keep the installed packages to minimum and use the `--local` flag to skip globally installed packages.

fabiobaltieri, 2023-10-24:
> Sure a CI action could be used to automate it. A release manager could also use a virtual environment to keep the installed packages to minimum and use the `--local` flag to skip globally installed packages.

I think it's a good idea to avoid adding more manual steps to the release process if possible, it's hard enough to get the current process executed consistently considering that a release manager does one release every couple of years or less. If it can be automated I think it'd be alright.

dkalowsk, 2023-10-24:
After discussion with the release team, there is no strong opposition from the team.  There are many concerns about the process and the edge cases introduced with the change.  Request is to bring to the discussion to the Process WG (@keith-zephyr) and 
Security WG (@fabiobaltieri).

cfriedt, 2023-10-24:
> The unknown here is how this will impact folks working between main development and multiple release branches.

One solution is to have a separate Python virtual environment locally for each release branch one is dealing with, and main of course.

Thanks for proposing this. It's a concern for us at Meta because we tend to ratchet-down our python utilities (security concerns with PyPI, etc).

cfriedt, 2023-10-24:
> _If_ the Python package versions are to be frozen at release time, we'd need some way of unfreezing them again for the next release cycle. This could be as simple as reverting the freeze commit, but it's not a very clean process imho.

@henrikbrixandersen -  Wouldn't it effectively be done by creating the release branch? I guess that would be predicated on having versions there in the first place, but that could be done in `main`. No unfreezing required

keith-zephyr, 2023-10-25:
From the release meeting, everyone agreed it was a good idea. Concerns:
* Additional load on release team
* Security implications - how to manage security patches to pip dependencies

@fabiobaltieri - If we decide to do this, we should update tooling to automatically create the dependencies.
@nashif - what's the expected outcome? @dkalowsk - that downloading a Zephyr LTS release builds out of the box.
@nashif - what about OS updates?
@fabiobaltieri - proposal is to do a pip freeze, creating a new requirements file matching the release.  Users need to use the venv with the frozen pip requirements.
@nashif  - how to we guarantee this works with new Linux distros?
@fabiobaltieri - currently users can only see the current requirements, but have no data on which specific version should be used.
@fabiobaltieri - This issue carries over to other areas (cmake for instance).  @dkalowsk - cmake versions are checked throughout the project
@fabiobaltieri - this should be easy to implement.  Users just include a different requirements.txt
@dkalowsk - How to verify tooling in the backport process?  
@fabiobaltieri - CI runs on the release branches.
@dkalowsk - How are the docker images used with the branches?  @nashif - the release branches use on older docker images. so the CI system now doesn't catch these problems.
@fabiobaltieri - requirements.txt is processed at build time for the docker image, so the docker image is stable.  But we wouldn't be able to replicate this if the image needs to be rebuilt.
@dkalowsk - No objections to the idea.
@nashif - can we create this manually to dry run?
@fabiobaltieri - at a minimum we need an automated script to generate the list of all python packages. 
@fabiobaltieri and @dkalowsk to work on the proof of concept.

dkalowsk, 2024-01-06:
This is now done I believe

closed 2024-01-06 by dkalowsk