zephyrproject-rtos/zephyr #64289: net: lwm2m: send timeout callback on close

RomainPelletant, 2023-10-24:
**Describe the bug**
LwM2M client implements the send method. A callback is called on result (success or failed/timeout)
On LwM2M connection close (L4_DISCONNECTED), the timeout callback shall be called.

**To Reproduce**
Steps to reproduce the behavior:
1. lwm2m send data registering a callback
2. lwm2m close
3. callback is never called

**Expected behavior**
Timeout callback shall be called

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain 0.16.1
 - Commit 3.5

I suggest the following patch

[001_lwm2m_timeout_on_reset.patch](https://github.com/zephyrproject-rtos/zephyr/files/13112691/001_lwm2m_timeout_on_reset.patch)

If you are OK I can open a PR
Thanks for your help :)

cuevana8, 2023-10-24:
+1..[.](https://github.com/flslib/fl-studio-cracked)[@@@](https://github.com/zephyrproject-rtos/zephyr/issues/64289#issue-1958807422)

henrikbrixandersen, 2023-10-24:
Please submit code change proposals as pull requests, not patches attached to issues.

rlubos, 2023-10-24:
It'd be indeed easier to discuss the changes over the PR.

There's a dedicated callback fn for send operation (`send_status_cb`), I think it could be used instead. Especially, that it allows to report a failure(`LWM2M_SEND_STATUS_FAILURE`) instead of a timeout, which IMO fits better.

As for the second change, we shouldn't really call `lwm2m_engine_context_close()` directly from `lwm2m_rd_client_stop()`, as the engine may still need to handle the deregistration process. LwM2M context is closed in `ENGINE_DEREGISTERED` state handler.

RomainPelletant, 2023-10-24:
@rlubos thanks for your message.

On send method, lwm2m_send_cb_t is registered to get result : 

```
static void send_reply(enum lwm2m_send_status status)
{
	int ret;

	if (status == LWM2M_SEND_STATUS_SUCCESS) {
		ret = 0;
	} else {
		ret = -EAGAIN;
	}
	...
}
...
int ret = lwm2m_send_cb(lwm2m_rd_client_ctx(),
				&LWM2M_OBJ(LWM2M_OBJECT_BINARYAPPDATACONTAINER_ID, 0, 0, 0), 1U,
				send_reply);
```
This proposal is to call registered callback, with the timeout status. But you are right : a failed status is more appropriate.

lwm2m_rd_client_stop() gives the ability to close without deregistering which is useful when connection is over, and then waiting for connection to be up, then client start, etc...

rlubos, 2023-12-12:
@RomainPelletant Ping, any updates on this?

closed 2024-02-26 by github-actions[bot]