zephyrproject-rtos/zephyr #64290: net: lwm2m: send callback behavior on blockwise operation

RomainPelletant, 2023-10-24:
**Describe the bug**
Initiate blockwise operation on send method will never call send callback.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable blockwise feature
2. send data greater than COAP BLOCK SIZE and register send callback

**Expected behavior**
The send callback shall be called to get send result

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain 0.16.1
 - Zephyr 3.5

I am not a fan of the following patch, mainly solving the issue with user_data fields.
Keep msg context in user_data, on each new block, keep that context.
Ignore continue (2.31) message, and only result on the end of transmission
[008_lwm2m_callback_context.patch](https://github.com/zephyrproject-rtos/zephyr/files/13112780/008_lwm2m_callback_context.patch)

henrikbrixandersen, 2023-10-24:
Please submit code change proposals as pull requests, not patches attached to issues.

rlubos, 2023-10-24:
Isn't the `CONTINUE` code already handled at the CoAP library level, so that the reply callback is only called on the final block response? For reference: https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/coap/coap.c#L1744

Also I don't think I follow the user data change, isn't `struct lwm2m_message` preserved for the consecutive blocks? CC @SeppoTakalo

SeppoTakalo, 2023-10-24:
I see.. 
`build_msg_block_for_send()` will reset the message structure, and get rid of CoAP reply structure that was having the `user_data` pointer.

PR #64310 (2023-10-25): net: lwm2m: Keep user_data between blocks
CoAP reply structure user_data should be kept between ongoing blocks, so that callbacks for LwM2M send continue to work on blockwise transfers.

Fixes #64290
commit cf513451 (2023-10-25): net: lwm2m: Keep user_data between blocks

CoAP reply structure user_data should be kept between
ongoing blocks, so that callbacks for LwM2M send continue
to work on blockwise transfers.

Fixes #64290
closed 2023-10-25 by carlescufi