zephyrproject-rtos/zephyr #64292: dts: bindings: add power pulse duration in bg95 modem cellular

RomainPelletant, 2023-10-24:
**Is your enhancement proposal related to a problem? Please describe.**
Modem is based on BG95 MiniPCIe hardware, the power pulse duration set by default at 1500ms is sometime not sufficient

**Describe the solution you'd like**
Add the (already existing) power_pulse_duration_ms as an optional parameter by default

**Describe alternatives you've considered**
Patch the quectel,bg95.yaml file

The patch proposed:
[012_power_pulse_duration.patch](https://github.com/zephyrproject-rtos/zephyr/files/13112879/012_power_pulse_duration.patch)

henrikbrixandersen, 2023-10-24:
Please submit code change proposals as pull request, not patches attached to issues.

RomainPelletant, 2023-10-24:
According to PR comments, this kind of stuff shall be implemented in the driver and not dealed in device tree space

closed 2023-10-24 by RomainPelletant