zephyrproject-rtos/zephyr #64294: Different image rotation behaviour when using LittleFS as image source for LVGL

Kampi, 2023-10-24:
## Describe the bug

### The working setup

I have an application that prints a cursor on a display and rotates this cursor based on a sensor value by using this code:

**Cursor**

[iaq_img_cursor_png.txt](https://github.com/zephyrproject-rtos/zephyr/files/13113029/iaq_img_cursor_png.txt)

**Code**
```
LV_IMG_DECLARE(iaq_img_cursor_png);
lv_img_set_src(ui_ImgCursor, &iaq_img_cursor_png);

lv_obj_set_width(ui_ImgCursor, LV_SIZE_CONTENT);
lv_obj_set_height(ui_ImgCursor, LV_SIZE_CONTENT); 
lv_obj_set_x(ui_ImgCursor, 0);
lv_obj_set_y(ui_ImgCursor, -52);
lv_obj_set_align(ui_ImgCursor, LV_ALIGN_CENTER);
lv_obj_add_flag(ui_ImgCursor, LV_OBJ_FLAG_ADV_HITTEST);
lv_obj_clear_flag(ui_ImgCursor, LV_OBJ_FLAG_SCROLLABLE);
lv_img_set_pivot(ui_ImgCursor, 7, 63);
lv_img_set_angle(ui_ImgCursor, 0);

void iaq_app_ui_home_set_iaq_cursor(uint16_t iaq)
{
    int16_t angle;

    // Convert the IAQ value into an angle. Use the table from
    //  https://community.bosch-sensortec.com/t5/MEMS-sensors-forum/BME680-688-IAQ-meaning/td-p/45196
    // as reference.
    // 0        -> -900
    // >351     -> +900
    //  y = mx + b
    //      b = -900
    //      m = 5.12 -> 5
    // Use 5 to store everything in an int16_t. The error is 0.45° With IAQ 351
    angle = 5 * (int16_t)iaq - 900;

    lv_img_set_angle(ui_ImgCursor, angle);
}
```

**This looks as shown below**

- Initial state
[image]

- First sensor update
[image]

### The faulty setup

I replace the lines

```
LV_IMG_DECLARE(iaq_img_cursor_png);
lv_img_set_src(ui_ImgCursor, &iaq_img_cursor_png);
```

with

```
lv_img_set_src(ui_ImgCursor, "/lvgl_lfs/iaq_cursor.bin");
```

because I want to store the assets in an external SPI flash. The SPI flash is filled with the converted image from the [LVGL Image Converter](https://lvgl.io/tools/imageconverter)

[image]

The resulting binary is saved on the flash memory and the application finds the file system and the image

```
PS F:\Git\ZSWatch\app> west upload_fs --type lfs
Creating image
F:\Git\ZSWatch/app/src/images/binaries
root F:\Git\ZSWatch/app/src/images/binaries/lvgl_lfs dirs [] files ['iaq_cursor.bin']
Copying F:\Git\ZSWatch/app/src/images/binaries/lvgl_lfs\iaq_cursor.bin to iaq_cursor.bin
Uploading image
Connecting to JLink...
Connecting to nRF5340_XXAA...
Connected, send RTT_FLASH_LOAD_BOOT_MODE to RAM_ADDR...
Done. Reset into flash loader mode...
Done. Starting RTT transfer...
RTT started, 3 up bufs, 3 down bufs.
up channels:
    0: name = 'Terminal', size = 1024 bytes, flags = 0
    1: name = '', size = 0 bytes, flags = 0
    2: name = 'FlashLoaderChannel', size = 4105 bytes, flags = 2
down channels:
    0: name = 'Terminal', size = 16 bytes, flags = 0
    1: name = '', size = 0 bytes, flags = 0
    2: name = 'FlashLoaderChannel', size = 4105 bytes, flags = 2
FILENAME lvgl_resources
[00:00:00.059,692] <inf> spi_nor: at25sl128a@0: SFDP v 1.6 AP ff with 2 PH
[00:00:00.059,722] <inf> spi_nor: PH0: ff00 rev 1.6: 16 DW @ 30
[00:00:00.062,194] <inf> spi_nor: at25sl128a@0: 16 MiBy flash
[00:00:00.062,194] <inf> spi_nor: PH1: 011f rev 1.0: 2 DW @ 80
[00:00:01.362,640] <inf> fs_nvs: 8 Sectors of 4096 bytes
[00:00:01.362,670] <inf> fs_nvs: alloc wra: 3, f80
[00:00:01.362,670] <inf> fs_nvs: data wra: 3, b39
[00:00:02.104,217] <inf> littlefs: littlefs partition at /lvgl_lfs
[00:00:02.104,248] <inf> littlefs: LittleFS version 2.5, disk version 2.0
[00:00:02.104,614] <inf> littlefs: FS at at25sl128a@0:0x0 is 512 0x1000-byte blocks with 512 cycle
[00:00:02.104,644] <inf> littlefs: sizes: rd 1024 ; pr 512 ; ca 4096 ; la 4096
[00:00:02.125,091] <inf> littlefs: Automount /lvgl_lfs succeeded
*** Booting Zephyr OS build zephyr-v3.4.0-3979-g2ea2d37d60ad ***
[FILE] iaq_cursor.bin (size = 2650)
[00:00:02.165,771] <wrn> main: SPI Flash Loader Boot Mode
[00:00:03.066,528] <dbg> zsw_rtt_flash_loader: find_partition_id_from_label: Found target partition id: 0
Load sequence received: LOADER_START:lvgl_lfs_partition partition ID: 0
Filesize: 2097152
Time taken: 208 ms2 len: 4096
Took 208 ms Througput:  472615.3846153846 kbps
Sending LOADER_END
JLink disconnected, exiting...
```

I have a different behavior after the first rotation of the cursor. The cursor disappears.

- Initial state
[image]

- First sensor update
[image]

I don´t understand it. Why does the image source (LittleFS instead of program memory) changes the behavior of the rotation in LVGL?

## To Reproduce

I can provide the complete project when required. I haven´t prepared the commit with the issue yet.

### Environment (please complete the following information):
 - OS: Windows
 - Toolchain Zephyr SDK

**west.yml**
```
manifest:
  self:
    west-commands: scripts/west-commands.yml

  remotes:
    - name: zephyrproject-rtos
      url-base: https://github.com/zephyrproject-rtos

  projects:
    - name: zephyr
      remote: zephyrproject-rtos
      revision: 2ea2d37d60ada33bdf051eb9d699f4db9123c3ff
      import:
        name-blocklist:
          - ci-tools
          - hal_altera
          - hal_cypress
          - hal_infineon
          - hal_microchip
          - hal_nxp
          - hal_openisa
          - hal_silabs
          - hal_xtensa
          - hal_ti
          - loramac-node
          - net-tools
          - openthread
          - edtt
          - trusted-firmware-m
          - sof
  self:
    west-commands: scripts/west-commands.yml
```

**iaq_cursor.bin**

Please rename the file to `.bin`.

[iaq_cursor.txt](https://github.com/zephyrproject-rtos/zephyr/files/13113012/iaq_cursor.txt)

**Cursor.png**

Initial image file of the cursor. Please rename it to `.png`.

[Cursor.txt](https://github.com/zephyrproject-rtos/zephyr/files/13113060/Cursor.txt)

faxe1008, 2023-10-25:
@Kampi You seem to be using the ZSWatch for reproducing the issue. I know that the project also support the `native_posix` target and seems to be enabling the flash simulator as well: [native_posix.conf](https://github.com/jakkra/ZSWatch/blob/main/app/boards/native_posix.conf). Is the issue also reproducible with that?

Kampi, 2023-10-26:
Hi @faxe1008 
that's a really good idea. I discuss the process to do it with @jakkra, test it and share the results with you.

Kampi, 2023-10-26:
Hi @faxe1008 
I have discussed this with @jakkra and we are not sure about the source of this bug. @jakkra has implemented a raw flash API for LVGL for use without file system to the flash and the bug also occurs with this API. So we came up with the idea that the bug isn´t related to a Zephyr FS port.

In parallel, we try to test it with the posix port.

Btw. I have pushed the branch with this bug: https://github.com/jakkra/ZSWatch/tree/Sensors_and_Project_improvements

faxe1008, 2023-11-13:
Where you able to reproduce the issue on `native_posix` in the meantime? I have started to work on a MRE based on the test that is already in the repo.

closed 2024-01-28 by github-actions[bot]