zephyrproject-rtos/zephyr #64295: Binary descriptor sample doesn't work on windows

kartben, 2023-10-24:
**Describe the bug**

On Windows, I can't seem to be able to build the bindesc sample. It fails with ``ninja: error: failed recompaction: Permission denied``

**To Reproduce**
Steps to reproduce the behavior:
1.  west build -p auto -b frdm_k64f samples/subsys/bindesc/hello_bindesc

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**

Full log for ``west build -p auto -b stm32l1_disco samples/subsys/bindesc/hello_bindesc -- -DCMAKE_VERBOSE_MAKEFILE=ON`` when build folder already contains a first failed attempt at building

```
 ~#@❯  west build -p auto -b stm32l1_disco samples/subsys/bindesc/hello_bindesc -- -DCMAKE_VERBOSE_MAKEFILE=ON         ❮  17s 694ms   
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base (cached)).
-- Application: C:/Users/kartben/zephyrproject/zephyr/samples/subsys/bindesc/hello_bindesc
-- CMake version: 3.27.0-rc4
-- Cache files will be written to: C:/Users/kartben/zephyrproject/zephyr/.cache
-- Zephyr version: 3.5.99 (C:/Users/kartben/zephyrproject/zephyr)
-- Found west (found suitable version "1.0.0", minimum required is "0.14.0")
-- Board: stm32l1_disco
-- Found host-tools: zephyr 0.16.3 (C:/Users/kartben/zephyr-sdk-0.16.3)
-- Found toolchain: zephyr 0.16.3 (C:/Users/kartben/zephyr-sdk-0.16.3)
-- Found BOARD.dts: C:/Users/kartben/zephyrproject/zephyr/boards/arm/stm32l1_disco/stm32l1_disco.dts
-- Generated zephyr.dts: C:/Users/kartben/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: C:/Users/kartben/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: C:/Users/kartben/zephyrproject/zephyr/build/zephyr/dts.cmake
Cygwin WARNING:
  Couldn't compute FAST_CWD pointer.  This typically occurs if you're using
  an older Cygwin version on a newer Windows.  Please update to the latest
  available Cygwin version from https://cygwin.com/.  If the problem persists,
  please see https://cygwin.com/problems.html

Parsing C:/Users/kartben/zephyrproject/zephyr/Kconfig
Loaded configuration 'C:/Users/kartben/zephyrproject/zephyr/build/zephyr/.config'
No change to configuration in 'C:/Users/kartben/zephyrproject/zephyr/build/zephyr/.config'
No change to Kconfig header in 'C:/Users/kartben/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- Configuring done (6.0s)
-- Generating done (0.3s)
-- Build files have been written to: C:/Users/kartben/zephyrproject/zephyr/build
←[92m-- west build: building application
←[0mChange Dir: 'C:/Users/kartben/zephyrproject/zephyr/build'

Run Build Command(s): C:/ProgramData/chocolatey/bin/ninja.exe -v
[1/14] cmd.exe /C "cd /D C:\Users\kartben\zephyrproject\zephyr\build\zephyr && C:\Users\kartben\zephyrproject\.venv\Scripts\python.exe C:/Users/kartben/zephyrproject/zephyr/scripts/build/parse_syscalls.py --scan C:/Users/kartben/zephyrproject/zephyr/include --scan C:/Users/kartben/zephyrproject/zephyr/drivers --scan C:/Users/kartben/zephyrproject/zephyr/subsys/net --json-file C:/Users/kartben/zephyrproject/zephyr/build/zephyr/misc/generated/syscalls.json --tag-struct-file C:/Users/kartben/zephyrproject/zephyr/build/zephyr/misc/generated/struct_tags.json --file-list C:/Users/kartben/zephyrproject/zephyr/build/zephyr/misc/generated/syscalls_file_list.txt "
[2/14] cmd.exe /C "cd /D C:\Users\kartben\zephyrproject\zephyr\build\zephyr && C:\Users\kartben\zephyr-sdk-0.16.3\arm-zephyr-eabi\bin\arm-zephyr-eabi-gcc.exe -x assembler-with-cpp -undef -MD -MF linker_zephyr_pre0.cmd.dep -MT linker_zephyr_pre0.cmd -D_LINKER -D_ASMLANGUAGE -imacros C:/Users/kartben/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -IC:/Users/kartben/zephyrproject/zephyr/include -IC:/Users/kartben/zephyrproject/zephyr/build/zephyr/include/generated -IC:/Users/kartben/zephyrproject/zephyr/soc/arm/st_stm32/stm32l1 -IC:/Users/kartben/zephyrproject/zephyr/drivers -IC:/Users/kartben/zephyrproject/zephyr/soc/arm/st_stm32/common -IC:/Users/kartben/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -IC:/Users/kartben/zephyrproject/zephyr/modules/cmsis/. -IC:/Users/kartben/zephyrproject/modules/hal/stm32/stm32cube/stm32l1xx/soc -IC:/Users/kartben/zephyrproject/modules/hal/stm32/stm32cube/stm32l1xx/drivers/include -IC:/Users/kartben/zephyrproject/modules/hal/stm32/stm32cube/stm32l1xx/drivers/include/Legacy -IC:/Users/kartben/zephyrproject/modules/hal/stm32/stm32cube/common_ll/include -D__GCC_LINKER_CMD__ -DLINKER_ZEPHYR_PREBUILT -E C:/Users/kartben/zephyrproject/zephyr/soc/arm/st_stm32/stm32l1/linker.ld -P -o linker_zephyr_pre0.cmd && "C:\Program Files\CMake\bin\cmake.exe" -E cmake_transform_depfile Ninja gccdepfile C:/Users/kartben/zephyrproject/zephyr/samples/subsys/bindesc/hello_bindesc C:/Users/kartben/zephyrproject/zephyr C:/Users/kartben/zephyrproject/zephyr/build C:/Users/kartben/zephyrproject/zephyr/build/zephyr C:/Users/kartben/zephyrproject/zephyr/build/zephyr/linker_zephyr_pre0.cmd.dep C:/Users/kartben/zephyrproject/zephyr/build/CMakeFiles/d/14501e8e435b2247f884a3646eefd7747033a70ee29079083ba7f7a480c80feb.d"
[3/14] cmd.exe /C "cd /D C:\Users\kartben\zephyrproject\zephyr\build\zephyr\subsys\bindesc && "C:\Program Files\CMake\bin\cmake.exe" C:/Users/kartben/zephyrproject/zephyr/build"
FAILED: zephyr/subsys/bindesc/CMakeFiles/bindesc_time_force_rebuild C:/Users/kartben/zephyrproject/zephyr/build/zephyr/subsys/bindesc/CMakeFiles/bindesc_time_force_rebuild
cmd.exe /C "cd /D C:\Users\kartben\zephyrproject\zephyr\build\zephyr\subsys\bindesc && "C:\Program Files\CMake\bin\cmake.exe" C:/Users/kartben/zephyrproject/zephyr/build"
Loading Zephyr default modules (Zephyr base (cached)).
-- Application: C:/Users/kartben/zephyrproject/zephyr/samples/subsys/bindesc/hello_bindesc
-- CMake version: 3.27.0-rc4
-- Cache files will be written to: C:/Users/kartben/zephyrproject/zephyr/.cache
-- Zephyr version: 3.5.99 (C:/Users/kartben/zephyrproject/zephyr)
-- Found west (found suitable version "1.0.0", minimum required is "0.14.0")
-- Board: stm32l1_disco
-- Found host-tools: zephyr 0.16.3 (C:/Users/kartben/zephyr-sdk-0.16.3)
-- Found toolchain: zephyr 0.16.3 (C:/Users/kartben/zephyr-sdk-0.16.3)
-- Found BOARD.dts: C:/Users/kartben/zephyrproject/zephyr/boards/arm/stm32l1_disco/stm32l1_disco.dts
-- Generated zephyr.dts: C:/Users/kartben/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: C:/Users/kartben/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: C:/Users/kartben/zephyrproject/zephyr/build/zephyr/dts.cmake
Cygwin WARNING:
  Couldn't compute FAST_CWD pointer.  This typically occurs if you're using
  an older Cygwin version on a newer Windows.  Please update to the latest
  available Cygwin version from https://cygwin.com/.  If the problem persists,
  please see https://cygwin.com/problems.html

Parsing C:/Users/kartben/zephyrproject/zephyr/Kconfig
Loaded configuration 'C:/Users/kartben/zephyrproject/zephyr/build/zephyr/.config'
No change to configuration in 'C:/Users/kartben/zephyrproject/zephyr/build/zephyr/.config'
No change to Kconfig header in 'C:/Users/kartben/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- Configuring done (6.3s)
-- Generating done (0.3s)
CMake Error:
  Running

   'C:/ProgramData/chocolatey/bin/ninja.exe' '-C' 'C:/Users/kartben/zephyrproject/zephyr/build' '-t' 'restat' 'build.ninja'

  failed with:

   ninja: error: failed recompaction: Permission denied

CMake Generate step failed.  Build files cannot be regenerated correctly.
ninja: build stopped: subcommand failed.

FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'C:\Users\kartben\zephyrproject\zephyr\build'
``` 

**Environment (please complete the following information):**
 - OS: windows
 - Toolchain: SDK 0.16.3
 - Commit SHA or Version used: 3.5.99 // 92045a021ed3f98716529a7f041923a662ce04b8

**Additional context**

N/A

nordicjm, 2023-10-24:
Needs more information to determine the issue, try `ninja -j1`, also try disabling anti-virus

kartben, 2023-10-24:
> Needs more information to determine the issue, try `ninja -j1`, also try disabling anti-virus

Thanks @nordicjm. No antivirus, full log below:

```
 ~#@❯  ninja -j1 -v                                                                                                                 ❮
[1/128] cmd.exe /C "cd /D C:\Users\kartben\zephyrproject\zephyr\build\zephyr && "C:\Program Files\CMake\bin\cmake.exe" -DZEPHYR_BASE=C:/Users/kartben/zephyrproject/zephyr -DOUT_FILE=C:/Users/kartben/zephyrproject/zephyr/build/zephyr/include/generated/version.h -DVERSION_TYPE=KERNEL -DVERSION_FILE=C:/Users/kartben/zephyrproject/zephyr/VERSION -DKERNEL_VERSION_CUSTOMIZATION="" -P C:/Users/kartben/zephyrproject/zephyr/cmake/gen_version_h.cmake"
-- Zephyr version: 3.5.99 (C:/Users/kartben/zephyrproject/zephyr), build: zephyr-v3.5.0-424-gc40a37c759ec
[2/128] cmd.exe /C "cd /D C:\Users\kartben\zephyrproject\zephyr\build\zephyr && "C:\Program Files\CMake\bin\cmake.exe" -DZEPHYR_BASE=C:/Users/kartben/zephyrproject/zephyr -DOUT_FILE=C:/Users/kartben/zephyrproject/zephyr/build/zephyr/include/generated/app_version.h -DVERSION_TYPE=APP -DVERSION_FILE=C:/Users/kartben/zephyrproject/zephyr/samples/subsys/bindesc/hello_bindesc/VERSION -DAPP_VERSION_CUSTOMIZATION="" -P C:/Users/kartben/zephyrproject/zephyr/cmake/gen_version_h.cmake"
[3/128] cmd.exe /C "cd /D C:\Users\kartben\zephyrproject\zephyr\build\zephyr && C:\Users\kartben\zephyrproject\.venv\Scripts\python.exe C:/Users/kartben/zephyrproject/zephyr/scripts/build/parse_syscalls.py --scan C:/Users/kartben/zephyrproject/zephyr/include --scan C:/Users/kartben/zephyrproject/zephyr/drivers --scan C:/Users/kartben/zephyrproject/zephyr/subsys/net --json-file C:/Users/kartben/zephyrproject/zephyr/build/zephyr/misc/generated/syscalls.json --tag-struct-file C:/Users/kartben/zephyrproject/zephyr/build/zephyr/misc/generated/struct_tags.json --file-list C:/Users/kartben/zephyrproject/zephyr/build/zephyr/misc/generated/syscalls_file_list.txt "
[4/128] cmd.exe /C "cd /D C:\Users\kartben\zephyrproject\zephyr\build\zephyr && C:\Users\kartben\zephyrproject\.venv\Scripts\python.exe C:/Users/kartben/zephyrproject/zephyr/scripts/build/gen_syscalls.py --json-file C:/Users/kartben/zephyrproject/zephyr/build/zephyr/misc/generated/syscalls.json --base-output include/generated/syscalls --syscall-dispatch include/generated/syscall_dispatch.c --syscall-list C:/Users/kartben/zephyrproject/zephyr/build/zephyr/include/generated/syscall_list.h  --split-type k_timeout_t --split-type k_ticks_t"
[5/128] cmd.exe /C "cd /D C:\Users\kartben\zephyrproject\zephyr\build\zephyr && C:\Users\kartben\zephyrproject\.venv\Scripts\python.exe C:/Users/kartben/zephyrproject/zephyr/scripts/build/gen_kobject_list.py --validation-output C:/Users/kartben/zephyrproject/zephyr/build/zephyr/include/generated/driver-validation.h --include-subsystem-list C:/Users/kartben/zephyrproject/zephyr/build/zephyr/misc/generated/struct_tags.json --verbose"
[6/128] cmd.exe /C "cd /D C:\Users\kartben\zephyrproject\zephyr\build\zephyr && C:\Users\kartben\zephyrproject\.venv\Scripts\python.exe C:/Users/kartben/zephyrproject/zephyr/scripts/build/gen_kobject_list.py --kobj-types-output C:/Users/kartben/zephyrproject/zephyr/build/zephyr/include/generated/kobj-types-enum.h --kobj-otype-output C:/Users/kartben/zephyrproject/zephyr/build/zephyr/include/generated/otype-to-str.h --kobj-size-output C:/Users/kartben/zephyrproject/zephyr/build/zephyr/include/generated/otype-to-size.h --include-subsystem-list C:/Users/kartben/zephyrproject/zephyr/build/zephyr/misc/generated/struct_tags.json --verbose"
[7/128] C:\Users\kartben\zephyr-sdk-0.16.3\arm-zephyr-eabi\bin\arm-zephyr-eabi-gcc.exe -DHSE_VALUE=8000000 -DKERNEL -DPICOLIBC_INTEGER_PRINTF_SCANF -DSTM32L151xB -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D_FORTIFY_SOURCE=1 -D_POSIX_C_SOURCE=200809 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -IC:/Users/kartben/zephyrproject/zephyr/kernel/include -IC:/Users/kartben/zephyrproject/zephyr/arch/arm/include -IC:/Users/kartben/zephyrproject/zephyr/include -IC:/Users/kartben/zephyrproject/zephyr/build/zephyr/include/generated -IC:/Users/kartben/zephyrproject/zephyr/soc/arm/st_stm32/stm32l1 -IC:/Users/kartben/zephyrproject/zephyr/drivers -IC:/Users/kartben/zephyrproject/zephyr/soc/arm/st_stm32/common -IC:/Users/kartben/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -IC:/Users/kartben/zephyrproject/zephyr/modules/cmsis/. -IC:/Users/kartben/zephyrproject/modules/hal/stm32/stm32cube/stm32l1xx/soc -IC:/Users/kartben/zephyrproject/modules/hal/stm32/stm32cube/stm32l1xx/drivers/include -IC:/Users/kartben/zephyrproject/modules/hal/stm32/stm32cube/stm32l1xx/drivers/include/Legacy -IC:/Users/kartben/zephyrproject/modules/hal/stm32/stm32cube/common_ll/include -fno-strict-aliasing -Os -imacros C:/Users/kartben/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -fno-printf-return-value -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m3 -mthumb -mabi=aapcs -mfp16-format=ieee --sysroot=C:/Users/kartben/zephyr-sdk-0.16.3/arm-zephyr-eabi/arm-zephyr-eabi -imacros C:/Users/kartben/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix [... cut ...]
No change to Kconfig header in 'C:/Users/kartben/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- Configuring done (6.3s)
-- Generating done (0.2s)
CMake Error:
  Running

   'C:/ProgramData/chocolatey/bin/ninja.exe' '-C' 'C:/Users/kartben/zephyrproject/zephyr/build' '-t' 'restat' 'build.ninja'

  failed with:

   ninja: error: failed recompaction: Permission denied

CMake Generate step failed.  Build files cannot be regenerated correctly.
ninja: build stopped: subcommand failed.
```

yonsch, 2023-11-26:
@kartben Unfortunately I don't have a Windows setup to try this on. Does it happen with other samples/boards as well?