zephyrproject-rtos/zephyr #64311: mgmt: img: nxp: rt1060: image upload causes a silent reset

kunoh, 2023-10-24:
**Describe the bug**
The image upload command can cause a non-deterministic silent reset as a simple print can be a way to solve the issue, probably just hiding it by moving some of the timing about bit.
The reset will occur on any random time during the upload, so the amount of data that has been transferred and written to the flash is different.

 - What target platform are you using?
   - NXP MIMXRT1060
 - What have you tried to diagnose or workaround this issue?
   - Adding/removing code can solve (hide) the issue making it difficult to diagnose.

**To Reproduce**
I have made a small example code that allows one to reproduce the issue. The issue was very reproducible on 3.4.0, as the reset would basically happen every time. I have now tried to make an example code based on 3.5.0 to make sure the issue still persisted, though the issue does not provoke the silent reset every time, it does it most of the time.
The example I have made is built upon the hello_world example on a separate fork;

https://github.com/zephyrproject-rtos/zephyr/commit/e120afeb8ed4060dceef8c16bdde03575067dcf2

Steps to reproduce the behavior:
1. Checkout my fork of zephyr with the modified hello_world example
2. Build a bootloader using the partition overlay that is in the hello_world example
3. Build the hello_world example
4. Flash both firmwares
5. Use any smp client to upload the image (mcumgr) - The silent reset should occur any time. (It may need a couples of tries before it happens)

**Expected behavior**
The firmware should be uploaded successfully without doing a reset

**Logs and console output**
```
Got Sensor Sample Data
Got Imu Sample Data
Got Sensor Sample Data
*** Booting Zephyr OS build zephyr-v3.5.0-1-ge120afeb8ed4 ***
Hello World! mimxrt1060_evk
TestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTt
SensorMock Init
Got Imu Sample Data
Got Sensor Sample Data
Got Sensor Sample Data
```

**Environment (please complete the following information):**
 - OS: 
   - Ubuntu 22.04
 - Toolchain (e.g Zephyr SDK, ...)
   -  Zephyr SDK 0.16.3
 - Commit SHA or Version used
   - Based on tag v3.5.0;
   https://github.com/zephyrproject-rtos/zephyr/commit/e120afeb8ed4060dceef8c16bdde03575067dcf2

**Additional context**
The firmware is built for XIP and the image upload command will do a full erase before writing the new firmware

kunoh, 2023-10-24:
CC @Cladoc

DerekSnell, 2023-10-24:
Hi @kunoh ,
What do you mean by "silent" reset?

The RT10xx devices have the Reset Status Register SRC_SRSR you can read, which reports the sources of the resets.  You can print this register out the console, and then clear all the flags.  When your undesired reset occurs, this register should then help identify what is causing it.

Let us know what you find

kunoh, 2023-10-25:
Hi @DerekSnell,

Sorry if the term "silent" is misleading, I mean that it just do a reset without entering the fault handler or halts.

The SRC_SRSR is 0x3, which points to a cpu_lockup or the SYSRESETREQ being set.
The SRC_GPR5 is 0x0, which may indicate it's a cpu_lockup and not the SYSRESETREQ being set.
The documentation describes that SRC_GPR5 needs to be set by SW when using SYSRESETREQ. 
Assuming any implementations that uses the SYSRESETREQ bit would also use the SRC_GPR5 to indicate that it is not a lock up.

From the CM7 documentation;
The processor enters a lockup state if a fault occurs when executing the NMI or HardFault handlers.

We have tried with a breakpoint in both the nmi handler and all the underlying fault type handlers without success.

DerekSnell, 2023-10-25:
Hi @kunoh ,
I am not sure that `SRC_GPR5` is written before SYSRESETREQ is triggered.  I would not make that assumption without confirming it.

But a lockup reset seems likely given your symptoms.  SYSRESETREQ is a software reset and is usually intentional.  Lockups can be erratic, and challenging to track down in the code.

It is not uncommon for a lockup to occur without trapping in a fault exception.  For example, if the CPU is not able to fetch instructions from the memory for some reason, a fault is triggered.  Then the CPU will attempt to fetch the fault exception handler address, or the instruction in that exception.  If that triggers a 2nd fault, the core will enter the lockup state.

But it is not clear to me how the image upload is related to the reset you see.  Your description and reproduction steps state the image must be uploaded.  But your console output shows the reset occurred while executing your modified hello_world app, not during the upload.  Does this mean the image upload succeeded, and then the uploaded app executes and has sporadic resets?  Does this mean if you flash the app in the same location, and execute the app without ever uploading it, the resets never occur?

Thanks

kunoh, 2023-10-25:
In order to reproduce it, one must execute the modified hello_world example, it includes the smp server and the image commands.
The hello_world example will run perfectly fine on the device, until the action/event of an image upload occur.
It will then at some point while writing the chunks of data from a binary file to the flash, do a reset.

The reset can only be triggered by executing the image upload command from a smp client tool.

The console output is to illustrate that the application is running fine and then suddenly prints out the boot logs because of a reset occured. The reset was triggered by a smp client which tried to upload a firmware using the image upload command.

@Cladoc is currently looking into moving the fault handler instructions into ITCM as they are currently in flash. The CPU can probably not fetch the instructions while writing to the flash, if a fault happens here.

DerekSnell, 2023-10-25:
Hi @kunoh ,
Yes, based on this, it does seem the issue is related to XIP from flash while updating the same flash.

If the flash is being erased or programmed, the app cannot access the flash.  This includes accessing the hardware vector table if it is in flash, and any ISRs.  It also includes all masters in the device that can access memory, like the CPUs, DMAs, USB, Ethernet, display driver, etc..   The flash driver should execute from RAM during the flash operation.  But it seems your app is still accessing the flash at some point.  I suspect if you moved all your code to ITCM, and then uploaded the image to flash, you would not see these resets.

Have you tried executing the [SMP server](https://docs.zephyrproject.org/latest/samples/subsys/mgmt/mcumgr/smp_svr/README.html) sample to see if it has a similar issue?  If SMP server does not show the issue, is there some difference between it and your application that may explain that?

Best regards

kunoh, 2023-10-26:
Hi @DerekSnell ,

Yes, I have tried with the SMP server sample and were not able to reproduce it.
I don't know if there could be any relations to the compiled binary size, as I need to add code until the size reaches a certain area - and just out of the box the SMP server sample is far away from this.

I can try to work some more with the SMP server sample and see if I can reproduce it using that.

I have made a small script that keeps adding a new print to the source code and then tests if the upload command still works. It seems like I can use prints to fine tune the issue occurrence.

Cladoc, 2023-10-26:
Hi @DerekSnell
First of all, thank you for taking time to comment on our issue that @kunoh opened. I would like to add two things to the conversation:

1) I do not know if the information is useful, but here are some notes for a little context regarding the come-and-go nature of the issue that we've experienced. This has been the main hassle in terms of debugging. 

Adding or removing functionality (just .text) can seemingly hide and with time (more code) later reintroduce the reset as we've seen during development. Because of this, in an attempt to make as little changes to the binary as possible now that we have a minimal stable example, I have tried moving just the vector table and cortex_m/fault_s.S assembly that defines the fault handler entry points to ITCM. 
Doing this, the reset still occurs during image upload, but a break point in fault_s.S is not hit. 
However, this is reasonable because we build with -Os, not debug optimizations, and the break points are thus more unreliable. The same is evident if I force, say, a bus fault.

However, building with debug optimizations  `CONFIG_DEBUG_OPTIMIZATIONS=y` hides the issue again, probably affecting the application in a way like adding or removing some amount of functionality will.
We have seen it occur sporadically with debug optimizations, but have not as of yet found a code state where the reset is easily reproducible. This all in all adds to the difficulty of debugging the reset.

If I move more to ITCM, like cortex_m/fault.c (in hopes of some logging), the reset disappears. This is again likely not because it solves the underlying issue, but instead something aligning in a certain way.

2) We have noted that whenever the reset occurs, it does so while the following while loop in flash_mcux_flexspi_nor.c:flash_flexspi_nor_write() is executing:

```
while (len) {
		/* If the offset isn't a multiple of the NOR page size, we first need
		 * to write the remaining part that fits, otherwise the write could
		 * be wrapped around within the same page
		 */
		i = MIN(SPI_NOR_PAGE_SIZE - (offset % SPI_NOR_PAGE_SIZE), len);
#ifdef CONFIG_FLASH_MCUX_FLEXSPI_NOR_WRITE_BUFFER
		memcpy(nor_write_buf, src, i);
#endif
		flash_flexspi_nor_write_enable(dev);
#ifdef CONFIG_FLASH_MCUX_FLEXSPI_NOR_WRITE_BUFFER
		flash_flexspi_nor_page_program(dev, offset, nor_write_buf, i);
#else
		flash_flexspi_nor_page_program(dev, offset, src, i);
#endif
		flash_flexspi_nor_wait_bus_busy(dev);
		memc_flexspi_reset(data->controller);
		src += i;
		offset += i;
		len -= i;
	}
```
Which I think also adds to the suspicion that a flash read-access is attempted by _something_ during the write operation. 

I guess the next step will be to try to get the smp_server sample to fail, possibly with debug optimizations.

Cheers

DerekSnell, 2023-10-27:
Hi @kunoh and @Cladoc ,
This sounds like a tricky issue to debug.  It does seem like a flash read is occurring at some point during a flash erase/program operation.

Have you already tried disabling the cache to help find the root cause?  If not, the cache may be part of the reason the issue seems so inconsistent.  If a flash read is triggering the resets, then much of the time that read may fetch from the cache, and not actually access the flash.  And small simple changes to the code will affect when that data is in cache versus a cache miss.  At least for diagnosing the root cause, disabling the cache may help the issue be more consistent, and easier to track down.

Have you tried disabling interrupts globally during the image upload?  When the flash driver performs an operation on the flash, it executes from RAM.  So I would expect that any flash read at that time would be triggered by an interrupt somehow.  If disabling all interrupts resolves the issue, but is not acceptable in the application, then you could narrow down which interrupt is causing the issue.

Let us know what you find

kunoh, 2023-10-30:
Hi @DerekSnell ,

We are a bit in doubt if we are disabling the cache correctly.
We have tried disabling `CONFIG_DCACHE` and `CONFIG_ICACHE`, but the Configuration and Control Register, CCR, still shows that I cache and D cache is enabled.
By disabling these two configs we can not reproduce the issue, so something is changing.
**Is it the correct way to disable the caches or can you let us know how it should be done?**

Regarding disabling global interrupts, we have not tried this yet and will get back to you once we know more.

Thank you.

DerekSnell, 2023-11-01:
Hi @kunoh ,
We have identified a potential issue in the flash driver that may be causing your issue.  Disabling interrupts or the cache will not prevent this issue.  BTW, the flash driver does disable interrupts during the flash operations.

There is some code in the driver that reads the device structure during the flash operations.  And that stuct is located in flash.  So this appears to lead to a Read-While-Write situation, where the CPU attempts to read from the flash during an operation.  

NXP will work to fix this driver to avoid that issue.  And since this issue is challenging to replicate, we would like your help to test the update, and see if it resolves your issue.

In the meantime, if you want to try to workaround this issue, the structs used by the driver need to be placed in RAM.  You can refer to the changes in https://github.com/zephyrproject-rtos/zephyr/commit/e58d0c3bb5a7d3e48bd60697578fa43797617c8 where this was done in the HyperFlash driver.

Thank you for raising this to us.

Cladoc, 2023-11-01:
Hi @DerekSnell,

That's great news. I just noticed something of the same today, but didn't get around to the memory map yet.
On my working commit that also sees the issue, but which was not the minimal example referred to in this issue that @kunoh worked on, I was easily able to reproduce the reset with the data cache disabled as you suggested. Adding/Removing functionality and different optimization options no longer affected the consistency of the issue to the same extent as before. The data cache line containing the struct located in flash was probably being evicted during the upload process at what seemed like random times. 

I noticed the issue being timing-sensitive. A breakpoint on the `flash_flexspi_not_wait_bus_busy` in the flash-page programming while loop 
```
#else
		flash_flexspi_nor_page_program(dev, offset, src, i);
#endif
		flash_flexspi_nor_wait_bus_busy(dev);                   <-------------------
```
Seems to give enough time to whatever race condition is occurring to finish. Continuous stepping with "continue" allowed the upload to progress with no reset (albeit slowly) as long as a breakpoint is present on said line.
The process of programming the flash page always fails on the first `flash_flexspi_nor_read_status` when derefencing `data->port` as part of the struct initialization:
```
flexspi_transfer_t transfer = {
		.deviceAddress = 0,
		.port = data->port,
		.cmdType = kFLEXSPI_Read,
		.SeqNumber = 1,
		.seqIndex = READ_STATUS_REG,
		.data = status,
		.dataSize = 1,
	};
```
Which makes sense if the accessed structs are in flash as you point out. I will temporarily move the structs to RAM then as per your suggestion. 

Unfortunately @kunoh is on holiday for a few weeks. I will try to let him know of the progress.
I will be available for testing your update with my own version of the application and @kunoh 's linked example.

Best Regards

Cladoc, 2023-11-01:
Hi @DerekSnell,
An additional question. We had some discussions regarding the memory placement of files concerning fault handling when performing operations like these. With the cache disabled I moved these to ITCM and occasionally managed to get some useful information on the type of fault occurring and the precise instruction addresses.

Is there any Kconfig flag that we have missed that ensures these are always placed in RAM memory? If not, do you know if there any considerations as to add such an option in the future for ease of use?

DerekSnell, 2023-11-01:
> is there any Kconfig flag that we have missed that ensures these are always placed in RAM?
Hi @Cladoc ,

I am not aware of a Kconfig to do that.  But if you have a solution, you are welcome to contribute it as a Pull Request.

Best regards

Cladoc, 2023-11-04:
Hi @DerekSnell,

I have edited my answer regarding the cache disable. It makes the reset easier to reproduce, but it seems there is still a race condition taking place.

From my understanding the HyperFlash workaround moves the controller device struct to RAM on driver initialization, by memcpy'ing it via the pointer in dev->config, to dev->data as seen below
```
static int flash_flexspi_hyperflash_init(const struct device *dev)
{
	const struct flash_flexspi_hyperflash_config *config = dev->config;
	struct flash_flexspi_hyperflash_data *data = dev->data;

	if (!device_is_ready(data->controller)) {
	/* Since the controller variable may be used in critical sections,
	 * copy the device pointer into a variable stored in RAM
	 */
	memcpy(&data->controller, config->controller, sizeof(struct device));
```
I am not sure this solves the entire problem. The struct pointed to by `dev` which holds the `data` and `config` pointers, is itself located in flash. Thus, while the workaround ensures the controller device struct is located in RAM, it might occasionally not be accessible by the `data` pointer in `dev`.

This explains why we have seen that the dereference of port:
```
static int flash_flexspi_nor_read_status(const struct device *dev,
		uint32_t *status)
{
	struct flash_flexspi_nor_data *data = dev->data;
	
	flexspi_transfer_t transfer = {
		.deviceAddress = 0,
		.port = data->port,
		.cmdType = kFLEXSPI_Read,
		.SeqNumber = 1,
		.seqIndex = READ_STATUS_REG,
		.data = status,
		.dataSize = 1,
	};
```
_can_ lead to a Bus Fault before the controller struct is accessed. This fault can occur both during flash erase and flash write, where the status is polled.

Would this be something NXP can look into and confirm or refute?

danieldegrasse, 2023-11-06:
@kunoh, can you try to reproduce this issue with #64873? I believe I've removed any flash access while programming in that PR. As @Cladoc correctly points out, the `dev` struct of the FlexSPI driver itself also cannot be dereferenced during programming, so the helper functions should only take a reference to the driver's `dev->data` structure as an argument.

Cladoc, 2023-11-08:
Hi @danieldegrasse, would you prefer @kunoh verified it (he is back in office in 2 weeks) since he was the one to open the issue, or would I be able to take a look?

Cheers

closed 2023-11-24 by kunoh