zephyrproject-rtos/zephyr #64327: Documentaion: CMSIS-DSP updates with the Zephyr 3.5 release

spe-ciellt, 2023-10-24:
**For others struggling with getting CMSIS DSP to work with the 3.5 release**

The split out of CMSIS-DSP to a separate module is not mentioned in any of the 3.5 release documentation, neither Migration Guide nor Release Notes. The solution is simply to add `cmsis-dsp` to the import list of `west.yml` .

The update of CMSIS-DSP has caused trouble with macro names colliding, in particular `ROUND_UP()` and `ROUND_DOWN()`. The CMSIS-DSP have wrapped their macro definitions in `#ifndef` while Zephyr hasn't. So my current solution is to - in the C file -  include all Zephyr files first and then include `arm_math.h`. I am fully aware of that this is a hack, but "desperate needs gives desperate measures".

Zephyr will not fix the macro names, I got referred to the Coding Guidelines, this chapter https://docs.zephyrproject.org/latest/contribute/coding_guidelines/index.html#rule-a-3-macro-name-collisions

I have added some comments on this to the ARM CMSIS DSP issues https://github.com/ARM-software/CMSIS-DSP/issues/99

henrikbrixandersen, 2023-10-24:
CC: @fabiobaltieri @jhedberg

fabiobaltieri, 2023-10-25:
@spe-ciellt thanks for reporting, the release notes are contributed as with the rest of the code and it happens that things get missed. I think it's probably worth adding a line post-release since many users are probably grabbing the notes from the lastest version anyway and we can backoprt it as well. Would you be able to open a PR to add that?

Long term, I see that the macro has been renamed upstream (https://github.com/ARM-software/CMSIS-DSP/commit/00bab39e8f64ca32aa6707ea5e1fa6ca95dd5276), probably worth updating the module so that the next release won't have the issue anymore (cc @stephanosio @galak @XenuIsWatching)